subsurface/subsurface #4052: Use imported GF (instead of the Tech menu entry) to calculate TTS

OlivierV78, 2024-01-19:
### Describe the issue:

- [ ] Bug
- [X] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:

Would it be possible to use the imported GF instead of the entered in the tech diving menu entry ?
Actually I have to switch the GF every time I change it for my dive.
Use of the imported from the computer would be better.

### Operating system:

Linux Kubuntu
Windows 10
French

Shearwater Perdix AI + Petrel 3

Yes ; on every OS

### Subsurface version:

6.0.5051-CICD

Yes

Official release

### Steps to reproduce:

Only open a dive

### Current behavior:

Subsurface uses the GF in the menu

### Expected behavior:

Should use the imported GF from the computer to calculate the TTS

### Additional information:

### Mentions:

[image]

atdotde, 2024-01-21:
Currently, this is not planned as the display GF is meant as as an analytic tool: You can check what kinds of GF combinations your dive profile is compatible with and for which not. Therefore the GF used to compute the ceiling is not fixed to the GF used by your dive computer during the dive (and of course many dive computers don't report a GF, many will not even have used one since there are many different decompression algorithms implemented in devices).

OlivierV78, 2024-01-22:
I understand.
But would it be possible to add this functionality (which is implemented in the shearwater logbook) ?
It is helpful to see the ceiling for a logged dive.

Another problem in subsurface-mobile : the GF always comme back to 30/75. Do you know this bug ?

mikeller, 2024-01-22:
> I understand. But would it be possible to add this functionality (which is implemented in the shearwater logbook) ?

Shearwater's logbook is providing a set of functionality that is different from what Subsurface is trying to provide: All they have to support is dive computers from one manufacturer, themselves, so they do not have to care that dive computers from different manufacturers use different formats to report 'extra' information like gradient factors, and use slightly different algorithms to calculate decompression data based on these gradient factors.
Subsurface on the other hand is trying to support as many dive computer manufacturers and models as possible, and there is limited interest in implementing functionality if it will work for just one manufacturer's dive computers.

> It is helpful to see the ceiling for a logged dive.

This is already supported in Subsurface - and based on the most relevant data: The ceiling that was displayed by the dive computer during the dive.
To get this just enable 'DC reported ceiling' in the profile options:

[image]

> Another problem in subsurface-mobile : the GF always comme back to 30/75. Do you know this bug ?

The gradient factor used for the calculated ceiling shown in the profile for logged dives is a setting that is adjustable in the preferences:

[image]

OlivierV78, 2024-01-23:
> This is already supported in Subsurface - and based on the most relevant data: The ceiling that was displayed by the dive computer during the dive. To get this just enable 'DC reported ceiling' in the profile options:

OK Thank you. I was wrong on this point and thought that the red ceiling was calculated too

> The gradient factor used for the calculated ceiling shown in the profile for logged dives is a setting that is adjustable in the preferences:

I know this menu entry on the computer and the mobile soft.
But I have a bug on this point on my smartphone (Galaxy S21) (only in the smartphone mobile version, not in the computer version) :
when I try to change the settings it always come back to 30/75.
I will open another issue for this with the screenshot
This issue can be closed

mikeller, 2024-01-23:
@OlivierV78:

 > But I have a bug on this point on my smartphone (Galaxy S21) (only in the smartphone mobile version, not in the computer version) :

Ah yes, this is actually a bug that has already been fixed (in #3912), and android versions including the fix are available from https://subsurface-divelog.org/current-release/.
(As an open source project we do not have the resources to keep up with the ever changing minimum version requirements enforced by Google for applications published to the Play store...)

OlivierV78, 2024-01-24:
> Ah yes, this is actually a bug that has already been fixed (in #3912), and android versions including the fix are available from https://subsurface-divelog.org/current-release/. (As an open source [...]

I did the update but same problem : only GF 30/75 and always come back to this setting.

mikeller, 2024-01-24:
> I did the update but same problem : only GF 30/75 and always come back to this setting.

Yes, you are right - looks like something is broken again. I'll have a look and fix.

mikeller, 2024-01-28:
@OlivierV78: The problem with setting a gradient factor in the mobile app should be fixed in https://github.com/subsurface/nightly-builds/releases/tag/v6.0.5078-CICD-release.

OlivierV78, 2024-01-28:
@mikeller Sorry buy always the same problem, but now it displays GF10/10 instead of GF 30/75 (even if the settings are on GF 80/80 for example)
Another problem is that the box "display the GF ON/OFF" don't work.

mikeller, 2024-01-28:
@OlivierV78:

> Sorry buy always the same problem, but now it displays GF10/10 instead of GF 30/75 (even if the settings are on GF 80/80 for example) 

That's odd - this is working fine on my Android device. Just to make sure this isn't a case of corrupted data, can you please uninstall Subsurface-mobile, and then re-install it, and try again.

> Another problem is that the box "display the GF ON/OFF" don't work.

Oh my, looks like there is more brokenness in this. :-)
I had never noticed these being broken, but I'll get them fixed.

OlivierV78, 2024-01-29:
I uninstalled, empty the cache (but not the data) and install v6.0.5079-CICD-release but same problem GF10/10.
Should I clear all the data (in this case I will check my computer backup before) ?

mikeller, 2024-01-29:
I would assume that the settings are persisted in the (local) data, so you will have to clear this.
If you are using cloud storage then all the data that has been synched (check on https://cloud.subsurface-divelog.org/) will be unaffected by this, you will just have to re-login to the cloud on the following install.

OlivierV78, 2024-01-29:
I cleaned the data (see picture in attachment), uninstalled, reboot, and installed again.
But always same problem

A strange thing : I did not have to give my login/password for cloud storage ; it has automatically come back

[image]
[image]
[image]

dirkhh, 2024-01-29:
> I cleaned the data (see picture in attachment), uninstalled, reboot, and installed again. But always same problem
> 
> A strange thing : I did not have to give my login/password for cloud storage ; it has automatically come back

Proof positive that you did NOT clear the data. Your mobile device decided to retain the log in information - and likely kept all the other settings as well (as those are stored in the same place).

OlivierV78, 2024-01-29:
OK but in first picture you can see that there is no data...
So where are this data to delete them manually ?

mikeller, 2024-01-29:
@OlivierV78: Clutching at straws here, but is this a corporate phone or one that came locked to a contract, meaning that it has got some sort of locking software installed?

OlivierV78, 2024-01-30:
No it is a personnal phone
Galaxy S21
Android 14
One UI 6.0

OlivierV78, 2024-01-31:
@mikeller 
No it is a personal phone
Galaxy S21
Android 14
One UI 6.0

OlivierV78, 2024-02-05:
I tried again :
- total deleting data off the app
- uninstall
- wipe cache for all the phone (at startup of the phone)

But same problem after install : subsurface finds all the date alone !
Where are this data stored ?

OlivierV78, 2024-02-06:
@mikeller 
Yes In the 6.0.5090 the ceiling settings "on/off" is fixed.
But when it is on I always have GF10/10 until when I set it to GF80/80 for example.
I never set GF to 10/10 manually. So I don't understand where this is coming from.
So always the same problem for me.

To have this setting  working I need to use the old Playstore version...

OlivierV78, 2024-02-08:
@mikeller 
I have cleared all the data and wipe cache and I tried again with 6.0.5094.
After doing this, I HAD TO LOG IN, no dives in the logbook and it took a few seconds to get the dives downloaded.
At fist opening the GF was 30/75.
After this when I put 80/80 it display 10/10.

mikeller, 2024-02-08:
@OlivierV78:
> After doing this, I HAD TO LOG IN, no dives in the logbook and it took a few seconds to get the dives downloaded.

That's progress!

> At fist opening the GF was 30/75. After this when I put 80/80 it display 10/10.

I just tested with 6.0.5094 and for me an 80/80 GF works on my device. How are you setting the GFlow / GFhigh settings? The UI components that we are using are a bit picky, and you either have to use the little [+] and [-] buttons, or if entering through the numerical keyboard you have to remove the '%' from the input.

OlivierV78, 2024-02-08:
@mikeller 
I cannot use + - (see screenshot) as the + is missing ; the screens seems to be cut.
With removing the %, I tried to put 80/80 but I got 80/10 and impossible change the 10 ! (I tried many times)

I noticed that the "-" is working with the GF Low but now with the GF high it is on 10 
But the "-" is working with the GF High when it is for example on 85. But after this working way it drops drops down to 10...

[image]

mikeller, 2024-02-12:
@OlivierV78: Not sure what is causing these fields to be truncated on your device. @dirkhh do we have anybody who has got a good grip on QML?

dirkhh, 2024-02-12:
This is 90% my work.
So the answer is really "try and error and asking on their chat server"

dirkhh, 2024-02-12:
I'd love to see the startup log file of the device, because this looks like our heuristic for picking grid width is failing. Often switching to a smaller font in the settings can help work around such issues.

OlivierV78, 2024-02-12:
@dirkhh @mikeller What do you need exactly ?

dirkhh, 2024-02-12:
Easiest would be to exit Subsurface-mobile, start it again, open the settings page that shows the problem, and then send a support request from the help menu.
That will include the logs I was hoping to see.

OlivierV78, 2024-02-12:
@dirkhh 
Here the file
[subsurface.log](https://github.com/subsurface/subsurface/files/14249384/subsurface.log)

dirkhh, 2024-02-12:
Yeah, you have one of those crazy annoying phones that for some reason lie about their screen resolution for a moment after the app first starts.
Initially your phone tells us that your screen is 360 x 725  - but then it changes its mind and claims 450x675, only to go back to 360 x 725. And depending on the exact sequence of resolutions that are being thrown at us, sometimes that confuses our sizing algorithm.

Can you in the settings -> theme manually select a smaller font? That should fix it.
If not, then I need to once again try to fine-tune the heuristic that tries to catch these situations. I really just wish the interaction between QML and some phones wasn't so painful.

OlivierV78, 2024-02-12:
@dirkhh 
Ok thank you : I was able to try some settings of my GF with a smaller font but it continue to be difficult because sometimes it drop down itself to GF10/* or GF*/10. In attachment the log

[subsurface.log](https://github.com/subsurface/subsurface/files/14249600/subsurface.log)

OlivierV78, 2024-02-14:
@mikeller @dirkhh 

Hi,

Thank you very much for the v6.0.5099 update that I tested.
I can now get the button to work in normal font but pressing remains difficult because it is small.
It would be better if it was a little bigger (e.g. two to three times wider).

Furthermore, by manipulating the buttons, one of the GFs resets sometimes to 10, for example, as soon as you change lines.

dirkhh, 2024-02-14:
The buttons are for small changes - I generally think it's better to just enter the GF that you want into the text field. Maybe I should just do away with the buttons and make this a simple text field in the first place.

mikeller, 2024-02-15:
> The buttons are for small changes - I generally think it's better to just enter the GF that you want into the text field. Maybe I should just do away with the buttons and make this a simple text field in the first place.

I changed it from a simple text field in #3912, to get improved options for input validation (i.e. limiting to the actual 10 to 150). Plus some users had complained about the keyboard popping out and overlapping the edit when trying to edit, which can be worked around by using the buttons.

dirkhh, 2024-02-15:
valid points. I need to play with this on device to see what would make these easier to use, I guess.
more stuff for the long todo list working on Subsurface-mobile

mikeller, 2024-02-15:
@dirkhh: #4097 should improve the situation for these settings. Turns out I did not read the (right) documentation for QML when I did #3912, and did not realise that a `valueFromText` was needed.

OlivierV78, 2024-02-15:
> The buttons are for small changes - I generally think it's better to just enter the GF that you want into the text field. Maybe I should just do away with the buttons and make this a simple text field in the first place.

@dirkhh Use of text fields is very buggy : most time it drops down to GF 10 when I use it, even if I remove the %.

@mikeller seems to know this problem because he said me in a previous post "The UI components that we are using are a bit picky, and you either have to use the little [+] and [-] buttons, or if entering through the numerical keyboard you have to remove the '%' from the input.

mikeller, 2024-02-17:
@OlivierV78: Try https://www.subsurface-divelog.org/latest-release/ - this should fix the gradient factor entry.

OlivierV78, 2024-02-17:
@mikeller Thank you. It is ok !

PR #4071 (2024-01-28): Mobile: Fix Gradient Factor Preference Setting.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix the persisting and use of gradient factor preferences for dive profiles in the mobile version.
Also rename the mobile backend gradient factor settings to make it obvious that they are used by the (not currently enabled) planner.

### Changes made:

1. fix the persisting and use of gradient factor preferences for dive prof [... cut]
PR #4079 (2024-02-05): Mobile: Fix Configuration of Ceiling Display.

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix the configuration of the deco ceilings in the mobile version.

### Changes made:

1. make the settings work;
2. remove reading of the dive computer ceiling from git;
3. hide the gradient factor in the profile when the calculated ceiling is not shown;
4. when the calculated ceiling is disabled in the settings, disable editin [... cut]
closed 2024-02-17 by OlivierV78