subsurface/subsurface #4053: AppImage crash with current build

exxus, 2024-01-19:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Running the Appimage on Arch Linux fails.
Self compiled version works fine.

Error message:
```
./Subsurface-v6.0.5062-CICD-release.AppImage 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

[1]    157414 IOT instruction (core dumped)  ./Subsurface-v6.0.5062-CICD-release.AppImage
```

### Operating system:

Arch Linux Linux 6.6.12-1-lts
Gnome 45.3, Wayland

### Subsurface version:

Crash with
./Subsurface-v6.0.5062-CICD-release.AppImage
./Subsurface-v6.0.5059-CICD-release.AppImage
./Subsurface-v6.0.5054-CICD-release.AppImage

No crash when building from source.

### Steps to reproduce:

1) run AppImage.

### Current behavior:

AppImage crashes.

### Expected behavior:

No crash.

### Additional information:

```
(gdb) run
Starting program: /home/exus/Downloads/Subsurface-v6.0.5062-CICD-release.AppImage 

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
[Thread debugging using libthread_db enabled]                                                                                                                                                                                                 
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 158819]                                                                                                                                                                                              
process 158816 is executing new program: /tmp/.mount_Subsurz2HSmx/usr/bin/subsurface
warning: Loadable section ".dynstr" outside of ELF segments
  in /tmp/.mount_Subsurz2HSmx/usr/bin/subsurface
[Thread debugging using libthread_db enabled]                                                                                                                                                                                                 
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffe45ff6c0 (LWP 158825)]                                                                                                                                                                                                      
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

Thread 1 "AppRun" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;                                                                                                                                                       
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff6eac8a3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff6e5c668 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff6e444b8 in __GI_abort () at abort.c:79
#4  0x00000000005165c7 in messageHandler(QtMsgType, QMessageLogContext const&, QString const&) (type=<optimized out>, msg=<optimized out>) at /__w/subsurface/subsurface/subsurface-desktop-main.cpp:236
#5  0x00007fffef29c6b0 in qt_message(QtMsgType, QMessageLogContext const&, char const*, __va_list_tag*) () at /tmp/.mount_Subsurz2HSmx/usr/bin/../lib/libQt5Core.so.5
#6  0x00007fffef285fd4 in QMessageLogger::fatal(char const*, ...) const () at /tmp/.mount_Subsurz2HSmx/usr/bin/../lib/libQt5Core.so.5
#7  0x00007fffeff4ef48 in QGuiApplicationPrivate::createPlatformIntegration() () at /tmp/.mount_Subsurz2HSmx/usr/bin/../lib/libQt5Gui.so.5
#8  0x00007fffeff4f25d in QGuiApplicationPrivate::createEventDispatcher() () at /tmp/.mount_Subsurz2HSmx/usr/bin/../lib/libQt5Gui.so.5
#9  0x00007fffef47c60f in QCoreApplicationPrivate::init() () at /tmp/.mount_Subsurz2HSmx/usr/bin/../lib/libQt5Core.so.5
#10 0x00007fffeff50d8c in QGuiApplicationPrivate::init() () at /tmp/.mount_Subsurz2HSmx/usr/bin/../lib/libQt5Gui.so.5
#11 0x00007ffff0965479 in QApplicationPrivate::init() () at /tmp/.mount_Subsurz2HSmx/usr/bin/../lib/libQt5Widgets.so.5
#12 0x000000000050f18b in main(int, char**) (argc=<optimized out>, argv=0x7fffffffdbf8) at /__w/subsurface/subsurface/subsurface-desktop-main.cpp:44
```

### Mentions:

dirkhh, 2024-01-20:
Now that I read through this one more time... so it looks to me like you are only running wayland and no traditional X11 server. Does that sound right? I haven't used Arch... in a while :)
I need to figure out if / how we can enable Wayland in an AppImage. That's a new challenge for me.

I do happen to know that you are up and running with a locally compiled version - but still, I'd love for this to work for more people.

Thanks for the report.

exxus, 2024-01-20:
Thanks for the comment. Since this seems to be wayland related, I looked deeper into it and found the following:

The crash is caused by my environment variable: QT_QPA_PLATFORM=wayland
Since a run a wayland session, I have this environment variable set to wayland which is useful for some qt based applications.

Unfortunately, this causes problems for qt applications that don't support wayland.

Setting the environment variable to 'xcb' or an empty string fixes the problem.

```
QT_QPA_PLATFORM=xcb ./Subsurface-v6.0.5062-CICD-release.AppImage
QT_QPA_PLATFORM= ./Subsurface-v6.0.5062-CICD-release.AppImage
```

closed 2024-01-20 by exxus

dirkhh, 2024-01-20:
Oh awesome. Thank you so much for figuring this out and sharing it.
Much appreciated!