subsurface/subsurface #4056: Gas switches / current gas missing in UDDF export

KG32, 2024-01-20:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When exporting a UDDF, `<switchmix/>` elements are missing. 

### Operating system:

macOS Sonoma 14.0 (23A344), English

Macbook Pro 16, Apple M1 Pro

### Subsurface version:

5.0.10

Not tested

Official release

### Steps to reproduce:

- Export dive(s) with gas switches as UDDF
- Open the exported logbook

### Current behavior:

- Gas switches are gone, all dives seem to be on air only
- In the .uddf file there's no `<switchmix/>` elements in respective `<waypoint>` elements
### Expected behavior:

- `<switchmix/>` elements are present in the .uddf file
- gas switches are visible when opening exported UDDF

### Additional information:

### Mentions:

closed 2024-01-30 by dirkhh

mikeller, 2024-01-30:
@KG32: Can you please test with https://github.com/subsurface/nightly-builds/releases/tag/v6.0.5081-CICD-release to verify that the UDDF files produced by Subsurface (now with `<switchmix/>` elements) are correctly parsed by the application you want to use them for?

KG32, 2024-02-03:
@mikeller seems like the exported UDDF files contain correct `<switchmix/>` elements now. Thanks a lot!

PR #4075 (2024-01-30): Desktop: Fix the Generation of <switchmix> Elements in the UDDF Export.

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix the generation of <switchmix> elements in the UDDF export. Also change the ids of gasmixes to include the helium fraction, and fractions of a percent, in order to make UDDF export suitable for technical diving.

### Changes made:

1. fix the generation of <switchmix> elements in the UDDF export;
2. [... cut]