subsurface/subsurface #4086: Crashes in Dive Planner

BlackFire01, 2024-02-06:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When using the dive planner, there are two quite annoying crashes I encountered. This is bad because if not saved, all the planned deco gases etc. are cleared and you start from scratch. In both of these cases, when the action leading up to the crash is made, the app freezes and then crashes completely.
The first one sometimes appears when deleting a tank. See below for how to reproduce it, in my case 100% of the time.
The second one is a bit more troubling. It happens when you have the tissue heatmap enabled with multiple gasses. If you then try to set a surface interval after the dive to judge off-gasing effectiveness, the app just crashes. You cannot set the interval at all. Also occurs if you accelerate the decompression with more than one gas already set.

### Operating system:
My system is running Windows 11, Version 10.0.23615 Build 23615.
I have tested both crashes with English and German language setting, they occur regardless.

### Subsurface version:
I am running Subsurface 6.0.5079-CICD-release.
The issues have been occurring since I started working with the dive planner.
I did not compile the code myself.

### Steps to reproduce:

Both crashes are completely independent of algorithm choice.

Crash 1 - Tank Removal
1) Go to Dive Planner.
2) Set the first depth marker to something deeper (e. g. 60m - 200ft).
3) Set the second default depth marker at the same depth for some time (I chose the default 19 minutes).
4) Add a tank not containing air. I chose EAN50. Tank properties should not matter, but I selected 12l steels and 200bar initial pressure.
5) Add another tank not containing any of the currently available gases. I chose EAN99.
6) Now simply delete the first tank you added (should be in the middle) and see the window vanish.

Crash 2 - Surface Interval
1) Go to Dive Planner.
2) Add another tank (even air works).
4) Enable tissue heatmap.
5) Now, add a surface interval on the bottom of the middle column of the dive parameters. Crash ensues.

### Current behavior:
Well, the app crashes.

The second problem does NOT occur if the additional tanks/gasses are set to "unused" before adding the interval. The problem does not occur at all if there is only one used gas, even if the bottom gas is not air.

### Expected behavior:
In the first case, the gas should simply be removed and the decompression schedule re-adjusted if necessary.
For the second part, one should expect a surface interval to show effectiveness of off-gassing in heatmap, typically used for accelerated deco planning.

### Additional information:

### Mentions:

dirkhh, 2024-02-07:
Well, I tried to reproduce this - but of course I avoid doing anything on Windows as much as I can. And on Linux I cannot reproduce the crash. So I'll keep looking at this - just so you know this isn't being ignored.

dirkhh, 2024-02-07:
Even on Windows I cannot reproduce a crash in the first scenario, but the dive planner window simply closes - which isn't the intended behavior for sure.
The second scenario does indeed create a crash. And the stack trace is rather weird.
```
#0  0x00007ffe2dcb6f0f in ntdll!RtlRegisterSecureMemoryCacheCallback ()
   from C:\WINDOWS\SYSTEM32\ntdll.dll
#1  0x00007ffe2dc81805 in ntdll!EtwLogTraceEvent () from C:\WINDOWS\SYSTEM32\ntdll.dll
#2  0x00007ffe2dcb5278 in ntdll!RtlRegisterSecureMemoryCacheCallback ()
   from C:\WINDOWS\SYSTEM32\ntdll.dll
#3  0x00007ffe2dc6c62e in ntdll!EtwLogTraceEvent () from C:\WINDOWS\SYSTEM32\ntdll.dll
#4  0x00007ffe2dbebe44 in ntdll!RtlGetCurrentServiceSessionId () from C:\WINDOWS\SYSTEM32\ntdll.dll
#5  0x00007ffe2dbeab11 in ntdll!RtlFreeHeap () from C:\WINDOWS\SYSTEM32\ntdll.dll
#6  0x00007ffe2d7ccadc in msvcrt!free () from C:\WINDOWS\System32\msvcrt.dll
#7  0x00007ff6476eb19e in ?? ()
#8  0x00007ff6475f9de1 in ?? ()
#9  0x00007ff6476086ef in ?? ()
#10 0x00007ffdd6c4bb63 in QEventDispatcherWin32::qt_metacall(QMetaObject::Call, int, void**) ()
   from C:\Program Files\Subsurface\Qt5Core.dll
#11 0x00007ffdd6b12639 in QAbstractItemModel::dataChanged(QModelIndex const&, QModelIndex const&, QVector<int> const&) () from C:\Program Files\Subsurface\Qt5Core.dll
#12 0x00007ff64762cb4f in ?? ()
#13 0x00007ffdd6c4bb63 in QEventDispatcherWin32::qt_metacall(QMetaObject::Call, int, void**) ()
   from C:\Program Files\Subsurface\Qt5Core.dll
#14 0x00007ffdd26c08b2 in non-virtual thunk to QSpinBox::~QSpinBox() ()
   from C:\Program Files\Subsurface\Qt5Widgets.dll
#15 0x00007ffdd26bd02a in QAbstractSpinBox::initStyleOption(QStyleOptionSpinBox*) const ()
   from C:\Program Files\Subsurface\Qt5Widgets.dll
#16 0x00007ffdd26bdd82 in QAbstractSpinBox::stepBy(int) ()
   from C:\Program Files\Subsurface\Qt5Widgets.dll
#17 0x00007ffdd26bc4ee in QAbstractSpinBox::contextMenuEvent(QContextMenuEvent*) ()
```

I clicked on the up arrow of the spin box, which seems to be #16 -- but then it seems to destroy the QSpinBox (#14), run through a dataChanged handler and then access freed memory. But all this is happening inside Qt.

So I'm not convinced that this is really where the error is - vs this is where it manifests. Of course the debugging experience on Windows is a bit challenging with a cross built up. But let's see if @bstoeger looks at this and says "BUT OF COURSE!!!"

bstoeger, 2024-02-07:
I can't reproduce this on current master with Linux. And the stacktrace looks very ominous / incomplete. Is this a recent phenomenon or does it also happen on older versions?

BlackFire01, 2024-02-07:
Just tested as well, both issues only seem to concern Windows, no problems on Linux. Yeah, the crashes have been happening since I started using subsurface for deco planning ~6 months ago.

bstoeger, 2024-02-07:
Sadly, I don't have access to a debug environment under Windows. Could you try running the Linux version under valgrind and see if it does something ominous when you execute the steps above? For some reason valgrind stopped working on my laptop. :(

bstoeger, 2024-02-07:
OK... Made it run on a Linux Desktop and got this:
```
==72145== Invalid read of size 4
==72145== at 0x4F8FFB: setup_gas_sensor_pressure (profile.c:778)
==72145== by 0x4F8FFB: create_plot_info_new (profile.c:1342)
==72145== by 0x3E5DC4: ProfileScene::plotDive(dive const*, int, DivePlannerPointsModel*, bool, bool, bool, bool, double, double) (profilescene.cpp:456)
==72145== by 0x3F554E: ProfileWidget2::plotDive(dive const*, int, int) (profilewidget2.cpp:214)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x9972B4F: QAbstractItemModel::dataChanged(QModelIndex const&, QModelIndex const&, QVector const&) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x420497: DivePlannerPointsModel::emitDataChanged() (diveplannermodel.cpp:467)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x8B98A51: QSpinBox::valueChanged(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B95876: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B965E0: QAbstractSpinBox::stepBy(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B941E0: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8A2BB27: QWidget::event(QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== Address 0x13f706f8 is 0 bytes after a block of size 8 alloc'd
==72145== at 0x4845828: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==72145== by 0x4F8F21: setup_gas_sensor_pressure (profile.c:754)
==72145== by 0x4F8F21: create_plot_info_new (profile.c:1342)
==72145== by 0x3E5DC4: ProfileScene::plotDive(dive const*, int, DivePlannerPointsModel*, bool, bool, bool, bool, double, double) (profilescene.cpp:456)
==72145== by 0x3F554E: ProfileWidget2::plotDive(dive const*, int, int) (profilewidget2.cpp:214)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x9972B4F: QAbstractItemModel::dataChanged(QModelIndex const&, QModelIndex const&, QVector const&) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x420497: DivePlannerPointsModel::emitDataChanged() (diveplannermodel.cpp:467)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x8B98A51: QSpinBox::valueChanged(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B95876: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B965E0: QAbstractSpinBox::stepBy(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B941E0: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145==
==72145== Invalid write of size 4
==72145== at 0x4F8FFD: setup_gas_sensor_pressure (profile.c:777)
==72145== by 0x4F8FFD: create_plot_info_new (profile.c:1342)
==72145== by 0x3E5DC4: ProfileScene::plotDive(dive const*, int, DivePlannerPointsModel*, bool, bool, bool, bool, double, double) (profilescene.cpp:456)
==72145== by 0x3F554E: ProfileWidget2::plotDive(dive const*, int, int) (profilewidget2.cpp:214)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x9972B4F: QAbstractItemModel::dataChanged(QModelIndex const&, QModelIndex const&, QVector const&) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x420497: DivePlannerPointsModel::emitDataChanged() (diveplannermodel.cpp:467)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x8B98A51: QSpinBox::valueChanged(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B95876: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B965E0: QAbstractSpinBox::stepBy(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B941E0: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8A2BB27: QWidget::event(QEvent*) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== Address 0x13f912a8 is 0 bytes after a block of size 8 alloc'd
==72145== at 0x4845828: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==72145== by 0x4F8F39: setup_gas_sensor_pressure (profile.c:756)
==72145== by 0x4F8F39: create_plot_info_new (profile.c:1342)
==72145== by 0x3E5DC4: ProfileScene::plotDive(dive const*, int, DivePlannerPointsModel*, bool, bool, bool, bool, double, double) (profilescene.cpp:456)
==72145== by 0x3F554E: ProfileWidget2::plotDive(dive const*, int, int) (profilewidget2.cpp:214)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x9972B4F: QAbstractItemModel::dataChanged(QModelIndex const&, QModelIndex const&, QVector const&) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x420497: DivePlannerPointsModel::emitDataChanged() (diveplannermodel.cpp:467)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x8B98A51: QSpinBox::valueChanged(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B95876: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B965E0: QAbstractSpinBox::stepBy(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B941E0: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145==
==72145== Invalid write of size 4
==72145== at 0x4F9031: setup_gas_sensor_pressure (profile.c:784)
[... cut ...]
==72145== by 0x420497: DivePlannerPointsModel::emitDataChanged() (diveplannermodel.cpp:467)
==72145== by 0x9A15311: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.10)
==72145== by 0x8B98A51: QSpinBox::valueChanged(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B95876: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B965E0: QAbstractSpinBox::stepBy(int) (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145== by 0x8B941E0: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.10)
==72145==
```

bstoeger, 2024-02-07:
The problem is that the dive has 2 cylinders, but there is a gas change event for cylinder with index 2 (i.e. a third cylinder).

Can you check if the second crash is fixed with this:
```
diff --git a/core/profile.c b/core/profile.c
index 2a6e633f1..1806a9140 100644
--- a/core/profile.c
+++ b/core/profile.c
@@ -768,7 +768,7 @@ static void setup_gas_sensor_pressure(const struct dive *dive, const struct dive
 		int cyl = ev->gas.index;
 		int sec = ev->time.seconds;
 
-		if (cyl < 0)
+		if (cyl < 0 || cyl >= dive->cylinders.nr)
 			continue;
 
 		last[prev] = sec;
```

Note - this is _not_ a fix, just a test to narrow down the problem. I will only be able to check for the root cause of the problem tonight.

atdotde, 2024-02-07:
> On 7. Feb 2024, at 11:19, bstoeger ***@***.***> wrote:
> 
> The problem is that the dive has 2 cylinders, but there is a gas change event for cylinder with index 2 (i.e. a third cylinder).
> 

I see this happening when single stepping in a debugger. Somehow, the cylinder gets deleted before the corresponding gaschange event gets deleted.

Best
Robert

bstoeger, 2024-02-07:
Could it be this gem:
```
        /* FIXME: The planner uses a dummy cylinder one past the official number of cylinders
         * in the table to mark no-cylinder surface interavals. This is horrendous. Fix ASAP. */
```
? OMG.

bstoeger, 2024-02-07:
And there is more of that, for example in `calculate_sac`. :(

atdotde, 2024-02-07:
I guess, you need to change the initialization of the arrays in setut_gas_sensor_pressures as well.

bstoeger, 2024-02-07:
Yes. Last attempt, I have to leave now.
```
diff --git a/core/profile.c b/core/profile.c
index 2a6e633f1..97730818e 100644
--- a/core/profile.c
+++ b/core/profile.c
@@ -683,11 +683,13 @@ static void calculate_sac(const struct dive *dive, const struct divecomputer *dc
 	const struct event *ev = NULL;
 	bool *gases, *gases_scratch;
 
-	gases = calloc(pi->nr_cylinders, sizeof(*gases));
+	/* FIXME: The planner uses a dummy one-past-end cylinder for surface air! */
+	int num_cyl = pi->nr_cylinders + 1;
+	gases = calloc(num_cyl, sizeof(*gases));
 
 	/* This might be premature optimization, but let's allocate the gas array for
 	 * the fill_sac function only once an not once per sample */
-	gases_scratch = malloc(pi->nr_cylinders * sizeof(*gases));
+	gases_scratch = malloc(num_cyl * sizeof(*gases));
 
 	for (int i = 0; i < pi->nr; i++) {
 		struct plot_data *entry = pi->entry + i;
@@ -751,12 +753,14 @@ static void setup_gas_sensor_pressure(const struct dive *dive, const struct dive
 	if (pi->nr_cylinders == 0)
 		return;
 
-	int *seen = malloc(pi->nr_cylinders * sizeof(*seen));
-	int *first = malloc(pi->nr_cylinders * sizeof(*first));
-	int *last = malloc(pi->nr_cylinders * sizeof(*last));
+	/* FIXME: The planner uses a dummy one-past-end cylinder for surface air! */
+	int num_cyl = pi->nr_cylinders + 1;
+	int *seen = malloc(num_cyl * sizeof(*seen));
+	int *first = malloc(num_cyl * sizeof(*first));
+	int *last = malloc(num_cyl * sizeof(*last));
 	const struct divecomputer *secondary;
 
-	for (i = 0; i < pi->nr_cylinders; i++) {
+	for (i = 0; i < num_cyl; i++) {
 		seen[i] = 0;
 		first[i] = 0;
 		last[i] = INT_MAX;
@@ -768,8 +772,10 @@ static void setup_gas_sensor_pressure(const struct dive *dive, const struct dive
 		int cyl = ev->gas.index;
 		int sec = ev->time.seconds;
 
-		if (cyl < 0)
+		if (cyl < 0 || cyl >= num_cyl) {
+			fprintf(stderr, "setup_gas_sensor_pressure(): invalid cylinder idx %d\n", cyl);
 			continue;
+		}
 
 		last[prev] = sec;
 		prev = cyl;
```

dirkhh, 2024-02-07:
I think this last attempt looks sane. Well, somewhat sane, given the challenges with the underlying approach that the planner takes, but at least this should avoid crashes, right?
I guess the only way to test this is to merge this and create a test build?

bstoeger, 2024-02-07:
Note that this can only explain the second crash, as it only happens with a surface interval.

bstoeger, 2024-02-07:
> I think this last attempt looks sane. 

Looking a bit closer, the changes to `calculate_sac` are wrong. The loop should probably just exit once it reaches the surface air "cylinder".

bstoeger, 2024-02-07:
Or actually, the old code seem to be safe. The "surface gas mix" is treated gracefully in `get_gasmix_from_event`. Though I'm not sure if it is correct to use the surface interval in SAC calculation....?
@atdotde?

bstoeger, 2024-02-07:
> I guess the only way to test this is to merge this and create a test build?

Since you can reproduce this, you could try a local build, no?

dirkhh, 2024-02-07:
> > I guess the only way to test this is to merge this and create a test build?
> 
> Since you can reproduce this, you could try a local build, no?

I tried the latest build and the second crash is addressed.
And even better, the first issue with the plan disappearing ALSO seems to be addressed.

dirkhh, 2024-02-07:
@BlackFire01 would you please test the latest CICD build https://github.com/subsurface/nightly-builds/releases/download/v6.0.5093-CICD-release/subsurface-6.0.5093-CICD-release-installer.exe

BlackFire01, 2024-02-07:
I have tested the new build and both issues have been fixed! Thanks for the extremely fast and competent support, I will gladly continue using the app and recommend it to my tech group!

closed 2024-02-07 by BlackFire01