subsurface/subsurface #4096: release 6.0.5054 download links offline

core-code, 2024-02-13:
e.g.
https://subsurface-divelog.org/downloads/Subsurface-6.0.5054-CICD-release.dmg
=>
404

mikeller, 2024-02-13:
It looks like the binaries on the web server have been updated to 5082, but the links on the Current Release page are still pointing to 5054, @dirkhh can you please have a look?

@core-code you can download the binaries for the latest current release (5082) from https://subsurface-divelog.org/downloads/ for the time being.

dirkhh, 2024-02-13:
webserver stumbled over its feet and lost the updated data for the current release. Please try again.

core-code, 2024-02-13:
thanks, download works now.

since i am already typing, just FYI the meta-data doesn't line up. CFBundleVersion and CFBundleShortVersionString are stuck at 6.0.5082 while they should be 6.0.5094 - screenshot attached

[image]

closed 2024-02-13 by core-code

dirkhh, 2024-02-13:
I noticed that while testing the broken link.
Will fix and replace

dirkhh, 2024-02-13:
The file has been replaced. Please re-download. My apologies for the mixups.

core-code, 2024-02-13:
thanks, confirmed fixed

core-code, 2024-02-15:
@dirkhh on the topic of Mac release abnormalities, why do the disk-images contain the zipped app alongside the non-zipped app?

```ls /Volumes/Subsurface-6.0.5094 
Applications		Subsurface-6.0.5094.zip	Subsurface.app
```

if you make the window wider you see it:

[image]

dirkhh, 2024-02-15:
I figured out yesterday how to avoid that. So this won't be the case anymore starting with the next one.
Initially the stapling of the notarization failed otherwise.

mikeller, 2024-02-15:
> Initially the stapling of the notarization failed otherwise.

Need to use iStaples for Mac... 😝