subsurface/subsurface #4119: Crash when creating a journey or adding a dive to a journey

OlivierV78, 2024-03-04:
### Describe the issue:

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The mobile app crash when creating a journey or adding a dive to a journey

### Operating system:

Android

Galaxy S21

No

### Subsurface version:

6.0.5112-CICD

Nightly build

### Steps to reproduce:

Open subsurface
Create a new journey with long hold on a dive
OR add a dive to a existing journey

### Current behavior:

App crash

### Expected behavior:

App not crash

### Additional information:

Log in attachment but as I have to launch subsurface again after crash I don't know if the crash can be seen in the log

[subsurface.log](https://github.com/subsurface/subsurface/files/14479928/subsurface.log)

### Mentions:

mikeller, 2024-03-04:
Confirmed the behaviour in the android app. Doesn't happen on the desktop. Still need to figure out how to generate / access a log of the crash.

bstoeger, 2024-03-11:
Does it happen on mobile-on-desktop? There you can simply run it in a debugger.

mikeller, 2024-03-11:
I've never actually done a mobile-on-desktop build - looking into this now.

`adb logcat` when running on Android gives me the following:

```
03-11 21:47:12.222  4665  4785 I Subsurface: INFO: "17.230: showProgress: [Undo]Changes saved:'add 1 dives to trip [#1]'.\nUndo possible via context menu"
03-11 21:47:12.897  4665  4785 I Subsurface: INFO: Object 0x7a9ce06fc0 destroyed while one of its QML signal handlers is in progress.
03-11 21:47:12.897  4665  4785 I Subsurface: Most likely the object was deleted synchronously (use QObject::deleteLater() instead), or the application is running a nested event loop.
03-11 21:47:12.897  4665  4785 I Subsurface: This behavior is NOT supported!
03-11 21:47:12.897  4665  4785 I Subsurface: qrc:/org/kde/kirigami.2/private/ContextDrawerActionItem.qml:84: function() { [native code] }
```

mikeller, 2024-03-11:
No dice with mobile-on-desktop - adding dives to a trip or creating a trip works just fine.

mikeller, 2024-03-11:
Second time's the charm:

```
INFO: Object 0x55555856cdd0 destroyed while one of its QML signal handlers is in progress.
Most likely the object was deleted synchronously (use QObject::deleteLater() instead), or the application is running a nested event loop.
This behavior is NOT supported!
qrc:/org/kde/kirigami.2/private/ContextDrawerActionItem.qml:84: function() { [native code] }

Thread 1 "subsurface-mobi" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:44
Download failed: Invalid argument.  Continuing without source file ./nptl/./nptl/pthread_kill.c.
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6)
    at ./nptl/pthread_kill.c:89
#3  0x00007ffff53588e6 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
#4  0x00007ffff533c8b7 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff595e0d3 in qt_message_fatal
    (message=<synthetic pointer>..., context=<optimized out>)
    at global/qlogging.cpp:1914
#6  QMessageLogger::fatal
    (this=this@entry=0x7fffffff7b40, msg=msg@entry=0x7ffff6399f18 "Object %p destroyed while one of its QML signal handlers is in progress.\nMost likely the object was deleted synchronously (use QObject::deleteLater() instead), or the application is running a nested e"...) at global/qlogging.cpp:893
#7  0x00007ffff62793e8 in QQmlData::destroyed
    (this=<optimized out>, object=0x55555856cdd0) at qml/qqmlengine.cpp:1936
#8  0x00007ffff5bd0706 in QObject::~QObject
    (this=this@entry=0x55555856cdd0, __in_chrg=<optimized out>)
    at kernel/qobject.cpp:1019
#9  0x00007ffff6e6c4aa in QQuickItem::~QQuickItem
--Type <RET> for more, q to quit, c to continue without paging--c
    (this=this@entry=0x55555856cdd0, __in_chrg=<optimized out>)
    at items/qquickitem.cpp:2388
#10 0x00007ffff47566e7 in QQuickControl::~QQuickControl
    (this=this@entry=0x55555856cdd0, __in_chrg=<optimized out>)
    at qquickcontrol.cpp:1002
#11 0x00007ffff47373cf in QQuickAbstractButton::~QQuickAbstractButton
    (this=this@entry=0x55555856cdd0, __in_chrg=<optimized out>)
    at qquickabstractbutton.cpp:491
#12 0x00007fffccc84f21 in QQuickItemDelegate::~QQuickItemDelegate
    (this=0x55555856cdd0, __in_chrg=<optimized out>)
    at ../../../include/QtQuickTemplates2/5.15.10/QtQuickTemplates2/private/../../../../../src/quicktemplates2/qquickitemdelegate_p.h:57
#13 QQmlPrivate::QQmlElement<QQuickItemDelegate>::~QQmlElement
    (this=0x55555856cdd0, __in_chrg=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtQml/qqmlprivate.h:144
#14 QQmlPrivate::QQmlElement<QQuickItemDelegate>::~QQmlElement
    (this=0x55555856cdd0, __in_chrg=<optimized out>)
    at /usr/include/x86_64-linux-gnu/qt5/QtQml/qqmlprivate.h:144
#15 0x00007ffff5bc446e in QObjectPrivate::deleteChildren
    (this=this@entry=0x5555584ea650) at kernel/qobject.cpp:2137
#16 0x00007ffff5bd0a94 in QObject::~QObject
    (this=this@entry=0x5555584ea570, __in_chrg=<optimized out>)
    at kernel/qobject.cpp:1115
#17 0x00007ffff6e6c4aa in QQuickItem::~QQuickItem
    (this=this@entry=0x5555584ea570, __in_chrg=<optimized out>)
    at items/qquickitem.cpp:2388
#18 0x00007ffff6f69745 in QQuickImplicitSizeItem::~QQuickImplicitSizeItem
    (this=0x5555584ea570, __in_chrg=<optimized out>)
    at items/qquickimplicitsizeitem_p.h:60
#19 0x00007ffff6ff3481 in QQuickColumn::~QQuickColumn
    (this=0x5555584ea570, __in_chrg=<optimized out>)
    at ../../include/QtQuick/5.15.10/QtQuick/private/../../../../../src/quick/items/qquickpositioners_p.h:235
#20 QQmlPrivate::QQmlElement<QQuickColumn>::~QQmlElement
    (this=0x5555584ea570, __in_chrg=<optimized out>)
    at ../../include/QtQml/../../src/qml/qml/qqmlprivate.h:144
#21 QQmlPrivate::QQmlElement<QQuickColumn>::~QQmlElement
[... 187 lines cut ...]
#94 0x00007ffff3e76b2c in g_main_dispatch (context=0x7fffec000ec0)
    at ../../../glib/gmain.c:3476
#95 0x00007ffff3ed246f in g_main_context_dispatch_unlocked
    (context=0x7fffec000ec0) at ../../../glib/gmain.c:4284
#96 g_main_context_iterate_unlocked.isra.0
    (context=context@entry=0x7fffec000ec0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../../../glib/gmain.c:4349
#97 0x00007ffff3e74d20 in g_main_context_iteration
    (context=0x7fffec000ec0, may_block=1) at ../../../glib/gmain.c:4414
#98 0x00007ffff5bf4daa in QEventDispatcherGlib::processEvents
    (this=0x555555dadf40, flags=...) at kernel/qeventdispatcher_glib.cpp:423
#99 0x00007ffff5b9815b in QEventLoop::exec
    (this=this@entry=0x7fffffffd710, flags=..., flags@entry=...)
    at ../../include/QtCore/../../src/corelib/global/qflags.h:69
#100 0x00007ffff5ba0904 in QCoreApplication::exec ()
    at ../../include/QtCore/../../src/corelib/global/qflags.h:121
#101 0x00007ffff664b610 in QGuiApplication::exec ()
    at kernel/qguiapplication.cpp:1863
#102 0x00007ffff7400bf9 in QApplication::exec ()
    at kernel/qapplication.cpp:2832
#103 0x000055555568ac8c in run_mobile_ui
    (initial_font_size=initial_font_size@entry=11)
    at /home/mikeller/git/subsurface/subsurface-helper.cpp:189
#104 0x0000555555682d0d in main (argc=<optimized out>, argv=<optimized out>)
    at /home/mikeller/git/subsurface/subsurface-mobile-main.cpp:95
```

dirkhh, 2024-03-11:
Oh how I love modern programming. 100 layers of stack 🤷 
I think in level 38 we see that you are checking your cloud connection - which seems to have nothing to do with the situation described earlier, but whatever... checkServer in level 35 then causes the event loop to run - and that ends up deleting a QML object that is likely the parent of that cloud connection.

IOW, the UI has moved on and we are doing something in the background (like, writing things to the cloud), but the thing in the background is still connected to the UI element. Weird that this should show up now - did we make any significant changes there?

mikeller, 2024-03-12:
> Oh how I love modern programming. 100 layers of stack 🤷 I think in level 38 we see that you are checking your cloud connection - which seems to have nothing to do with the situation described [...]

Good point - running the event loop from inside an event handler is probably a very bad idea. There's another report of a similar problem that could be caused by this: https://groups.google.com/g/subsurface-divelog/c/z7X575UN4Qs/m/w2FpNDzGBgAJ

> IOW, the UI has moved on and we are doing something in the background (like, writing things to the cloud), but the thing in the background is still connected to the UI element. Weird that this should show up now - did we make any significant changes there?

Not that I am aware - but then it's about 70 layers of stack for the change to hide in - I'll have another look.

bstoeger, 2024-03-30:
Yes, this whole thing is super-sketchy. Probably it should only sync with the cloud from the main-loop. Is that possible?

I really wonder why this didn't happen before.

bstoeger, 2024-03-30:
Also happens with `v5.0.10` from Oct 2022: This is an old bug that happens with `autosync` on.

Frankly, I would just do the following:
```
diff --git a/mobile-widgets/qmlmanager.cpp b/mobile-widgets/qmlmanager.cpp
index c47bec59e..07a0d47d9 100644
--- a/mobile-widgets/qmlmanager.cpp
+++ b/mobile-widgets/qmlmanager.cpp
@@ -1441,19 +1441,8 @@ void QMLManager::addDiveToTrip(int id, int tripId)
 
 void QMLManager::changesNeedSaving(bool fromUndo)
 {
-	// we no longer save right away on iOS because file access is so slow; on the other hand,
-	// on Android the save as the user switches away doesn't seem to work... drat.
-	// as a compromise for now we save just to local storage on Android right away (that appears
-	// to be reasonably fast), but don't save at all (and only remember that we need to save things
-	// on iOS
-	// on all other platforms we just save the changes and be done with it
 	mark_divelist_changed(true);
 	emit syncStateChanged();
-#if defined(Q_OS_IOS)
-	saveChangesLocal(fromUndo);
-#else
-	saveChangesCloud(false, fromUndo);
-#endif
 	updateAllGlobalLists();
 }
```

And then add a timer that runs in the main event loop and every 10 sec or so does
```
	if (unsavedChanges()) {
		appendTextToLog("saving changes");
		saveChangesCloud(false);
		appendTextToLog("done trying to save to git local / remote");
	}
```
Or perhaps keep the iOS distinction to avoid random application hangs.

bstoeger, 2024-03-30:
Alternatively, remove the autosync option and only sync on state-change, application exit or manually by the user?

bstoeger, 2024-04-05:
Hopefully fixed with #4150. Closing for now.

closed 2024-04-05 by bstoeger