subsurface/subsurface #4127: SIGABRT when Saving to the Cloud

mikeller, 2024-03-10:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Subsurface crashes with a SIGABRT when saving to the cloud.

### Operating system:

Linux Ubuntu 23.10.

### Subsurface version:

Latest on `master`: `8b5812bc2c99fc603a59bf4d3fd6c776dc418db2`

### Steps to reproduce:

- start Subsurface
- Open cloud storage
- Save to cloud storage

### Current behavior:

- Subsurface crashes
- 
### Expected behavior:

- Subsurface stores data to the cloud

### Additional information:

Two different crashes can be observed, they seem to be dependent on the existing data in the cloud storage account (or no crash on an empty cloud account):

```
realloc(): invalid next size

Thread 1 "subsurface" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737252829312) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737252829312) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737252829312) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737252829312, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff54a3476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff54897f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff54ea676 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff563cb77 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff5501cfc in malloc_printerr (str=str@entry=0x7ffff563a83e "realloc(): invalid next size") at ./malloc/malloc.c:5664
#7  0x00007ffff5505c7c in _int_realloc (av=av@entry=0x7ffff567bc80 <main_arena>, oldp=oldp@entry=0x5555581d43a0, oldsize=oldsize@entry=48, nb=nb@entry=80) at ./malloc/malloc.c:4842
#8  0x00007ffff5506862 in __GI___libc_realloc (oldmem=0x5555581d43b0, bytes=64) at ./malloc/malloc.c:3494
#9  0x00007ffff7eca578 in ?? () from /lib/x86_64-linux-gnu/libgit2.so.1.1
#10 0x00007ffff7f49859 in ?? () from /lib/x86_64-linux-gnu/libgit2.so.1.1
#11 0x00007ffff7f4c0b4 in ?? () from /lib/x86_64-linux-gnu/libgit2.so.1.1
#12 0x00007ffff7f648c6 in git_treebuilder_insert () from /lib/x86_64-linux-gnu/libgit2.so.1.1
#13 0x0000555555924bbc in tree_insert (dir=0x555556001230, name=0x55557643c408 "2018", mkunique=<optimized out>, id=0x7fffffffce40, mode=GIT_FILEMODE_TREE)
    at /home/mikeller/git/subsurface/core/save-git.cpp:525
#14 0x0000555555924eee in write_git_tree (tree=tree@entry=0x7fffffffd000, result=result@entry=0x7fffffffcfa0, repo=<optimized out>) at /usr/include/c++/11/bits/basic_string.h:194
#15 0x0000555555926a88 in do_git_save (info=0x7fffffffd0c0, select_only=false, create_empty=<optimized out>) at /home/mikeller/git/subsurface/core/save-git.cpp:1330
#16 0x000055555592e5e8 in save_dives_logic (filename=0x555576218538 "https://ssrf-cloud-eu.subsurface-divelog.org/git/subsurface-test@ike.ch[subsurface-test@ike.ch]", select_only=<optimized out>,
    anonymize=<optimized out>) at /home/mikeller/git/subsurface/core/save-xml.cpp:825
#17 0x000055555579834a in MainWindow::on_actionCloudstoragesave_triggered (this=0x555555ffffb0) at /home/mikeller/git/subsurface/desktop-widgets/mainwindow.cpp:449
#18 0x00005555556ef763 in MainWindow::qt_metacall (this=0x555555ffffb0, _c=QMetaObject::InvokeMetaMethod, _id=8, _a=0x7fffffffd2d0)
    at /home/mikeller/git/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:371
#19 0x00007ffff5ce04e5 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff742bbe6 in QAction::triggered(bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff742e8fc in QAction::activate(QAction::ActionEvent) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff75b9142 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... 71 lines cut ...]
#38 0x00007ffff75c0d39 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#39 0x00007ffff74754ee in QWidget::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#40 0x00007ffff7432713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#41 0x00007ffff743a364 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#42 0x00007ffff5ca8e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#43 0x00007ffff7438e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#44 0x00007ffff748f843 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#45 0x00007ffff7491fd5 in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#46 0x00007ffff7432713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#47 0x00007ffff5ca8e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#48 0x00007ffff67ad307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#49 0x00007ffff6782a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#50 0x00007ffff0d5ad6e in ?? () from /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#51 0x00007ffff4222d3b in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#52 0x00007ffff4278258 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#53 0x00007ffff42203e3 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#54 0x00007ffff5d020b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#55 0x00007ffff5ca775b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#56 0x00007ffff5cafcf4 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#57 0x00005555556ddc43 in main (argc=<optimized out>, argv=<optimized out>) at /home/mikeller/git/subsurface/subsurface-desktop-main.cpp:110
```

### Mentions:

@bstoeger, @dirkhh

dirkhh, 2024-03-10:
So the first one is in git_treebuilder and looks VERY much like a memory corruption issue. That sucks as this could cause data loss.

And while the second one is actually in the progress update code, it also immediately makes me think that we walked all over the memory management code.

So I'm wondering if the re-implementation of the leaky memory management in C ended up causing things to boom.

I'll try to reproduce this here to see if I can make more sense of either of these crashes.

One of the neat things to try would be to compile Subsurface with ASAN enabled - that often does a good job at pin-pointing stuff like this.

mikeller, 2024-03-11:
```
[Thread 0x7fffca1b1640 (LWP 46934) exited]
=================================================================
==46862==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x604002bc28f1 at pc 0x555555df1eeb bp 0x7fffffffbc10 sp 0x7fffffffbc00
WRITE of size 1 at 0x604002bc28f1 thread T0
    #0 0x555555df1eea in fp_get_data[abi:cxx11](fingerprint_table*, unsigned int) /home/mikeller/git/subsurface/core/device.cpp:302
    #1 0x555555f683f4 in save_one_fingerprint /home/mikeller/git/subsurface/core/save-git.cpp:889
    #2 0x555555f683f4 in save_settings /home/mikeller/git/subsurface/core/save-git.cpp:901
    #3 0x555555f683f4 in create_git_tree /home/mikeller/git/subsurface/core/save-git.cpp:1026
    #4 0x555555f683f4 in do_git_save /home/mikeller/git/subsurface/core/save-git.cpp:1324
    #5 0x555555f789f2 in save_dives_logic /home/mikeller/git/subsurface/core/save-xml.cpp:825
    #6 0x555555a214a6 in MainWindow::on_actionCloudstoragesave_triggered() /home/mikeller/git/subsurface/desktop-widgets/mainwindow.cpp:449
    #7 0x55555580b123 in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) /home/mikeller/git/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:284
    #8 0x555555812c02 in MainWindow::qt_metacall(QMetaObject::Call, int, void**) /home/mikeller/git/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:371
    #9 0x7ffff4f8e4e4  (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2f14e4)
    #10 0x7ffff6766be5 in QAction::triggered(bool) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x165be5)
    #11 0x7ffff67698fb in QAction::activate(QAction::ActionEvent) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1688fb)
    #12 0x7ffff68f4141  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x2f3141)
    #13 0x7ffff68fbd38  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x2fad38)
    #14 0x7ffff67b04ed in QWidget::event(QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1af4ed)
    #15 0x7ffff676d712 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16c712)
    #16 0x7ffff6775363 in QApplication::notify(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x174363)
    #17 0x7ffff4f56e39 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2b9e39)
    #18 0x7ffff6773e46 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x172e46)
    #19 0x7ffff67ca842  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1c9842)
    #20 0x7ffff67ccfd4  (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1cbfd4)
    #21 0x7ffff676d712 in QApplicationPrivate::notify_helper(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16c712)
    #22 0x7ffff4f56e39 in QCoreApplication::notifyInternal2(QObject*, QEvent*) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2b9e39)
    #23 0x7ffff5a5d306 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x141306)
    #24 0x7ffff5a32a2b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x116a2b)
    #25 0x7fffecdf6d6d  (/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5+0x73d6d)
    #26 0x7ffff35abd3a in g_main_context_dispatch (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x55d3a)
    #27 0x7ffff3601257  (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0xab257)
    #28 0x7ffff35a93e2 in g_main_context_iteration (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x533e2)
    #29 0x7ffff4fb00b7 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x3130b7)
    #30 0x7ffff4f5575a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2b875a)
    #31 0x7ffff4f5dcf3 in QCoreApplication::exec() (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2c0cf3)
    #32 0x5555557dada4 in main /home/mikeller/git/subsurface/subsurface-desktop-main.cpp:110
    #33 0x7ffff476ad8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #34 0x7ffff476ae3f in __libc_start_main_impl ../csu/libc-start.c:392
    #35 0x5555557f8b14 in _start (/home/mikeller/git/install-root/bin/subsurface+0x2a4b14)

0x604002bc28f1 is located 0 bytes to the right of 33-byte region [0x604002bc28d0,0x604002bc28f1)
allocated by thread T0 here:
    #0 0x7ffff767c1e7 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x7ffff4bbcd0b in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct(unsigned long, char) (/lib/x86_64-linux-gnu/libstdc++.so.6+0x14bd0b)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/mikeller/git/subsurface/core/device.cpp:302 in fp_get_data[abi:cxx11](fingerprint_table*, unsigned int)
Shadow bytes around the buggy address:
  0x0c08805704c0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fd
  0x0c08805704d0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fd
  0x0c08805704e0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fd
  0x0c08805704f0: fa fa fd fd fd fd fd fd fa fa fd fd fd fd fd fa
  0x0c0880570500: fa fa fd fd fd fd fd fa fa fa fd fd fd fd fd fa
=>0x0c0880570510: fa fa fd fd fd fd fd fa fa fa 00 00 00 00[01]fa
  0x0c0880570520: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
[... cut ...]
[Thread 0x7fffe0ab3640 (LWP 46877) exited]
[Thread 0x7fffe12b4640 (LWP 46876) exited]
[Thread 0x7fffe1ab5640 (LWP 46875) exited]
[Thread 0x7fffe22b6640 (LWP 46874) exited]
[Thread 0x7fffe32b8640 (LWP 46872) exited]
[Thread 0x7fffecd82640 (LWP 46869) exited]
[Thread 0x7fffed6fe640 (LWP 46868) exited]
[Thread 0x7fffedeff640 (LWP 46867) exited]
[Thread 0x7ffff12ed0c0 (LWP 46862) exited]
[Thread 0x7fffe2ab7640 (LWP 46873) exited]
[New process 46862]
[Inferior 1 (process 46862) exited with code 01]
```

bstoeger, 2024-03-11:
Well, yes. These were bound to happen as I noted in the PR description. And I'm sure there will be more.

This one is sadly a consequence of the weak type-system: chars are accepted as integers and vice-versa. I would have appreciated a narrowing-conversion warning from the compiler.

commit 311c04c8 (2024-03-11): core: don't do C-style string initialization in core/device.cpp

The std::string was wrongly initialized (character and length
were exchanged) and then filled with C-style array accesses.

Instead, reserve the memory and using string concetanation.
This will not break, even if we get the length wrong. In the
worst case, it will cause reallocation.

Fixes #4127.
commit c7077653 (2024-03-11): core: don't do C-style string initialization in core/device.cpp

The std::string was wrongly initialized (character and length
were exchanged) and then filled with C-style array accesses.

Instead, reserve the memory and use string concetanation.
This will not break, even if we get the length wrong. In the
worst case, it will cause reallocation.

Fixes #4127.
PR #4129 (2024-03-11): Correctly initialize string in core/device.cpp
When initializing a string with multiple characters, first comes the length, then the size. Not the other way around.

Fixes #4127.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

See commit description.

### Related issues:

Fixes #4127.
commit c7a929a8 (2024-03-11): Correctly initialize string in core/device.cpp

When initializing a string with multiple characters, first
comes the length, then the size. Not the other way around.

Fixes #4127.