subsurface/subsurface #4138: Import from Mares Quad fails

nelsg, 2024-03-16:
### Describe the issue:

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I trying to import from my computer (Mares Quad) and it fails with this error:

[image]

Logs are:

````
Subsurface: v6.0.5112.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (62a29eea15137ba7b0f2e10fae095517cc9d8341)
[303.955073] INFO: Open: name=/dev/tty.Bluetooth-Incoming-Port
[303.956196] INFO: Configure: baudrate=115200, databits=8, parity=2, stopbits=0, flowcontrol=0
[303.956562] INFO: Timeout: value=3000
[303.956564] INFO: DTR: value=0
[303.956680] INFO: RTS: value=0
[303.956812] INFO: Purge: direction=3
[303.957009] INFO: Write: size=2, data=C267
[306.960954] INFO: Read: size=0, data=
[306.960958] ERROR: Failed to receive the answer. [in src/mares_iconhd.c:213 (mares_iconhd_packet)]
[306.960960] INFO: Sleep: value=100
[307.061606] INFO: Purge: direction=1
[307.061945] INFO: Write: size=2, data=C267
[310.064959] INFO: Read: size=0, data=
[310.064963] ERROR: Failed to receive the answer. [in src/mares_iconhd.c:213 (mares_iconhd_packet)]
[310.064966] INFO: Sleep: value=100
[310.165371] INFO: Purge: direction=1
[310.165574] INFO: Write: size=2, data=C267
[313.165603] INFO: Read: size=0, data=
[313.165606] ERROR: Failed to receive the answer. [in src/mares_iconhd.c:213 (mares_iconhd_packet)]
[313.165608] INFO: Sleep: value=100
[313.265735] INFO: Purge: direction=1
[313.265916] INFO: Write: size=2, data=C267
[316.270971] INFO: Read: size=0, data=
[316.270975] ERROR: Failed to receive the answer. [in src/mares_iconhd.c:213 (mares_iconhd_packet)]
[316.270978] INFO: Sleep: value=100
[316.375195] INFO: Purge: direction=1
[316.375506] INFO: Write: size=2, data=C267
[319.377347] INFO: Read: size=0, data=
[319.377357] ERROR: Failed to receive the answer. [in src/mares_iconhd.c:213 (mares_iconhd_packet)]
````

### Operating system:

Mac Book Pro

[image]

### Subsurface version:

SubSurface 6.0.5112-CICD-release

### Steps to reproduce:

### Current behavior:

Import fails

### Expected behavior:

Import succeed

### Additional information:

### Mentions:

jefdriesen, 2024-07-17:
The log indicates you tried to use `/dev/tty.Bluetooth-Incoming-Port`. That is not the correct serial port.

mikeller, 2026-10-17:
Hi @nelsg,

Thanks for reporting the import failure with your Mares Quad. Could you confirm the exact COM port (Windows) or BLE device you selected when attempting the download? This often turns out to be a port-selection mismatch rather than a genuine driver fault, so confirming that first will help us narrow down whether there's a real bug here.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.