subsurface/subsurface #4140: Add export option "CSV Dive sites sorted by Country" and change geo nodes in dive sites XML

hwillemen, 2024-03-19:
### Describe the issue:

- [ ] Bug
- [X] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:

Because the way the Dive sites xml is structured it is hard to parse in external programs.
As an example: there is no schema that can import the xml into Excel because of the non-unique "geo" tag.

Solution 1: CHANGE xml export:
Since every geo cat has a specific meaning it would be more readable and better parse-able to make them all unique:
  <geo cat='1' -> <Ocean value='Foo'/>
  <geo cat='2' -> <Country value='Foo'/>
  <geo cat='3' -> <State value='Foo'/>
  <geo cat='4' -> <County value='Foo'/>
  <geo cat='5' -> <Town value='Foo'/>
  <geo cat='6' -> <City value='Foo'/>
(For clarity the format geo1ocean might even be better)
It would also help to give the option to sort the dive sites by Country and then by site name.
This makes more sense than sorting by uuid.

Solution 2: ADD export option for sorted CSV:
Properties like; uuid, name, gps, description, Ocean, Country, State, County, Town, City, Notes
can easily be columns for an CSV.
Again, this should be sorted by Country and then by name, this makes the most sense I think.
"CSV Dive sites by Country/name" would be descriptive.

### Operating system:

Android & Windows: not relevant

### Subsurface version:

6.0.5112-CICD-release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

hwillemen, 2024-04-04:
I have solved this problem myself for now with a python script:
https://github.com/hwillemen/subsurface-divelogs-scripts/blob/main/Divesites_sorter.py

Best would be to simply support this by the program itself.

poehoes, 2024-04-19:
Nice, perhaps you could add a small non-sorted subsurface.xml and a sorted .csv to your repository so people can test this (and compare the result with the csv).

hwillemen, 2024-05-13:
I did, as well with a readme and some other info.

mikeller, 2026-10-18:
Hi @hwillemen,

Splitting the two asks here:

Restructuring the dive-site XML's geo elements into named fields: not something we're going to do, it would be a breaking format change affecting backward compatibility with existing dive-site XML/cloud sync and third-party tooling (including the scuba-sites.com integration), for what's primarily a parsing-convenience improvement. Your own Python script already solves that parsing pain point externally without touching the canonical format, which seems like the better path for this specific need.

A dedicated "CSV Dive sites sorted by Country" export option: lower-risk and more plausible as an additive feature, and your script is a solid field-mapping reference if anyone picks it up, but with only light engagement on this so far, it's not something we're actioning right now. We'll leave the door open if there's renewed interest.

Closing this issue on that basis. Please reopen if there's broader interest in the CSV export option specifically.

closed 2026-10-18 by mikeller