subsurface/subsurface #4147: Question: Cressi: Bluetooth Android via Homemade ESP32 BT-Serial to IRDA bridge.

hb9eue, 2024-03-28:
Hi Devs

Sorry for being to lazy to dig into the depths of the source.

I own a Cressi Leonardo and recently got Cressi Donatello for another family member.

The Cressi Leonardo uses serial IRDA @115200 baud for communication with it's vendor specific, very expensive RS232 cradle.
An ESP32 with a TFDU4101 can also do IRDA for a fraction of that price.

I tinkered around a bit, and after commenting out DTR/RTS toggles, which are required to initialize the vendor craddle, but which of course cause the USB-Serial interface of an ESP32 bord to reboot, I got import for the Leonardo working!

The newly purchased Donatello is advertised to communicate via Bluetooth-Serial! Yeah!

What a disappointment, when I realized you once more have to buy a very expensive additional bluetooth interface. But wait, the Donatello has some hole at the exact same location on which the Leonardo has it's IRDA interface. Is that Bluetooth vendor interface a simple IRDA to Bluetooth-Serial interface?

Let's comment out the DTR/RTS toggles in libdivecomputer for the goa/donatello model.

Success! I immediately can connect to the donatello and get's it's firmware version and modell information. (no dives yet).

And ESP32 has bluetooth too. So why not try to reconstruct that vendor bluetooth to IRDA interface with an ESP32 and then use Bluetooth-Serial to import the dives. In theory, that should also work from the subsurface anrdoid app.

Unfortunately I am stuck with the android app. I am able to pair Android to my ESP32 with the Bluetooth-Serial service. And I am able to select the bluetooth device in subsurface android.

But it looks like android is never properly opening the serial connection when I select 'import'. On the USB serial monitor on the ESP32 which I now can use to log some debugging information, I never get any data via bluetooth serial from the android app.

When I select the Bluetooth Device in subsurface andoid, vendor and type of the divecomputer also are being preset to something that is probably deducted from the BT MAC Address of my ESP32.

So that makes me wonder. Does the android app somehow check the MAC address of the BT device somehow defaults to only speak the BT protocol associated with that MAC address?

-Benoit-

KonradIT, 2024-07-20:
Hi @hb9eue 

I'm interested in your work, how did you wire the TFDU4101 to the ESP32? Breakout board? How did you encode whatever is received on the ESP32 from the IR sensor so Subsurface could read it?

hb9eue, 2024-08-16:
Sorry for the late reply.

You can switch the uart port of the esp32 to IRDA mode. So no need for any further encoding. The only real issue I faced is that the TFDU4101 is receiving the echo of the character being sent on the RX side. So I drop the character if it's the same I just sent. This would most probably not work in a real dupliex communication. Thankfully the cressi divecomputer receive the command to dump their memory and then send the dump, so it's 'half duplex' which works.

I wired the TFDU4101 as per data-sheet with a resistor (don't know remember much) in front of the RX led to limit current.

Here is my arduino code for the Heltec ESP32 board:

```
#include <BTAddress.h>
#include <BTAdvertisedDevice.h>
#include <BTScan.h>
#include <BluetoothSerial.h>

#define BUILTIN_LED 2

HardwareSerial irda(2); // GPIO 17: TXD U2  +  GPIO 16: RXD U2
// NOT the ones labeled TX and RX, real GPIO 17 and 16 on Heltec
int serialByte;
int irdaByte;
bool sent;
bool usbmode = false;
String device_name = "DiveComputerBT";

// Check if Bluetooth is available
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
  #error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

// Check Serial Port Profile
#if !defined(CONFIG_BT_SPP_ENABLED)
  #error Serial Port Profile for Bluetooth is not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);
    Serial.begin (115200); // (USB + TX/RX) to check
    irda.begin(115200); // GPIO 17: TXD U2  +  GPIO 16: RXD U2

    //UART_CONF0_REG  Configuration register 0
    //UART0: 0x3FF40020
    //UART1: 0x3FF50020
    //UART2: 0x3FF6E020

    WRITE_PERI_REG( 0x3FF6E020 , READ_PERI_REG(0x3FF6E020) | (1<<16 ) | (1<<9 ) );  //UART_IRDA_EN + UART_IRDA_DPLX  "Let there be light"
    //Serial.print("Reg: "); Serial.println(READ_PERI_REG(0x3FF6E020),BIN);  //For Debug only
    //Serial.println("Ready");
    digitalWrite(BUILTIN_LED, HIGH);
    delay(100);
    digitalWrite(BUILTIN_LED, LOW);
//    irda.print("{123DBA}");
  SerialBT.begin(device_name); //Bluetooth device name
  // SerialBT.deleteAllBondedDevices(); // Uncomment this to delete paired devices; Must be called after begin
  Serial.printf("The device with name \"%s\" is started.\nNow you can pair it with Bluetooth!\n", device_name.c_str());
}//setup

void loop() {
  if (usbmode) {
    while (Serial.available()) {
      serialByte = Serial.read();
      irda.print((char) serialByte);
      sent = true;
      delay(5); // Obtained by try and error wait until a potential echo is received
      break; // Immediately check if we received and echo
    }
  } else {
    while (SerialBT.available()) {
      serialByte = SerialBT.read();
      irda.print((char) serialByte);
      Serial.print((char) serialByte);
      sent = true;
      delay(5); // Obtained by try and error wait until a potential echo is received
      break; // Immediately check if we received and echo
    }
  }
  while (irda.available()) {
    irdaByte = irda.read();
    if ((irdaByte == serialByte) && sent) {
    //if (irdaByte == serialByte) {
      //Serial.print("Echo:");
      //Serial.println((char) irdaByte);
    } else {
      //Serial.print("Received:");
      //Serial.println((char) irdaByte);
      if (usbmode) {
        Serial.print((char) irdaByte);
      } else {
        SerialBT.print((char) irdaByte);
        Serial.print((char) irdaByte);
      }
    }
    sent = false;
  }
}
```

sebdbr, 2024-11-01:
Hi @hb9eue !
Since the Cressi BT interface for Leonardo is no longer available, I'm also interested in your work, and i'm trying to build one for my son, who would like to record his dives on his PC.
As a newbie, I need more information :
I guess that I have to build subsurface from sources AFTER modifying the cressi_leonardo.c file in subfolder subsurface/libdivecomputer/src/...
Am I right ?
But which lines should I comment out exactly ?
From line 199 to 212 ?

````
// Set the RTS line.
	status = dc_iostream_set_rts (device->iostream, 1);
	if (status != DC_STATUS_SUCCESS) {
		ERROR (context, "Failed to set the RTS line.");
		goto error_free;
	}

	// Set the DTR line.
	status = dc_iostream_set_dtr (device->iostream, 1);
	if (status != DC_STATUS_SUCCESS) {
		ERROR (context, "Failed to set the DTR line.");
		goto error_free;
	}

	dc_iostream_sleep (device->iostream, 200);
````

Hope that you could help me !

On the other hand, have you finally found a way to make it work with the android app ?

Seb

hb9eue, 2024-11-01:
Hi Seb

It depends on how you connect the ESP32 with your computer. If you use Bluetooth serial, subsurface does not need any modification.

Only if you connect the ESP32 via USB Serial, you need to modify the source as you have to remove the part which toggles the DTR line to initialize the original craddle as this causes a reset of the ESP32.

To disable this, in ./libdivecomputer/src/cressi_leonardo.c

function: cressi_leonardo_device_open

Just comment out this whole block as I did here:
[...]
```
        // Set the timeout for receiving data (1000 ms).
        status = dc_iostream_set_timeout (device->iostream, 1000);
        if (status != DC_STATUS_SUCCESS) {
                ERROR (context, "Failed to set the timeout.");
                goto error_free;
        }

/*
        // Set the RTS line.
        status = dc_iostream_set_rts (device->iostream, 1);
        if (status != DC_STATUS_SUCCESS) {
                ERROR (context, "Failed to set the RTS line.");
                goto error_free;
        }

        // Set the DTR line.
        status = dc_iostream_set_dtr (device->iostream, 1);
        if (status != DC_STATUS_SUCCESS) {
                ERROR (context, "Failed to set the DTR line.");
                goto error_free;
        }

        dc_iostream_sleep (device->iostream, 200);

        // Clear the DTR line.
        status = dc_iostream_set_dtr (device->iostream, 0);
        if (status != DC_STATUS_SUCCESS) {
                ERROR (context, "Failed to clear the DTR line.");
                goto error_free;
        }
*/
        dc_iostream_sleep (device->iostream, 1000);
        dc_iostream_purge (device->iostream, DC_DIRECTION_ALL);

```
[...]

And no, unfortunately still no success with the Android app. I guess the app check the Bluetooth MAC address to match a device and as at least my ESP32 seems to match some other device type, it does not attempt to connect using the cresi leonardo or donatello protocol.

-Benoît-

sebdbr, 2024-11-16:
Hello @hb9eue,
First of all, thanks for your tip.
After some struggles to adapt your DIY interface for the NodeMCU esp32 (on which the UART2 was out of order...) and tfdu4800  (not exactly the same pinout than the 4801) that i had, it seems to me I  managed to get an answer from the Leonardo on USB serial port, but there are still some problems.
Here are some examples of the libdivecomputer log in Subsurface.

> Subsurface: v6.0.5282.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (a2a5bb53d670dc520bf8035b18681dfdd9b18bfa)
[3175,516999] INFO: Open: name=/dev/ttyUSB0
[3175,525601] INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
[3175,531581] INFO: Timeout: value=1000
[3175,531599] INFO: Sleep: value=100
[3175,631686] INFO: Purge: direction=3
[3175,632444] INFO: Write: size=8, data=7B3132334442417D
[3175,670707] INFO: Read: size=7, data=C000380000F8C0
[3175,670720] ERROR: Unexpected answer byte. [in src/cressi_leonardo.c:326 (cressi_leonardo_device_dump)]

> Subsurface: v6.0.5282.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (a2a5bb53d670dc520bf8035b18681dfdd9b18bfa)
[9503,232466] INFO: Open: name=/dev/ttyUSB0
[9503,241359] INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
[9503,241388] INFO: Timeout: value=1000
[9503,241394] INFO: Sleep: value=100
[9503,342144] INFO: Purge: direction=3
[9503,342750] INFO: Write: size=8, data=7B3132334442417D
[9503,380888] INFO: Read: size=7, data=80000000008080
[9503,380900] ERROR: Unexpected answer byte. [in src/cressi_leonardo.c:326 (cressi_leonardo_device_dump)]

If you have any ideas...

-Seb-

mikeller, 2026-10-17:
Hi @hb9eue,

Really interesting DIY project, thanks for the write-up. The Android app's Bluetooth detection relies on recognising a paired device's advertised identity (MAC address/name pattern) as belonging to a known vendor/model, by design, so a homemade ESP32 bridge that doesn't present as a genuine Cressi device won't be picked up, even though it works fine on desktop once DTR/RTS toggling is patched out.

Closing this as expected behaviour for a DIY device outside the supported detection list, rather than a bug. A "manually force vendor/model" override is a reasonable feature idea if more people hit this, but we haven't seen that demand yet. Please reopen (or file a feature request) if that changes.

closed 2026-10-17 by mikeller