subsurface/subsurface #4153: Can't set more than one buddy / dive guide in dive notes

dgouju, 2024-04-01:
### Describe the issue:
- [ X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I can't set more than one buddy in dive notes. Same with the dive guide (sometimes you may have more than one for various reasons)

### Operating system:
Linux & iOS

### Subsurface version:
Linux: 6.0.5112-CICD
iOS: 6.0.5124
Note: I know that in older releases (I mean in some 5.X) it was possible because I have old dives with 2 buddies & 2 dive guides

### Steps to reproduce:
1°) Import a dive
2°) In Notes, you can add only 1 Buddy and 1 Dive guide

### Expected behavior:
Being able to add more than one Buddy (up to 4 at least) and more than one Dive guide (I would say at least 2)

mikeller, 2024-04-01:
Seems to be working fine for me here on linux:

[image]

Can you please add a detailed list of steps of how you are entering the list of comma separated buddies / dive guides that shows where this is failing for you?

dgouju, 2024-04-02:
Thanks @mikeller , I think it was then more a UX point. I tried to:
- click on the right side of the red box, expecting creating a new green box
- press "return" to create like a new tag in some other tools

But obviously if I add a comma, it works ;-) 

This can be either closed or be moved as a UX feature request?

Thanks!

dgouju, 2024-04-05:
Let's close ;-)

closed 2024-04-05 by dgouju