subsurface/subsurface #4154: iOS app cloud issues

lhangart, 2024-04-08:
### Describe the issue:

- [ ] Bug

### Issue long description:
Current iOS app failed to sync data with cloud. Deleted app and app data and reinstalled app. Now, I cannot log in any longer. Same email/password combination that works on the MacOS desktop app raises 'incorrect email / password combination' error. Resetting password did not fix the issue.

### Operating system:
iOS 17.4.1

### Subsurface version:
6.0.5.5124

### Steps to reproduce:
Deleted app and app data and reinstalled app, reset password

dirkhh, 2024-04-08:
please try to log into the cloud and reproduce the failure. Then switch to no-cloud mode (in order to get to the main menu) and send a support request from within the app. That way I'll get relevant logs that might help understand what's going wrong.

The most common issue are incorrectly spelled email addresses (I had people yell at me for doubting that they knew their own email address, only to later say "you know what, never mind, I fixed it) and subtly different behavior with special characters in the password -- I highly recommend to just use ascii letters and numbers.

closed 2024-04-09 by lhangart

dirkhh, 2024-04-09:
I'm glad it worked out.
The only log that would have been interesting would have been a log of a failure. Now that it works, there won't be anything interesting in there.

You can simply create your own tank that matches what you need. Please do so in the desktop app and save at least one dive using that tank. After you sync this with the phone, that tank will be available on the phone in the dropdown as well.

Copying default equipment / people is easily possible in the desktop version. Trying to do this in the mobile version is beyond the skills of the current set of developers, I think.

lhangart, 2024-04-10:
Hi Dirk,

Thanks for your explanations. I managed to make my own tank and it synced to the phone and also discovered the ‘copy dive components’ feature 🙈

Best,

Lars

lhangart, 2024-04-12:
Hi Dirk,

The synch issue occurred again, but only when my iPhone was connected to the local WiFi network. After I switched to 3G, it synched again.. I submitted the logs from within the app.

Best,

Lars

mikeller, 2024-04-12:
Hi Lars.

> The synch issue occurred again, but only when my iPhone was connected to the local WiFi network. After I switched to 3G, it synched again.. I submitted the logs from within the app.

What kind of WiFi network are you using? Residential broadband with access point / public WiFi / corporate WiFi?
From your description it looks like the problem could be overly aggressive firewalling / misconfiguration / general access limitations between the WiFi network and the internet.

lhangart, 2024-04-12:
Hi Dirk,

It is a corporate/edu network (The Scripps Research Institute). Also, our IT is currently redesigning our WiFi architecture to improve security, so it’s quite possible that something was blocked by this network. What port are you using for the sync?

L

Subsurface-divelog, 2024-04-12:
It's https. My guess is that they are using a man in the middle reverse proxy and that Subsurface says "no thank you" to that 

Generally I recommend not to use any corporate resources for non corporate purposes. Separate devices, separate infrastructure. 

IT organizations are in the business of assuming that they own everything and control everything and that anything they don't know is by definition evil and needs to be taken over. I generally decline to allow IT controlled devices onto my personal networks, nor do I connect personal devices to corporate networks. 

(and yes, ironically I work in a gigantic IT organization of some 20k people) 

/D