subsurface/subsurface #4155: Suunto Vyper Import suddenly broken in newest windows version

matthiashoeller, 2024-04-12:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I have been using the "Import from Dive computer" function with my Suunto Vyper and also my wife's Suunto Vyper for a long time without problems.
This is also still working flawlessly with windows build 6.0.5094.
However, after updating to 6.0.5112, the function no longer works. When I try to import the latest dives from my Suunto Vyper, then subsurface no longer recognizes the dive computer's serial number (it is displayed as "0"), and subsequently gets confused when I try downloading the dives from both Suunto Vypers, as it can no longer separate the two dive computers.
After downgrading back to 6.0.5094, everything is back to normal again, the import function successfully recognizes both dive computer's serial numbers, and everything works as usual. 
(I did not try with 6.0.5101)

### Operating system:

Windows 11 Pro

Dell XPS15 9510

### Subsurface version:

6.0.5112 (this version has the bug)

6.0.5094 (this version does not have the bug)

official release

### Steps to reproduce:

1) Connect Suunto Vyper Dive Computer
2) Set Suunto Vyper Dive Computer to "TR-PC"
3) Select Import from dive computer, hit download
4) Optional: repeat with second Suunto Vyper

### Current behavior:

1) After downloading latest dives from Suunto Vyper #1, the serial number in the "additional info" tab is wrongly displayed as "0"
2) Subsequently, when trying to download latest dives from Suunto Vyper #2, subsurface fails to recognize which dives are new and downloads only 1 dive instead of all new ones, and also does not map the dives from the two dive computers correctly

### Expected behavior:

1) After downloading latest dives from Suunto Vyper #1, the serial number in the "additional info" tab is correctly displayed
2) Subsequently, when trying to download latest dives from Suunto Vyper #2, subsurface recognizes which ones are new, downloads all new ones, and maps the dives from the two dive computers correctly.
The "expected behaviour" 1) and 2) is precisely what happens in 6.0.5094 and works flawlessly there.

### Additional information:

At the moment, I do not have them. 
If they are required, feel free to contact me and I will try to provide them, if that helps (matthias.hoeller@rwth-aachen.de)

### Mentions:

bstoeger, 2024-04-13:
This is interesting. I touched quite a lot of code concerning string-handling recently. I see how that could break parsing of the serial number.

Is there a way to get the git-tags of `6.0.5112` and `6.0.5094`?

mikeller, 2024-04-13:
> Is there a way to get the git-tags of `6.0.5112` and `6.0.5094`?

We aren't tagging the nightly releases any more in the `subsurface` repository, but the changeset IDs are a4091189b014566dd40d3bb35c22c620e041d1e4 and 5250a86277102567666d39cf99306f7a78700c16 respectively.

I'll be adding a script to find changeset IDs based on nightly release number.

matthiashoeller, 2024-04-13:
> We aren't tagging the nightly releases any more in the `subsurface` repository

Small remark - the releases in question (6.0.5112 and 6.0.5094) aren't "nightly" releases. These are (were) official windows releases.

mikeller, 2024-04-13:
> Small remark - the releases in question (6.0.5112 and 6.0.5094) aren't "nightly" releases. These are (were) official windows releases.

Same difference really - the 'current' releases are just a subset of the 'latest' / nightly releases that are selected for not having any regressions reported against them. The builds and the version numbers are the same as the 'latest' builds they are based on, to avoid confusion.
Also, the 'latest' and the 'current' builds are both 'official' - just with a slightly different expectation re. stability.

matthiashoeller, 2024-04-13:
> Same difference really [...]

Thank you for clarification!

bstoeger, 2024-04-15:
> We aren't tagging the nightly releases any more in the `subsurface` repository, but the changeset IDs are [a409118](https://github.com/subsurface/subsurface/commit/a4091189b014566dd40d3bb35c22c620e041d1e4) and [5250a86](https://github.com/subsurface/subsurface/commit/5250a86277102567666d39cf99306f7a78700c16) respectively.

The only thing related to import I see between these to IDs is a `libdivecomputer` update. Could that be the culprit?

matthiashoeller, 2024-05-02:
> The only thing related to import I see between these to IDs is a `libdivecomputer` update. Could that be the culprit?

Would that be from 0.7.0 to 0.8.0? Nothing related is mentioned in their release notes..

Is there anything else I can contribute? Would it help if I tried to reproduce the problem with the intermediate subsurface version 6.0.5101 (I did not try with that one so far)?

bstoeger, 2024-05-02:
> Would that be from 0.7.0 to 0.8.0? Nothing related is mentioned in their release notes..

0.7.0 and 0.8.0? I'm confused about the version numbers...
 
> Is there anything else I can contribute? Would it help if I tried to reproduce the problem with the intermediate subsurface version 6.0.5101 (I did not try with that one so far)?

Is there a digit missing after 510? Yes, that would definitely help as it would narrow down the set of potential culprits.

If you know how to compile for Windows (I don't), you could try to do a `git bisect`.

matthiashoeller, 2024-05-02:
> > Would that be from 0.7.0 to 0.8.0? Nothing related is mentioned in their release notes..
> 0.7.0 and 0.8.0? I'm confused about the version numbers...

I was referring to the release version numbers of libdivecomputer [https://github.com/libdivecomputer/libdivecomputer/releases](https://github.com/libdivecomputer/libdivecomputer/releases), since you mentioned that this lib was different between the two subsurface releases in question.

> Is there a digit missing after 510? Yes, that would definitely help as it would narrow down the set of potential culprits.

Now I'm the one who is confused... I didn't write 510, I wrote 5101, so yes, there is another digit "1" at the end :)

Anyways, I will perform this test tomorrow and post the results here.

> If you know how to compile for Windows (I don't), you could try to do a `git bisect`.

Sorry, no idea what any of that means... I am afraid I don't know how to compile at all.

mikeller, 2024-05-03:
Apologies, it is on my todo list to look into this and see what is going on. I just haven't found the time for it yet.

mikeller, 2024-05-03:
With a bit of luck this should fix it: https://github.com/subsurface/libdc/pull/60

matthiashoeller, 2024-05-03:
Wow that was fast, thank you very much!

If you want, I can test it immediately, but I would need a compiled windows .exe for that. Is that possible? Or should we wait for the next official release and I test it then whenever that comes out?

closed 2024-05-03 by mikeller

matthiashoeller, 2024-05-03:
I don't know if somebody is still reading this, but just in case: I can confirm that with the newest nightly build (6.0.5156) the problem is, indeed, solved, and the serial number is once again displayed correctly.
(Screenshots of 5112 with bug, and 5156 with bugfix, below)

[image]
[image]