subsurface/subsurface #4159: subsurface android location problems

casparjaeqx, 2024-04-20:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I am running subsurface mobile on an android 14 phone i have tryed the version from the google play store and the one from the website( both realease and dev versions) and i am not able to use any of the gps based featuers like the location service. If i go in to the side menue and select location it only show the option for map but not any of the other options like it shows in the documentation. Attatched is a screen shot
### Operating system:

android 14 in english

samsung galaxy s22

same hapens on android 13

### Subsurface version:

6.0.5141

i have not tryied, where can i find older versions

both officiale release and test build with same resualt 

### Steps to reproduce:

1) download the app
2) login with cloud credentials
3) check the location meneu

### Current behavior:

it only shows map in the location meneu

### Expected behavior:

show more than only map like the show gps fixes option or the run location service buton 

### Additional information:

### Mentions:

dirkhh, 2024-04-20:
these options were removed years ago and are no longer supported.

closed 2024-04-20 by dirkhh

casparjaeqx, 2024-04-20:
> these options were removed years ago and are no longer supported.
why where they rememoved, it sounded like such a cool feature.

atdotde, 2024-04-25:
You don't want to give random apps access to GPS data on your phone as this could have severe privacy implications (the app could track your whereabouts). Therefore Andoid puts lots of restrictions on apps that want to ask access to this information. Unfortunately, these restrictions change between Android versions and always adopting to the latest version turned out to be far too time consuming for the subsurface developers (and in particular Dirk), in particular given that this feature was never really working reliably. 

There is however a simple way to capture GPS data of a dive site and add that to Subsurface: Simply use your phone to take a picture at the dive site (that will contain the location information) and later, using the desktop version of Subsurface, add that picture to the dive. If the dive does not have a location, yet, it will take the geo data from the picture.