subsurface/subsurface #4163: Mobile App: You cannot revert accidentally removed dives

SeppoTakalo, 2024-04-23:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When the side menu from `...` button is displayed, it is really easy to accidentally click "Remove" and it will immeadiately remove a dive.
A popup shows up saying something about dive that has been removed, there is even "Cancell" button. Clicking "Cancel" does not do anything.
Revert buttons on `...` menu are not doing anything.

How can I get back those removed dives?

Can you add confirmation before deleting the dive?

### Operating system:

Apple iPhone
iOS 17.4.1

### Subsurface version:

Subsurface mobile Beta 6.0.5124.0

### Steps to reproduce:

Open a subsurface Mobile
Click `...` on botton right corner.
Click delete/remove button.
Try to revert the action.

### Current behavior:

### Expected behavior:

I would expect that the app would ask be a confirmation before removing the dive.
Or at least it should allow me to revert the action.

### Additional information:

### Mentions:

mikeller, 2024-04-23:
Sorry to hear that you accidentally deleted one or more of your dives. If you used cloud storage to backup / synchronise your dives it will be possible to get them restored from there.

> When the side menu from `...` button is displayed, it is really easy to accidentally click "Remove" and it will immeadiately remove a dive. 

After you open the context menu you have to explicitly click on the button labeled 'Delete dive' with the icon of a trash can - not sure how you would 'accidentally' click this?

[image]

> A popup shows up saying something about dive that has been removed, there is even "Cancell" button. Clicking "Cancel" does not do anything.

As the popup says, you can undo the action from the context menu. And the popup even offers you a button that brings you to said context menu. Not sure this can be made more explicit?

>  Revert buttons on `...` menu are not doing anything.

I just tried this multiple times on Android, and 'Undo delete 1 dive(s)...' in the context menu duefully undeleted the dive every case. There is a slim chance that this is broken only on iOS, but I can not test this for lack of a device.

[image]

> Can you add confirmation before deleting the dive?

I am really not sure if it makes sense to ask the user if they really want to delete a dive right after they clicked 'Delete dive'...

SeppoTakalo, 2024-04-23:
https://github.com/subsurface/subsurface/assets/3104794/b30df4cc-a50e-4052-928d-03d39c0f8ced

Here is a sample of the problem.
Sorry about the language, it is in Finnish and I did not want to boot my iPhone.

I'll explain what happens:
1. I open up the Subsurface and you see there is 456 dives.
2. I open up the `...` side panel
3. I click the Delete button
4. Popup appears, I immediately click the Undo button
5. Popup disappear. But the dive is not restored.
6. There is now 455 dives
7. I open up the `...` side panel
8. Cancel/Undo button is grayed out. I cannot cancel it

So I have now lost two dives from my log.

ps. I hope these can be recovered from Cloud storage. Who should I contanct?

mikeller, 2024-04-23:
Ok, I can see the problem.
We did have a recent change that touched the undo / redo functionality on mobile (#4150), so a possible explanation for the differences between what you get and what I get was that undo / redo was not working before this.
From the looks of it the latest iOS version was pushed into the Apple store before this change.
@dirkhh, if you've got the time would it be possible to push another version into the Apple store? Maybe a build of 322c1b56b2d99300b91eec457d407d7aad25a973 (before @bstoeger's latest refactoring).

Also, @dirkhh is the person who can help with restoring dives from cloud storage - or you can do it yourself if you are reasonably good with git.

SeppoTakalo, 2024-04-23:
> Also, @dirkhh is the person who can help with restoring dives from cloud storage - or you can do it yourself if you are reasonably good with git.

I'm reasonable good with git.
Just tell me how can I access the cloud storage. I assume now that it is actually a Git repo where my changes are commits.

mikeller, 2026-10-17:
Hi @SeppoTakalo,

Sorry you (and a second user) lost dives to this. Likely cause: the App Store build you were running predated a prior undo/redo refactor (PR #4150) that's already in master, so the broken build in your hands was built before that fix landed rather than reflecting a defect still present in current source. mikeller couldn't reproduce the problem on Android with current code.

Closing this on that basis, but with lower confidence than usual since we don't have an explicit confirmation that a refreshed App Store/TestFlight build resolves it for you. Please reopen if Undo still silently does nothing after deleting a dive on a current App Store or TestFlight release, in the meantime, recovering a deleted dive from your cloud storage history remains the fallback.

closed 2026-10-17 by mikeller