subsurface/subsurface #4174: Mares Smart Air - "Dive data import error"

Ricool06, 2024-05-02:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I have a Mares Smart Air with a Mares Dive Link USB connector.
I set the dive computer to PC READY mode.
I connect the Dive Link clip.
I plug the USB cable into the Dive Link clip.

I select Import > Import dive from computer in Subsurface.
I select Mares Smart Air from the dropdowns.
The only COM port available is COM3 and it is automatically selected.

I click Download, and after a second or two, a message pops up:
[image]

### Operating system:

Windows 11
What device are you using?
Laptop is Gigabyte AORUS 7 KB laptop.
Only answer this question if you have tried: Does the same happen on another OS?
Yes, also happens on 2023 Macbook Pro,

### Subsurface version:

6.0.5112 Official Release

### Steps to reproduce:

1) I have a Mares Smart Air with a Mares Dive Link USB connector.
1) I set the dive computer to PC READY mode.
1) I connect the Dive Link clip.
1) I plug the USB cable into the Dive Link clip.
1) I select Import > Import dive from computer in Subsurface.
1) I select Mares Smart Air from the dropdowns.
1) The only COM port available is COM3 and it is automatically selected.
1) I click Download.

### Current behavior:

Error message when clicking download.

### Expected behavior:
*
A successful import of my dives.

### Additional information:

[subsurface.log](https://github.com/subsurface/subsurface/files/15193360/subsurface.log)

### Mentions:

anubisg1, 2024-05-18:
I have the same exact error with Mares Quad Air.

anubisg1, 2024-05-18:
i can also confirm that it's working if i use version 5.0.10 , i will test different 6.0.5 releases trying to identy the moment it breaks

mikeller, 2024-05-19:
This seems to be another instance of a known problem, discussed here: https://groups.google.com/g/subsurface-divelog/c/nu4SHPL1xTY/m/c4D6Ngd4AwAJ

Unfortunately a bug in the Mares importer in libdivecomputer made it into Subsurface when libdivecomputer was updated to the latest version in 6.0.5108 - so any releases before that should be fine.

I'll have a look if it is possible to get this fix for the Mares importer into Subsurface without creating too much of a mess.

closed 2024-05-24 by mikeller