subsurface/subsurface #4175: Addition of freediving-specific features

sjaeckel, 2024-05-03:
### Describe the issue:
- [x] Discussion request

### Issue long description:
Hi,

I'm integrating the Cressi Nepto (c.f. libdivecomputer/libdivecomputer#33) which is a pure freediving DC.

There are currently three questions that came up, while looking into how the data of the DC is handled in Subsurface.

I'm using the Cressi vocabulary here to keep it simple:
* `dip` - you're in the water and you go for a single "freedive"
* `dip time` - the time of a single dip
* `dive time` - all dips added up
* `surface time` - either "the time between two dips", or "all time in between dips added up"
* `session time` - `dive time` + `surface time` (of all dips) -- that's equivalent to a 'dive' in Subsurface terms

1. Currently Subsurface works on full seconds, but the Nepto provides a sample rate of 0.5 seconds, how could this be handled?
2. Subsurface has currently no way to differentiate between dip, dive, and session time. How could this be done best?
3. the DC provides "advanced statistics" per dip: SurfaceTime, MaxDepth, DipTime, MinTemperature, MaxDescentSpeed, MaxAscendSpeed, TaravanaViolation. Would it make sense to integrate those as well? Personally I would like to see them.

ad 2.: I'm aware that there's the option to split a dive, but IMO that doesn't really make sense to split each session up into maybe tens of single dives.
ad 3.: Taravana is similar to decompression sickness, and it can occur when there's not enough surface time in between two dips, so I'd like if that were displayed as a warning of a specific dip.
ad 2. and 3.: if we'd find a solution how to split the profile view up (without really splitting it up visually) and make each dip somehow "selectable", we already had a way how to decide to display the advanced stats.

I'm aware that all those changes aren't trivial and e.g. 1) would most likely touch a majority of the codebase. That's why I decided to start a discussion first whether those changes could be acceptable and how they should best be implemented.

Thanks in advance.

mikeller, 2024-05-05:
Hi Steffen.

It is an interesting point that you are raising.
I have to preface my reply with the comment that Subsurface so far has a focus on SCUBA diving.
I think for further considerations we have to keep this in mind, and not try to make Subsurface into one application that does multiple things, and probably will end up none of them well, and even more struggle to get sufficient maintenance.
So my hunch for how to go on about this is to figure out how we can make this work by building on top of what we already have, and then see if what we come up will actually have a chance to be used by freedivers, and will hence be worthwhile doing.

One question that I have is: Are there other freediving logging applications out there, or what do freedivers use to upload their dives?

Based on the preface, what I see as realistic at this point in time will be to add support for the Cressi Nepto (and potentially other, similar free diving computers) as a 'normal' Subsurface dive computer:
- have the dive log parser reduce the sample rate to 1 sample per second by interpolating the values;
- accept that profiles will be visualised as a 'dive' / series of 'dips', and that the user will have to zoom in to see a dip in detail.

Subsurface allows certain events appearing in the time line in dive logs on dive computers (gas changes, deco state changes, limit violations) to be added to the log and visualised in the dive profile - this could probably be used to add 'dip start' / 'dip end' markers, and can certainly be used to visualise any kinds of violations at the point that it is reported by the dive computer.
I think this should be a starting point for adding freediving specific information in a way that re-uses existing facilities.
These markers could then be used to do things like calculated per-dip statistics, or automatically zoom into one dip. All of this functionality could potentially live in a new 'Freedive' tab in the information pane that is only shown when the current dive is a freedive.

What are your thoughts on this approach?

sjaeckel, 2024-05-07:
Hi Mike,

thanks a lot for your input.

> Are there other freediving logging applications out there, or what do freedivers use to upload their dives?

Not that I'm aware of. I also stopped looking immediately after I saw that there's a well maintained and supported solution that 1. is FLOSS and 2. already supports most DC's.

> * have the dive log parser reduce the sample rate to 1 sample per second by interpolating the values;

OK, I will see how to do this best. FYI the Mares Smart Apnea is fully supported and libdivecomputer already pushes sub-second values to Subsurface when reading it out. I guess I can live with the fact of having only 1s of resolution for now, but IMO this should be tackled after we got those first topics sorted out.

> * accept that profiles will be visualised as a 'dive' / series of 'dips', and that the user will have to zoom in to see a dip in detail.

That absolutely makes sense.

> Subsurface allows certain events appearing in the time line in dive logs on dive computers (gas changes, deco state changes, limit violations) to be added to the log and visualised in the dive [...]

That sounds like a plan.

I'll finish the basic Nepto integration for now, and see how far I can get with the dip related extension and the 'Freedive' specific part in the info pane. I'll keep you updated :)

atdotde, 2024-05-07:
Just a quick comment regarding seconds: Internally, Subsurface stores many (dimensionful) values as integers to prevent further complications from floating point maths. To this end, it uses "internal units" that are small enough that you don't lose any meaningful resolution from the quantization to integer values: All lengths (such as depth) are internally in mm, all pressures in mbar, and yes, you guessed right, all durations in seconds (even used when converting calendrical dates). That means it would require a huge rewrite of internal structures to allow for fractions of seconds and thus all formulas involving time would have to be changed. And no, switching to floating point is not an option since that brings its own problems with it (like comparing two floating point numbers for equality is not really defined). I don't think anyone in their right might would attempting that worth the effort.

sjaeckel, 2024-05-07:
I didn't look into detail, but a switch from seconds-based timekeeping to milliseconds-based doesn't seem like the horror you're trying to picture, at first glance.

We'll see. It's the last point on the list.

atdotde, 2024-05-07:
There are about 10^12 milliseconds since my first dive. That number requires about 40 bits to store.

dirkhh, 2024-05-07:
msec is not a reasonable target as general time unit. A fairly typical dive takes more than a million msec.

This reminds me of people arguing about the third digit behind the decimal separator for water salinity and mm precision for depths (I'm still annoyed that Linus did that). It doesn't matter if your (free) dive was 76 or 76.105 seconds. How do you even define start and end of a dive. 

Sure, vendors trying to sell you new gear will try anything to differentiate, but to me this is just pointless noise. And it has significant impact on things that I do care about (like size of logbook data, transmission time to the cloud, etc).

This is a scuba dive log first and foremost. Yes, you can deal with CCR and even free diving if you want to, but the target audience (and vast majority of user base) is scuba divers. The majority of dive computers still uses 10 to 30 seconds sample rate. And if you compare a dive profile from a 10s computer to that from one of the silly 1s sample rate computers, you gain absolutely nothing, instead you get visualization of the waves above you and the noise in your pressure sensor. 

So here is a very loud and hard NO on msec resolution from me.

sjaeckel, 2024-06-05:
@dirkhh I have one question regarding the IMO most important part of your statement.

> This is a scuba dive log first and foremost. Yes, you can deal with CCR and even free diving if you want to, but the target audience (and vast majority of user base) is scuba divers.

Do you want it to stay like that or would you accept if I implemented some freedive-specific features?

dirkhh, 2024-06-05:
we already accepted tons of changes that were targeting CCR and Freediving use cases - but always in the context where these were additional components that were relevant to that audience.
ms dive time precision is a (sorry, no offense, but let's be crisp here) completely ridiculous idea. There is no way to measure the duration of a dive with that precision. I will argue that at the speed the human body can deal with pressure changes - plus the lack of precision of actual pressure sensors - even 1/10th of a second precision is nothing but a marketing gimmick, but I have already been informed that I am WRONG about this.
So yes, I would respond differently if the proposal was to move to 1/10th of a second precision.
But even then I would argue about the incredible amount of disruption this would bring to all parts of the software (and I mean absolutely everything, every dive related calculation, UI element, the file format, everything). This would be a massive change for really really marginal advances to the users (besides the usual fake precision games).

sjaeckel, 2024-06-05:
Thanks for your prompt reply.

> msec is not a reasonable target as general time unit. A fairly typical dive takes more than a million msec.

In a world where 64bit integers are the norm we could even choose nano seconds as resolution and we could still count longer than subsurface currently can.

That's a lot of "could" and not any "should" or "must". I'm not saying that I want it or would implement it. For now.

> This reminds me of people arguing about the third digit behind the decimal separator for water salinity and mm precision for depths (I'm still annoyed that Linus did that). It doesn't matter if your (free) dive was 76 or 76.105 seconds. How do you even define start and end of a dive.

Maybe I want to see how stable/balanced I am in the pool while diving on distance in the same depth!?
Maybe I want to see the moment when I did an arm stroke!?

> Sure, vendors trying to sell you new gear will try anything to differentiate, but to me this is just pointless noise.

I bought the cheapest OK'ish quality second-hand DC for free-diving which is 5 years old ... and it supports 500ms accuracy ... FYI some Mares ones even have 250ms accuracy.

> And it has significant impact on things that I do care about (like size of logbook data, transmission time to the cloud, etc).

No, it won't. At least not to the first two things you care about. The only thing it has real impact on is RAM usage and maybe some CPU cycles of the device where subsurface runs. The formatting of timestamps is pretty standard and if your DC doesn't report sub-second values, this will result in absolutely no difference for you. We won't start storing `12:34` as `12:34.000`.

-----

libdivecomputer has already changed its internal accuracy to msec, i.e. Subsurface already gets sub-second values if the DC supports it (e.g. some Mares DCs). 

Currently it's cutting off the sub-second part and thereby loosing this information.

Plotting happens on the same second multiple times, which is no problem at all IMO. It's not pretty, but also not that important for now.

I've opened PR #4237 to address this issue and PR #4236 to add the dip details to the information tab.

Let's keep the discussion about the "what and why" in this issue and the discussions about the "how" in the PRs.

sjaeckel, 2024-06-11:
From https://github.com/subsurface/subsurface/pull/4237#issuecomment-2160235130

> > I still fundamentally think that ms resolution is nonsense, just as mm depth is nonsense.
> 
> I still fundamentally think that a factor 1000 doesn't matter if one has 64bit integers available.

Just FTR: I would already have given up on this discussion if you wouldn't have made `ds` a viable option for you.

For me it's either ms or keep it as it is.

mikeller, 2026-10-18:
Hi @sjaeckel,

Following up on this after a long gap, after discussion, we've decided not to make freediving-specific support a priority area for Subsurface right now. Freedivers aren't a main focus of the app's current direction, and that's the deciding factor here, not any issue with the design work itself.

Your PRs (#4236, #4237) and the design discussion we had are solid groundwork, the 1Hz resample-on-import approach, treating a freediving session as a single dive with dip boundaries via the existing event/marker mechanism, and the proposed Freedive info-pane tab are all still there if you or anyone else wants to pick them up independently. We're not closing the door on an external contributor finishing and maintaining this, just not committing project resources to drive it ourselves at this time.

Closing this as declined for now on prioritisation grounds. Thanks for the work already put into this, happy to review a rebased PR if you want to carry it forward yourself, now or later.

closed 2026-10-18 by mikeller