subsurface/subsurface #4176: Import: Download Dialog Hangs on Cancel after Successful Dive Download

mikeller, 2024-05-05:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When trying to cancel out of the download dives dialogue after successfully downloading one or more dives Subsurface hangs indefinitely.

### Operating system:

Linux ubuntu 23.10

### Subsurface version:

6.0.5156

### Steps to reproduce:

1. 'Import' / 'Import from dive computer';
2. Select computer that exists and has downloadable dives;
3. click 'Download';
4. wait until download has finished;
5. click 'Cancel'.

### Current behavior:

Subsurface hangs indefinitely.
Backtrace while hanging:
```
#0  delete_single_dive (idx=0)
    at /home/mikeller/git/subsurface/core/divelist.c:775
#1  0x00005555558d3317 in divelog::clear (this=0x555557a3c7e8)
    at /home/mikeller/git/subsurface/core/divelog.cpp:70
#2  0x000055555583f036 in DiveImportedModel::clearTable (this=0x555557a3c770)
    at /home/mikeller/git/subsurface/qt-models/diveimportedmodel.cpp:119
#3  0x000055555577af72 in DownloadFromDCWidget::on_cancel_clicked
    (this=0x7fffffffc830)
    at /home/mikeller/git/subsurface/desktop-widgets/downloadfromdivecomputer.cpp:549
#4  0x00005555556fd123 in DownloadFromDCWidget::qt_metacall
    (this=0x7fffffffc830, _c=QMetaObject::InvokeMetaMethod, _id=2, _a=0x7fffffffbc40)
    at /home/mikeller/git/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_downloadfromdivecomputer.cpp:226
#5  0x00007ffff5c8a23e in ??? () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff74deb96 in QAbstractButton::clicked(bool) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff74dee0e in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff74e09eb in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff74e0c0b in QAbstractButton::mouseReleaseEvent(QMouseEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff742db28 in QWidget::event(QEvent*) ()
--Type <RET> for more, q to quit, c to continue without paging--c
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff73eac82 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff73f31bd in QApplication::notify(QObject*, QEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff5c507a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff73f14b2 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff7446da1 in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff744a12f in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff73eac82 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff5c507a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff675abdb in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#20 0x00007ffff672e62c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#21 0x00007ffff0548b74 in userEventSourceDispatch(_GSource*, int (*)(void*), void*) () at /lib/x86_64-linux-gnu/libQt5WaylandClient.so.5
#22 0x00007ffff4084b2c in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff40e046f in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007ffff4082d20 in g_main_context_iteration ()
    at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007ffff5cabdaa in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff5c4f15b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff75faa37 in QDialog::exec() ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00005555557a97d8 in MainWindow::on_actionDownloadDC_triggered
    (this=0x555555fc3660)
    at /home/mikeller/git/subsurface/desktop-widgets/mainwindow.cpp:623
#29 0x00005555556ff973 in MainWindow::qt_metacall
    (this=0x555555fc3660, _c=QMetaObject::InvokeMetaMethod, _id=14, _a=0x7fffffffcbc0)
    at /home/mikeller/git/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:371
#30 0x00007ffff5c8a23e in ??? () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff73e4256 in QAction::triggered(bool) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#32 0x00007ffff73e6f57 in QAction::activate(QAction::ActionEvent) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#33 0x00007ffff756fd22 in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#34 0x00007ffff7577bdb in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#35 0x00007ffff742db28 in QWidget::event(QEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#36 0x00007ffff73eac82 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#37 0x00007ffff73f31bd in QApplication::notify(QObject*, QEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#38 0x00007ffff5c507a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#39 0x00007ffff73f14b2 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#40 0x00007ffff744786a in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#41 0x00007ffff744a12f in ??? () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#42 0x00007ffff73eac82 in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#43 0x00007ffff5c507a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#44 0x00007ffff675abdb in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#45 0x00007ffff672e62c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#46 0x00007ffff0548b74 in userEventSourceDispatch(_GSource*, int (*)(void*), void*) () at /lib/x86_64-linux-gnu/libQt5WaylandClient.so.5
#47 0x00007ffff4084b2c in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#48 0x00007ffff40e046f in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#49 0x00007ffff4082d20 in g_main_context_iteration ()
    at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#50 0x00007ffff5cabdaa in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#51 0x00007ffff5c4f15b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#52 0x00007ffff5c57904 in QCoreApplication::exec() ()
    at /lib/x86_64-linux-gnu/libQt5Core.so.5
#53 0x00005555556ed585 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/mikeller/git/subsurface/subsurface-desktop-main.cpp:107
```

### Expected behavior:

The download dialogue is closed.

### Additional information:

It looks like, after being called by `divelog::clear()`,  `delete_single_dive()` in `core/divelist.c` always exits early, as it cannot find the dive to be deleted. But I could not quite figure out what we are failing to hook up correctly when creating the temporary log in the download dialogue.

### Mentions:

@bstoeger I suspect that this is caused by 8cd451fc338da275f66b15181894e37621698109.

bstoeger, 2024-05-05:
> It looks like, after being called by divelog::clear(), delete_single_dive() in core/divelist.c always exits early, as it cannot find the dive to be deleted. 

My interpretation would be that we have a dive_table with a `NULL` entry.

bstoeger, 2024-05-05:
Can you check if this fixes (or improves) the problem:
```
diff --git a/core/divelist.c b/core/divelist.c
index 30a4de16e..88e94cd4b 100644
--- a/core/divelist.c
+++ b/core/divelist.c
@@ -772,10 +772,10 @@ struct dive *unregister_dive(int idx)
 void delete_single_dive(int idx)
 {
 	struct dive *dive = get_dive(idx);
-	if (!dive)
-		return; /* this should never happen */
-	remove_dive_from_trip(dive, divelog.trips);
-	unregister_dive_from_dive_site(dive);
+	if (dive) {
+		remove_dive_from_trip(dive, divelog.trips);
+		unregister_dive_from_dive_site(dive);
+	}
 	delete_dive_from_table(divelog.dives, idx);
 }
```
?

If this fixes it, we still should find the reason that there are `NULL` entries in the divetable. I don't think that's supported.

bstoeger, 2024-05-05:
Ooops.

This code is completely bogus.

`delete_single_dive` clears the dive from the global dive table.

bstoeger, 2024-05-05:
Please try this:
```
diff --git a/core/divelog.cpp b/core/divelog.cpp
index ce8d3c2f0..7951578dd 100644
--- a/core/divelog.cpp
+++ b/core/divelog.cpp
@@ -66,8 +66,7 @@ struct divelog &divelog::operator=(divelog &&log)
 
 void divelog::clear()
 {
-	while (dives->nr)
-		delete_single_dive(0);
+	clear_dive_table(log.dives);
 	while (sites->nr)
 		delete_dive_site(get_dive_site(0, sites), sites);
 	if (trips->nr != 0) {
```
or something similar.

mikeller, 2024-05-05:
> Please try this:

Yes, much better - I had similar thoughts, but wasn't sure if there was intent behind going through the global dive table for this.

Shall I pull request this?

bstoeger, 2024-05-05:
Yes, please.

PR #4179 (2024-05-05): Import: Fix Application Hang when Cancelling the Download Dialogue.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix a bug causing the 'Download from dive computer' dialogue to hang when the user attempts to cancel the dialogue after successfully downloading one or more dives.

### Changes made:

### Related issues:

Fixes #4176.

### Additional information:

### Release note:

###
commit 46cf2fc0 (2024-05-05): Import: Fix Application Hang when Cancelling the Download Dialogue.

Fix a bug causing the 'Download from dive computer' dialogue to hang
when the user attempts to cancel the dialogue after successfully
downloading one or more dives.

Fixes #4176.
closed 2024-05-05 by bstoeger