subsurface/subsurface #4185: Cloud Backend Source?

hughesjs, 2024-05-08:
### Describe the issue:
- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:
I'm wondering if/where the source-code for the cloud backend is? This is so that 1) I can see what's happening with my data and 2) consider self-hosting.

dirkhh, 2024-05-08:
it's not published as that would disclose my attempts to avoid the rampant abuse that this service saw in the early days.
I didn't appreciate people creating more than a TB in data transfer chargers in a day.
But if an attacker can see how I check for this, it's easy to work around it.

Additionally, as was discussed many times self hosting isn't supported. It's a trade-off between complexity and ability to support the average user of Subsurface.

Subsurface is open source. If you want to self host it should be quite easy for you to make the changes necessary to talk to your own backend.

closed 2024-05-08 by dirkhh

hughesjs, 2024-05-08:
That's fair enough, thanks for taking the time to answer my question.