subsurface/subsurface #4188: CCR Planner: Changing used gas from OXYGEN not possible and crashes Subsurface

linuxcrash, 2024-05-10:
### Describe the issue:
- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
If you create a CCR dive and then add the used CCR gases and Bailout tanks, the dive points are by default set to OXYGEN and can not be changed. By clicking on the dropdown and leaving the focus from the cell there Subsurface crashes after some seconds without any warning.

### Operating system:
Fedora 39

### Subsurface version:
Weekly build 6.0.5.1.44 and Nightly build 6.0.5.62. I used the AppImage and the COPR version.

### Steps to reproduce:
1. Plan a dive
2. Change to CCR
3. Add 1-2 gases for oxygen and diluent. 
4. Try to change gas used in Dive planner points to something else.

### Current behavior:
Crash

### Expected behavior:
Change to diluent gas possible.
[image]

bstoeger, 2024-05-11:
Thanks for the report.

For some reason, your screen shot doesn't show up.

This is hard for me to reproduce, since I have only a rudimentary understanding of what a CCR is. I tried to follow your steps and could add gas changes:
[image]

What am I doing wrong?

mikeller, 2024-05-11:
@bstoeger: I got a reproduction (try changing 'Used gas' in the 'Dive planner points' list), and I've got a fix which is trivial. But the underlying problem is that this dropdown is not populated at all, making it unusable:

```
diff --git a/qt-models/diveplannermodel.cpp b/qt-models/diveplannermodel.cpp
index b065138f6..40db8e48b 100644
--- a/qt-models/diveplannermodel.cpp
+++ b/qt-models/diveplannermodel.cpp
@@ -381,7 +381,7 @@ bool DivePlannerPointsModel::setData(const QModelIndex &index, const QVariant &v
                        if (value.toInt() >= 0)
                                p.cylinderid = value.toInt();
                        /* Did we change the start (dp 0) cylinder to another cylinderid than 0? */
-                       if (value.toInt() != 0 && index.row() == 0)
+                       if (value.toInt() >= 0 && index.row() == 0)
                                cylinders.moveAtFirst(value.toInt());
                        cylinders.updateTrashIcon();
                        break;
```

I'll investigate some more.

bstoeger, 2024-05-12:
Thanks, I didn't realize that you have to use the table, not the profile.. Now I also get the crash.

bstoeger, 2024-05-12:
Note that this doesn't have anything to do with CCR. The combobox doesn't work even in normal mode. And I distinctly remember that we had the exact same problem before.

bstoeger, 2024-05-12:
Must be my changes to the way the GasSelectionModel is reset on every invocation of the combobox.

bstoeger, 2024-05-12:
Well, there is one "obvious" bug: in `AirTypesDelegate::AirTypesDelegate`, the lambda makes a copy of the dive, whose gas list is, of course, not updated.

Sometimes, lambdas are terribly subtle. :(

bstoeger, 2024-05-12:
This seems to fix the problem (including your fix):
```
diff --git a/desktop-widgets/modeldelegates.cpp b/desktop-widgets/modeldelegates.cpp
index e432448dc..ccda7c9b4 100644
--- a/desktop-widgets/modeldelegates.cpp
+++ b/desktop-widgets/modeldelegates.cpp
@@ -371,8 +371,8 @@ void AirTypesDelegate::setModelData(QWidget *editor, QAbstractItemModel *model,
 }
 
 AirTypesDelegate::AirTypesDelegate(const dive &d, QObject *parent) :
-	ComboBoxDelegate([d] (QWidget *parent) { return new GasSelectionModel(d, parent); },
-			      parent, false)
+	ComboBoxDelegate([&d] (QWidget *parent) { return new GasSelectionModel(d, parent); },
+			       parent, false)
 {
 }
 
diff --git a/qt-models/diveplannermodel.cpp b/qt-models/diveplannermodel.cpp
index b065138f6..40db8e48b 100644
--- a/qt-models/diveplannermodel.cpp
+++ b/qt-models/diveplannermodel.cpp
@@ -381,7 +381,7 @@ bool DivePlannerPointsModel::setData(const QModelIndex &index, const QVariant &v
 			if (value.toInt() >= 0)
 				p.cylinderid = value.toInt();
 			/* Did we change the start (dp 0) cylinder to another cylinderid than 0? */
-			if (value.toInt() != 0 && index.row() == 0)
+			if (value.toInt() >= 0 && index.row() == 0)
 				cylinders.moveAtFirst(value.toInt());
 			cylinders.updateTrashIcon();
 			break;
```
But I still get these disconcerting messages
```
QAbstractItemView::commitData called with an editor that does not belong to this view
QAbstractItemView::closeEditor called with an editor that does not belong to this view
```

bstoeger, 2024-05-12:
Argh. I still can create crashes with this:

Go to planner. Don't add any gases. Select something from the Gas-Combobox. Crash. :(

bstoeger, 2024-05-12:
That seems to be a different (very old?) bug. Added a "fix" to the PR.

linuxcrash, 2024-05-12:
@bstoeger thanks for taking this up. I will test the next nighty that should have your commit tomorrow as soon as I'm back home.

commit 9e2dea98 (2024-05-12): planner: fix gas selection

The lambda that created the list of gases took a copy not a
reference of the planned dive. Of course, that never had its
gases updated. Ultimately this would crash, because this sent
an index of "-1" on change.

Fix by
1) Using a reference to the dive, not the copy
2) Catch an invalid "-1" index (by Michael Keller <github@ike.ch>)

Fixes #4188
PR #4190 (2024-05-12): planner: fix gas selection
The lambda that created the list of gases took a copy not a reference of the planned dive. Of course, that never had its gases updated. Ultimately this would crash, because this sent an index of "-1" on change.

Fix by
1) Using a reference to the dive, not the copy
2) Catch an invalid "-1" index (by Michael Keller <github@ike.ch>)

Fixes #4188

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fi [... cut]
commit d061a54e (2024-05-12): planner: fix gas selection

The lambda that created the list of gases took a copy not a
reference of the planned dive. Of course, that never had its
gases updated. Ultimately this would crash, because this sent
an index of "-1" on change.

Fix by
1) Using a reference to the dive, not the copy
2) Catch an invalid "-1" index (by Michael Keller <github@ike.ch>)

Fixes #4188
closed 2024-05-12 by bstoeger