subsurface/subsurface #4193: Mares Genius will connect via serial port (linux) bit ist a bluetooth device

jschuppe, 2024-05-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I have a mares genius dice computer an can read this on android with the divemate app. On my linux notebook I can connect the dive computer via bluetooth to my kubuntu os. If i choose Mare / Genius, the Subsurface software shows my in the third combobox all serial ports instead of the Bluetooth devices. The Android version of Subsurface can read all the dives (14 pices) from the computer in a span of 45 minutes :-(. But the read dives will not be imported into the dive log, all read dives are lost after i go to the dive log.

Mein Mares Genius lässt sich unter Android problemlos mit der Divemate App auslesen. Auf meinem Notebook verwende ich Kubuntu Linux und kann den Mares Genius problemlos mit dem Rechner paaren. Wenn ich im Subsurface Mare / Genius auswähle bekomme ich in der dritten Combobox alle Seriellports angezeigt, jedoch nicht die Bluetooth Geräte. Unter Android kann ich mit der App alle 14 Tauchgänge vom Tauchcomputer lesen, dauert jedoch ca. 45 Minuten. Wenn ich dann nach dem Laden in das Logbuch zurück gehe, dann sehe ich dort wieder 0 Tauchgänge, die gelesenen Daten wurden nicht übernommen.

### Operating system:
* Kubuntu Linux 23.10 mit KDE
* Android 13
* Mares Genius

### Subsurface version:
Version: 6.0.5173
installed via apt-get

### Steps to reproduce:

### Current behavior:

### Expected behavior:
* For Mares Genius I should choose Bluetooth devices instead of serial ports (linux)
* Store all read dives from Import to the Logbook on Android

### Additional information:

### Mentions:

mikeller, 2024-05-13:
> Sorry i have found the error on linux ....

Thank you for the update - do you care to share what the problem was, so others coming across this issue can benefit from your insights?

mikeller, 2026-10-17:
Hi @jschuppe,

Thanks for this report, it actually bundles two separate things and we'd like to pull them apart. First, could you confirm specifically why you'd expect the Genius to be recognised as Bluetooth rather than serial on Linux (e.g. is it paired as a BLE device at the OS level, or are you expecting Subsurface to offer a BLE option that isn't showing)? Second, on the data-loss claim, could you describe exactly what data went missing (which dives, which fields) and how you noticed, ideally with a before/after comparison or the affected log file?

If we don't hear back with those details within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.