subsurface/subsurface #4208: Issues With Window Rendering Under Wayland

hughesjs, 2024-05-24:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When I open any window except the main one (about/preferences/etc), it doesn't render completely until I either mouse over each control or drag it to another screen and back.

### Operating system:
Arch Linux

### Subsurface version:
6bdfee080dbf7d321e35158b4fd1a2d6fe5f367e

### Steps to reproduce:
Open about screen

### Current behavior:
It doesn't refresh the screen region, displaying whatever was last under it.

[image]

(As shown if I tab out and back to it)

[image]

Then if I mouse over some controls, they render one by one:

[image]

If I drag it to a different screen and back it forces a rerender and it looks fine.

### Expected behavior:
This just renders properly

### Additional information:
No errors in terminal.
Replicated in both Hyprland and KDE

### Mentions:

hughesjs, 2024-05-24:
Rebuilding with QT6 seems to have fixed this. So maybe we just close this as an upstream issue with QT 5.5.13?

hughesjs, 2024-05-24:
For anyone else who finds themselves here on Arch, you can use this PKGBUILD:

```
pkgname=subsurface-qt6-git
_pkgname=subsurface
pkgver=20240521.6bdfee080
pkgrel=1
pkgdesc='Divelog program with experimental QT6 support (fixes a few rendering issues under wayland)'
url='https://subsurface-divelog.org/'
license=('GPL2')
arch=('i686' 'x86_64')
makedepends=('git' 'cmake' 'asciidoc' 'qt5-tools')
depends=('libzip' 'libxml2' 'libxslt' 'sqlite' 'libusb' 'libgit2' 'googlemaps'
         'subsurface-libdc-git' 'qt6-svg' 'qt6-location' 'qt6-connectivity')
source=('git+https://github.com/subsurface/subsurface')
sha256sums=('SKIP')

# No QtLocation in Qt6 yet...
# https://bugreports.qt.io/browse/QTBUG-96795

provides=('subsurface')
conflicts=('subsurface')

pkgver() {
	cd "${srcdir}/${_pkgname}"
	git log -1 --format='%cd.%h' --date=short | tr -d -
}

build() {
	cd "${srcdir}/${_pkgname}"
	install -d build
	cd build
	cmake \
		-DCMAKE_BUILD_TYPE=Release \
		-DCMAKE_INSTALL_PREFIX=/usr \
		-DLIBDIVECOMPUTER_INCLUDE_DIR=/usr/include/libdivecomputer \
		-DBUILD_WITH_QT6:BOOL=ON \
		..
	make
}

package() {
	cd "${srcdir}/${_pkgname}"
	cd build
	make DESTDIR="${pkgdir}" install
}

```

hhassey, 2024-11-07:
I am using Arch Linux under Wayland and I have no issues. For whatever this is worth

mikeller, 2026-10-17:
Hi @ScottishTapWater,

Thanks for narrowing this down and testing it yourself. As you found, this was an upstream Qt5 Wayland rendering issue rather than a Subsurface-specific bug, rebuilding with Qt6 resolved it for you, and hhassey confirmed no rendering issues on Arch+Wayland either. The ongoing move to Qt6 builds is the long-term fix path here.

Closing this as resolved by the Qt6 migration. Please reopen if you still see windows failing to render properly under Wayland on a current Qt6 build.

closed 2026-10-17 by mikeller