subsurface/subsurface #4209: Mares Puck Prod unable to download all dives (ERROR: Unexpected answer byte. [in src/mares_iconhd.c:219 (mares_iconhd_packet)])

sulphur, 2024-05-25:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When downloading dives from Mares Puck Pro ( firmware version v4 ). Some dives are downloaded then error occures and dive computes restarts ... 
 
### Operating system:

Macbook Pro M1max with MacOS 14.5 (23F79)
### Subsurface version:

Subsurface-6.0.5144-CICD-release

### Steps to reproduce:

- Import From Dive computer
- Select Mares, Puck Pro, /dev/tty.SLAB_USBtoUART
- click on download 
 
### Current behavior:

download get interupted and dive computer resets

### Expected behavior:

To download all dives
### Additional information:

Log: 
```
Subsurface: v6.0.5144.0, built with libdivecomputer v0.9.0-devel-Subsurface-NG (62a29eea15137ba7b0f2e10fae095517cc9d8341)
[0.000000] INFO: Open: name=/dev/tty.SLAB_USBtoUART
[0.025289] INFO: Configure: baudrate=115200, databits=8, parity=2, stopbits=0, flowcontrol=0
[0.042031] INFO: Timeout: value=3000
[0.042040] INFO: DTR: value=0
[0.045013] INFO: RTS: value=0
[0.048000] INFO: Purge: direction=3
[0.050146] INFO: Write: size=2, data=C267
[0.060516] INFO: Read: size=1, data=AA
[0.067115] INFO: Read: size=140, data=000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005075636B2050726F000000000000000030342E30302E30300400000031342D30342D31352B8B4443303234000000000000000000000000000000000000000000000000000000
[0.067121] INFO: Read: size=1, data=EA
Event: vendor=000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000005075636B2050726F000000000000000030342E30302E30300400000031342D30342D31352B8B4443303234000000000000000000000000000000000000000000000000000000
[0.067294] INFO: Write: size=2, data=E742
[0.067364] INFO: Write: size=8, data=0C00000004000000
[0.069274] INFO: Read: size=1, data=AA
[0.075770] INFO: Read: size=4, data=F1EF0000
[0.075775] INFO: Read: size=1, data=EA
Event: model=24 (0x00000018), firmware=0 (0x00000000), serial=61425 (0x0000eff1)
[0.075884] INFO: Write: size=2, data=E742
[0.075999] INFO: Write: size=8, data=0120000004000000
[0.077882] INFO: Read: size=1, data=AA
[0.084325] INFO: Read: size=4, data=FFFFFFFF
[0.084330] INFO: Read: size=1, data=EA
[0.084390] INFO: Write: size=2, data=E742
[0.084513] INFO: Write: size=8, data=0130000004000000
[0.086322] INFO: Read: size=1, data=AA
[0.092793] INFO: Read: size=4, data=C0A70300
[0.092799] INFO: Read: size=1, data=EA
[0.092865] INFO: Write: size=2, data=E742
[0.092934] INFO: Write: size=8, data=C0A6030000010000
[0.094968] INFO: Read: size=1, data=AA
[0.132602] INFO: Read: size=256, data=3B0000020000AB00180000020000AB000D0000020000AB00010000020000AA00070000020000AB00060000020000AB00010000020000AB00010000020000AB00010000020000AB00010000020000AB00000000020000AB00010000020000AB00010000020000AB00010000020000AB00010000020000AB00010000020000AB00030000020000AB00010000020000AB00010000020000AB00030000020000AB0001000002F000DF01B30109003000160004007C0040050000160040062180400632807805000000000800A81F0000D700000000000404EF18EF18EF18EF18EF18EF18EF18EF18EF18EF189800C900C500B9021000100000000000000000000202
[0.132612] INFO: Read: size=1, data=EA
[0.132711] INFO: Write: size=2, data=E742
[0.132732] INFO: Write: size=8, data=C0A5030000010000
[0.134806] INFO: Read: size=1, data=AA
[0.172358] INFO: Read: size=256, data=1D0010023200A900320010022F00A900240010022C00A900400010022D00A900000010023300AA00010010023800AA00000010023A00AA00000010023B00AA000D0010023800AA005D0010023400AA00540010022C00AA008F0018022500AA00B50018021A00AA00970018021800AA00400000021500AA00410000020F00AA00580000020B00AA00780008020300AB006C0008020100AA00260000020000AB002D0000020200AB00000000020000AB001D0000020000AB00100000020400AA00000000020000AA003E0000020100AA000C0000020200AA00030000020100AA00180000020000AA002C0000020300AB00000000020300AB002D0000020300AB00
[0.172479] INFO: Read: size=1, data=EA
[0.172611] INFO: Write: size=2, data=E742
[0.172691] INFO: Write: size=8, data=C0A4030000010000
[0.174590] INFO: Read: size=1, data=AA
[0.212708] INFO: Read: size=256, data=430080023A00A900160080023A00A900150080023900A900230080023A00A900030080024100A900000080024400A900000080024400A900060080024600A900000080024800A900000080025000A900000080025100A900030000025000A900600000024900A900550000024700A900350000024500A900270000024400A900160000024300A900400000024300A900010000024400A9003A0000024300A900430000024100A900000000024200A900260000023F00A900410010023B00A900350010023A00A900180010023800A900290010023C00A900000010023F00A900000000024000A900270000023900A9007A0018023500A900520010023300A900

..... LONG AND REPETIVE blocks... ( deleted because body was to big )
[4.918025] INFO: Read: size=1, data=EA
[4.918641] INFO: Write: size=2, data=E742
[4.918868] INFO: Write: size=8, data=C02E030000010000
[4.920297] INFO: Read: size=1, data=AA
[4.958070] INFO: Read: size=256, data=0000008087009B001600008089009B00000000808E009B00000000808E009A001800008090009A000000008096009A00000000809B00990000000080A100990000000080AA00990000000080B100990000000080B100990012000080B400990000000080B500980000000080B40098002F000080B100980024000080B000980015000080B000980000000080AE00980051000080AE0098001B000080AC0098003B000080AD00980013000080AD00980000000080B300980000000080B500980000000080B90097000C000080BB00970024000080BA00970013000080BB00970001000080BE00970000000080C200970000000080C400970006000080C6009600
[4.958094] INFO: Read: size=1, data=EA
[4.958537] INFO: Write: size=2, data=E742
[4.958715] INFO: Write: size=8, data=C02D030000010000
[7.962395] INFO: Read: size=0, data=
[7.962487] ERROR: Failed to receive the answer. [in src/mares_iconhd.c:213 (mares_iconhd_packet)]
[7.962506] INFO: Sleep: value=100
[8.066648] INFO: Purge: direction=1
[8.068144] INFO: Write: size=2, data=E742
[8.068248] INFO: Write: size=8, data=C02D030000010000
[8.084369] INFO: Read: size=1, data=07
[8.084383] ERROR: Unexpected answer byte. [in src/mares_iconhd.c:219 (mares_iconhd_packet)]
[8.084664] INFO: Sleep: value=100
[8.186969] INFO: Purge: direction=1
[8.188264] INFO: Write: size=2, data=E742
[8.188479] INFO: Write: size=8, data=C02D030000010000
[8.190654] INFO: Read: size=1, data=00
[8.190658] ERROR: Unexpected answer byte. [in src/mares_iconhd.c:219 (mares_iconhd_packet)]
[8.190662] INFO: Sleep: value=100
[8.293864] INFO: Purge: direction=1
[8.295092] INFO: Write: size=2, data=E742
[8.295128] INFO: Write: size=8, data=C02D030000010000
[8.295511] INFO: Read: size=1, data=E1
[8.295515] ERROR: Unexpected answer byte. [in src/mares_iconhd.c:219 (mares_iconhd_packet)]
[8.295518] INFO: Sleep: value=100
[8.400586] INFO: Purge: direction=1
[8.402020] INFO: Write: size=2, data=E742
[8.402065] INFO: Write: size=8, data=C02D030000010000
[8.403153] INFO: Read: size=1, data=80
[8.403157] ERROR: Unexpected answer byte. [in src/mares_iconhd.c:219 (mares_iconhd_packet)]
[8.403164] ERROR: Failed to read the dive. [in src/mares_iconhd.c:808 (mares_iconhd_device_foreach_raw)]

```

### Mentions:

jefdriesen, 2024-07-17:
Can you try again with a more recent build? There is already a fix for this problem in newer builds.

rafsalvioni, 2024-08-08:
I have a Mares Puck Pro too. This issue occured with me too. But, after newest build, seems it is fixed. I downloaded my last dives (Aug 03, 2024) correctly using version 6.0.5214

sulphur, 2024-08-13:
Indeed fixed for me as well

closed 2024-08-13 by sulphur