subsurface/subsurface #4212: Errors with online storage

nsmtbkr, 2024-05-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Sub-Surface app does not appear to push latest dive (today) to online storage.  Tried multiple times with no success.
Sub-Surface website SSL certificate has expired, requires bypassing browser insecurity warnings to log in
Sub-Surface windows app tried multiple cloud storage locations, then throws a "Failure taking cloud storage online" error.

### Operating system:

iOS for app
Windows 11 Pro

### Subsurface version:

6.0.5144

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2024-05-27:
Thanks for reporting this.

Looks like it has already been fixed: https://groups.google.com/g/subsurface-divelog/c/P3lngV_1FOc/m/gto6zFy_DwAJ

nsmtbkr, 2024-05-27:
Confirmed fixed.  Thanks!

closed 2024-05-27 by nsmtbkr