subsurface/subsurface #4215: Map is not displaying

Julaye, 2024-05-28:
### Describe the issue:

- [ X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Since few development release, the google map is not displaying.

### Operating system:
MacOS 14.5 on MacBook Air M2

### Subsurface version:
6.0.5190-CICD-Release from subsurface daily 
Problem start with the 6.0.5185-CICD-Release, perhaps before.

### Steps to reproduce:
Launch the software, map is not displaying

### Current behavior:
Google map view is blank

### Expected behavior:
Google map view on the dive site

### Additional information:
Snapshot

### Mentions:
[image]

mikeller, 2024-05-28:
@notrege: It looks like this problem was introduced when we switched to using Qt 5.15.13 in #4180. Are you able to test and see if you can get a build of Qt 5.15.13 that can build a working version of the Google maps plugin? It looks like it gets built as it displays the minimap and buttons, but it does not show the full map.

notrege, 2024-05-28:
I'm away for the week, but will take a look when I get home.

notrege, 2024-05-31:
It looks like we are building a debug version of the plugin (libqtgeoservices_googlemaps_debug.dylib) and that doesn't get deployed.    I added a potential fix for this in https://github.com/subsurface/subsurface/pull/4224, but am hoping there is a better way to do this.

PR #4224 (2024-06-01): release build google maps
After the Mac QT upgrade to 5.15.13 google maps stopped working because a debug plugin was built and not deployed.   This changes forces a release build.   It may or may not be the best alternative, but if nothing else it's a starting point for discussion with people who know more about qmake than I do.

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [x ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
This changes forces a release build for the  [... cut]
closed 2024-06-01 by mikeller