subsurface/subsurface #4220: Profile does not show graph lines for Garmin MK3i ... works for Garmin MK2i

bwong2132, 2024-05-30:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The profile (graph) does not show depth or gas pressure for my new Garmin Descent MK3i.  That profile shows depth and pressure for my older Garmin MK2i. 

### Operating system:

MacBook Air - M2 chip. 
OS - Sanoma 14.4.1
The depth and pressure does not show up on subsurface on my iPhone. 

### Subsurface version:

6.0.5144 CICD release
not sure if this issue happens on other versions. I moved to Garmin MK3i yesterday. 
I am using the official released version. 

### Steps to reproduce:

- either connect to the Garmin Mk3i with the Android File Transfer to get to the FIT files or go to connect.garmin.com and export the FIT files.  
- import the FIT files into subsurface 
- look at the profile page

### Current behavior:

the depth, pressure, temperature lines etc do not show up

### Expected behavior:

I was expecting to see all the graph lines 

### Additional information:

yes. I am attaching two FIT files. 

### Mentions:

[2024-05-29-09-22-06.txt](https://github.com/subsurface/subsurface/files/15501869/2024-05-29-09-22-06.txt)

In the file above (2024-05-29-09-22-06.txt), change ".txt" to ".fit"

I am also attaching a FIT file from my Garmin MK2i that works. 
[14364933169_ACTIVITY.txt](https://github.com/subsurface/subsurface/files/15501889/14364933169_ACTIVITY.txt)

bwong2132, 2024-05-30:
Got it -- I see the MK3i is not yet supported.

mikeller, 2024-05-30:
Update to at least v6.0.5150-CICD-release from https://subsurface-divelog.org/ to get this to work.

bwong2132, 2024-05-30:
Hi - thanks for the suggestion. I upgraded to 6.0.5194-CICD-release. Deleted the MK3i dives. Then tried to import in the MK3i dive logs. When I clicked on the Import from Dive Computer, I got a red bar at the bottom with the following message "unlikely dive gas data from libdivecomputer: o2 = 0.000 he=0.0000. 

When I clicked on the download, I see my new FIT files. When I clicked on OK (to download), I got a spinning color wheel. After a while, I had to force stop subsurface. 

Bernie

bwong2132, 2024-05-30:
Wait - what the heck. I tried uploading the MK3i files again. This time, it worked. However, I still got the "unlikely dive gas" warning. Awesome. Thanks.

mikeller, 2024-06-02:
Should be fixed in Subsurface >= 6.0.5203, available from https://www.subsurface-divelog.org/latest-release/.