subsurface/subsurface #4221: Profile for Garmin MK3i shows 3 graph lines for gas pressure

bwong2132, 2024-05-30:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I imported the FIT files from my new Garmin MK3i into subsurface. In the profile (graph), I see three lines for gas pressure. In my previous dives (with the MK2i), I see only one graph line for gas pressure.  I am sure it must be related --- at the bottom of the page, I see an error message saying "unlikely dive gas data from libdivecomputer: o2=0.000 he=0.000"

See below 
[image]

### Operating system:

macOS Sonoma 14.4.1

MacBook Air M2

not sure

### Subsurface version:

6.0.5194-CICD-Release

on an earlier version, the profile does not show the gas pressure, depth, etc. on the graph

yes

### Steps to reproduce:

Import in a MK3i FIT log, look at the profile 

### Current behavior:

got warning saying unlikely gas data and profile shows 3 different lines for gas pressure

### Expected behavior:

no error message and only one line for gas pressure

### Additional information:

[2024-05-29-09-22-06.txt](https://github.com/subsurface/subsurface/files/15505487/2024-05-29-09-22-06.txt)
please change extension from txt to fit

none

### Mentions:

mikeller, 2024-05-30:
Hi Bernie.

Thanks for supplying the .fit file, I will have a look into what is going on and report back.

mikeller, 2024-05-30:
Looks like you are using an unusual setup that is somewhat of an edge case:
Your dive log shows that you had 3 tank sensors reporting, but you had only 1 gasmix configured on your dive computer. So I guess that you either dived with 3 tanks containing air, or you used extra tank sensors to monitor the remaining pressure of students or team mates.

Anyhow, the unintuitive warning was due to a shortcoming in the Garmin importer, which is fixed in https://github.com/subsurface/libdc/pull/64.
You will still get a warning telling you that there are more sensors than gasmixes though, but this is an appropriate observation now.

bwong2132, 2024-05-31:
OMG - you are right. I set up my MK3i with two other T2 transmitters - one for my wife and one for my daughter. I am sorry I forgot about it. In this case, I wonder if subsurface could color "my tank" with one color and the "other or guest" tanks with another color. That would be amazing. Or just graph "my tank".  But you are right -- I could live with the three graph lines. 

Thanks so much for checking!!!!!!!!!!!!!.

mikeller, 2024-05-31:
> I wonder if subsurface could color "my tank" with one color and the "other or guest" tanks with another color.

Good idea - I have submitted #4222 which will allow you to enter tank descriptions to set 'names' for the tanks that will show up in the profile.

closed 2024-06-02 by mikeller