subsurface/subsurface #4238: Dive Download Missing From Preferences

jntawel, 2024-06-06:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

In the latest dev release (6.0.5209-CICD-release) the "dive download" tab for the preferences no longer exists and the option to delete dive computers no longer exists either
### Operating system:

Windows 11

### Subsurface version:

6..0.5209-CICD-release

No, that feature still exists in the current stable release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2024-06-06:
This is by design - 'remembered' dive computers can now be deleted individually, directly in the 'Download from dive computer' dialog (the little 'thrash can' icon):

[image]

So there is no more need to manage this through the preferences.

mikeller, 2026-10-17:
Hi @jntawel,

This was intentional: the ability to delete individually remembered dive computers moved out of Preferences and into the "Download from dive computer" dialog itself (the trash-can icon next to each entry), so there's no need to manage it through Preferences any more.

Closing this as a confirmed relocation, not a regression. Please reopen if you can't find a way to remove a remembered dive computer in a current build.

closed 2026-10-17 by mikeller