subsurface/subsurface #4239: Imperial Tank Sizes broken after Crash

Patrick-BS, 2024-06-06:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After a crash in the Planner, the tanksizes are broken. When adding a tank in planner or to a dive there is the complete volume (Size x Pressure) shown in liters. 

### Operating system:

Win10

### Subsurface version:

6051.44

### Steps to reproduce:

No Idea...

### Current behavior:

Choose AL80 and it will add the follwing data to the table: AL80 | 2.265,3l | 207bar. This happens to all imerial sizes.

### Expected behavior:

Choose AL80 and it will add the follwing data to the table: AL80 | 11,1l | 207bar.

### Additional information:

I think the prefrences / config file is broken. But uninstalling, deleting the appdata folders and reinstall did not help.

### Mentions:

mikeller, 2024-06-07:
Looks like your preferences are set to a mix of metric and imperial - have you tried opening Preferences / Units and selecting 'metric'?

[image]

Julaye, 2024-06-07:
I had the same issue with ALxx tanks after a crash in the planner. I need to change manually the size of the tank.

I'm only in metrics

[image]

[image]

mikeller, 2024-06-08:
@Julaye: Can you reproduce the crash? It's hard to figure out what is causing the corruption without being able to reproduce.

Julaye, 2024-06-08:
Since the crash who seems to have caused the corruption on some tank volumes, the planification mode is very stable and I can't reproduce the crash :( sorry. 

I try again this morning playing with the planification mode but without success to crash it.

Anyway, how can I restore the tanks volume to initial data ? perhaps will help me to reproduce the crash.

mikeller, 2024-06-08:
Wait, you got the '1415.8l' tank volume despite having 'metric' selected in the preferences - that is very odd indeed. But the good news is that I found a reproduction - I'll be having a look to see what's going on.

Patrick-BS, 2024-06-08:
Yeah thats the whole point of it. I am using it all in metric. But since the AL80 is now set to 2.265,3l instead of 11.1 it calculates 2.265,3l x 207 bar = *way too much air* :)

Patrick-BS, 2024-06-08:
Where are the tank parameters stored? Maybe it wold be a quick fix to delete the properties and reinstall subsurface.

mikeller, 2024-06-08:
> Where are the tank parameters stored? Maybe it would be a quick fix to delete the properties and reinstall subsurface.

They are stored in the dive log. But for what I can see (without deeper investigation) the ridiculous sizes are calculated on the fly due to a bug, but not actually stored (until you use one of these tanks in one of your dive plans and store it).

Patrick-BS, 2024-06-08:
Hmm, i am quiete sure that i have this problem with any "imperial style cylinder" i never used , but i will check.

mikeller, 2024-06-08:
> Hmm, i am quiete sure that i have this problem with any "imperial style cylinder" i never used , but i will check.

If this is the case, would you be willing to share your dive log with me? You can send it to github@ike.ch.

commit 10fc3bfd (2024-06-09): Bugfix: Fix Incorrect Volumes Displayed for Tank Types.

Fix an issue introduced in #4148.
Essentially the refactoring missed the fact that in the imperial system
tank size is tracked as the free gas volume, but in the metric system
(which is the one used in most of Subsurface's calculations) tank size
is tracked as water capacity.
So when updating a tank template tracking imperial measurements, the
given (metric) volume in l has to be multiplied by the working pressure,
and vice versa.
This also combines all the logic dealing with `tank_info` data in one
place, hopefully making it less likely [... cut]
closed 2024-06-09 by bstoeger