subsurface/subsurface #4240: GF Low/High will reset to default value in Plan dive after restarting Subsurface

cchsieh, 2024-06-07:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I plan a dive, set the GF low/high which is different from the default value in Preferences. 
I save everything, and close the Subsurface. 
Then, I open Subsurface again, and choose "Edit dive in planner", and the GF low/high reset to the default value.

### Operating system:

Windows 11 Chinese Tranditional

### Subsurface version:

6.0.5204-CICD-release

don't know

official release

### Steps to reproduce:

### Current behavior:

GF low/high will reset to defalut value in "Edit dive in planer" after restarting the Subsurface program.

### Expected behavior:

The GF low/high would be the one I set, not default value.

### Additional information:

### Mentions:

mikeller, 2024-06-07:
In Subsurface there are two different gradient factor settings, and it's important to keep in mind where they are used:

1. there is one gradient factor setting in the Preferences / Tech setup. This is used to calculate the calculated ceiling that is shown in the dive profile when NOT in the planner:
[image]

2. there also is the gradient factor that can be set on the planner page, and is used to calculate the decompression schedule when planning a dive. But this is not persisted with the dive when it is saved, so when looking at a planned dive in the dive list, the setting in 1. is applied for the calculated ceiling.

cchsieh, 2024-06-07:
Thanks for your replying so quickly.

My preferences:
[image]

I create a new Plan dive, and modify GF low/high and save.
[image]

Than I reopen it by "Edit dive in planner" without exit Subsurface progrm.
[image]

The setting of GF low/high is the same as what I modified last time.
[image]

I shutdown/exit Subsurface program fully and start Subsurface again, and reopen the dive plan by "Edit dive in planner". The setting of GF low/high will change to the default value in preference.
[image]

It is weired for me if I planed and saved a dive last night and open it this morning but the setting is changed without any warning. 
 
Sorry for my bad English.

cchsieh, 2024-06-07:
BTW, "water type" has the same behavior. I set "water type" to "Sea water" and save the plan. After I exit Subsurface and then re-open it, the water type setting will change back to "Fresh water"

mikeller, 2024-06-08:
Yes, as I said in my first response, the gradient factor used for planning is NOT saved with the dive - the same as all the other planner settings that are not stored either. So when you re-edit the dive in planner, the gradient factor that is configured in planner is used. And this is read from the preferences at startup.
The fact that the water type is not stored when saving a dive in planner seems to be a bug though.

cchsieh, 2024-06-08:
I see. Thanks.

PR #4243 (2024-06-10): Desktop: Fix Inconsistencies in Handling of Salinity.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

1. add correct setting of the water type drop down for the dive shown initially after program start;
2. change salinity to have 3 decimals in planner, to make it consistency with the log.

### Related issues:

Fixes #4240.

### Additional information:

[image]

### Release note:
commit b126fccb (2024-06-10): Desktop: Fix Inconsistencies in Handling of Salinity.

- add correct setting of the water type drop down for the dive shown
  initially after program start;
- change salinity to have 3 decimals in planner, to make it consistency
  with the log.

Fixes #4240.

Reported-by: @ccsieh
closed 2024-06-10 by mikeller