subsurface/subsurface #4249: Not responding on start

remillc, 2024-06-15:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface doesn't start. It stays in a "not responding" state. The UI has not loaded.

This is not a fresh install, so it tries to open my local dive log, a 46 Mo file that contains 1300 dives.

### Operating system:

Windows version 23H2 (22631.3737) french

### Subsurface version:
Windows version installed with the official installer.

\> subsurface.exe --version
> Subsurface v6.0.5144-CICD-release,
> built with libdivecomputer v0.9.0-devel-Subsurface-NG (62a29eea15137ba7b0f2e10fae095517cc9d8341)
> built with Qt Version 5.15.12, runtime from Qt Version 5.15.12
> built with libgit2 1.1.0

### Steps to reproduce:
Don't know. The problem doesn't happen all the time.

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2024-06-16:
Can you please try starting Subsurface from the command line with:

```
"\Program Files\Subsurface\subsurface.exe" -v -v -v
```

and post the output in the console here?

remillc, 2024-06-16:
Thanks for replying Michael!

There is the output. I replaced some infos that looks personnal with "***". I can send you the original by another mean if needed.

```
Subsurface v6.0.5144-CICD-release,
built with libdivecomputer v0.9.0-devel-Subsurface-NG (62a29eea15137ba7b0f2e10fae095517cc9d8341)
built with Qt Version 5.15.12, runtime from Qt Version 5.15.12
built with libgit2 1.1.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 4.6."
loading "en-US" translations
QObject::connect(QQuickWindow, QDeclarativeGeoMap_QML_6): invalid nullptr parameter
DivePixmaps DPR: 1.000000 metrics: 24 16 sz_bigger: 40
loading dive data from ("F:/***/***/Mes documents/plongée/plongees.ssrf")

File locations:

Cloud email:christian.remillard@gmail.com
returning cloud URL "https://ssrf-cloud-us.subsurface-divelog.org/git/christian.remillard@gmail.com[christian.remillard@gmail.com]"
Local git storage: C:\Users\chris\AppData\Roaming\Subsurface/cloudstorage/e21702540e7f9d7f
Cloud URL: https://ssrf-cloud-us.subsurface-divelog.org/git/christian.remillard@gmail.com[christian.remillard@gmail.com]
Image filename table: C:\Users\chris\AppData\Roaming\Subsurface/hashes
IP used for cloud server access "184.163.73.248"
```

mikeller, 2024-06-17:
This looks like Subsurface is getting stuck when trying to access your log in cloud storage. Is there any kind of restrictive 'personal firewall' or security software running on this PC?

Can you try running Subsurface with the command line parameters `-v -v -v --user=test` (i.e. starting with a different, new user profile 'test')? This will mean Subsurface does not try to synchronise from the cloud on startup and should start fine.

remillc, 2024-06-17:
There is nothing special about the local network, no firewall or proxy settings. Actually, the configurations have not changed for a couple years. No new internet provider, switch or router either. No kind of security software either on the computer.

I recall I managed to open Subsurface after the problem occured for the first time. In fact, I use Subsurface every week after a dive to download the logs from my Petrel 3.

What changed though since I first witnessed the problem, is that I've had a very hard time getting the logs of the Petrel 3 from my Windows installation of Subsurface. So I started downloading it from my Android phone using the mobile version of Subsurface, and using Subsurface's cloud storage, I then open the log file on my Windows computer to complete log infos.

Today, I opened Subsurface and it started normally with an empty log. The last time I tried was with your instructions:

```
"\Program Files\Subsurface\subsurface.exe" -v -v -v
```

Once opened, I went to `File > Open cloud storage` and got my dives back. I can still do the test with a new user if you want. I'll pay attention next time I use my phone to get the dive logs to see if the problem occurs.

mikeller, 2024-06-18:
Sounds like this is working for you now, which this is the main thing.

Not sure why this stopped working temporarily, but if it is working now then we are probably not going to be able to reconstruct why it had stopped working.

remillc, 2024-06-18:
Yes. If the problem raises again, I'll get back to you.

Thanks for the assistance @mikeller.

remillc, 2024-08-08:
Ok, it did it again.

Starting Subsurface from the Windows start menu. A Subsurface window opens without the user interface, as it is initializing, but nothing else happens.

I killed the process, then ran the following command:

```
subsurface.exe -v -v -v --user=test
```
Subsurface started with a fresh new user. Here is what the console displayed (obfuscated the IP address):

```
Subsurface v6.0.5144-CICD-release,
built with libdivecomputer v0.9.0-devel-Subsurface-NG (62a29eea15137ba7b0f2e10fae095517cc9d8341)
built with Qt Version 5.15.12, runtime from Qt Version 5.15.12
built with libgit2 1.1.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 4.6."
using custom config for "Subsurface-test"
loading "fr_FR" translations
can't find Subsurface localization for locale "fr-CA"
QObject::connect(QQuickWindow, QDeclarativeGeoMap_QML_5): invalid nullptr parameter
DivePixmaps DPR: 1.000000 metrics: 24 16 sz_bigger: 40
loading dive data from ("C:\\Users\\chris\\AppData\\Roaming\\Subsurface\\chris.xml")

File locations:

Cloud email:
Unable to get local git directory
Cloud URL: No valid cloud credentials set.

Image filename table: C:\Users\chris\AppData\Roaming\Subsurface/hashes
IP used for cloud server access "xxx.xxx.xxx.xxx"
remember cloud server https://ssrf-cloud-us.subsurface-divelog.org/ based on IP location in  "North America\n"
```
I closed Subsurface, then started it again from the Windows menu and it started normally with my dive log.

So I don't need support since I can manage to use Subsurface, but am available if you want to run some tests. Else, I'll close this issue.

mikeller, 2026-10-17:
Hi @remillc,

Thanks for working through this with us. Since you mentioned you no longer need support and can manage day-to-day, and the hang didn't recur in a way we could pin down (it seemed tied to cloud-sync at startup per the verbose logs, but came and went without a clear trigger), we're closing this as not reliably reproducible. If it happens again, --user=test (or otherwise skipping cloud sync at startup) is worth trying as a quick workaround/diagnostic.

Please reopen with a fresh verbose log if the "not responding" hang comes back.

closed 2026-10-17 by mikeller