subsurface/subsurface #4251: visibility='3' added, when unsig ctrl-a in filtered view

kruegerha, 2024-06-18:
When a filtered subset of dives is displayed, and they are all selected with ctrl-a, visibility='3' is added to all of the selected dives, which did not have the visibility defined  before...

### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

31fca0f67

### Steps to reproduce:

1) filter a subset of dives, of which some have no visibility defined in the <dive ...> section
2) select all of them using cntrl-a
3) save and compare logbook

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2024-06-18:
I can not reproduce this with my test data set - the .ssrf file has the same number of occurrences of `visibility` before and after executing the steps described by you.
Are you able to create and share a .ssrf file that shows this problem?

kruegerha, 2024-06-18:
I prepared a stripped xml file (attached) which shows the problem.
1)  change to filtered view (filter on tag "test")
2)  ctrl-a does not work with this file... however, if you select all of the trips, in some of them the changes occur
  affects dive 21 (visibility added) and 28 (visibility changed?)

Hope this helps to identify the problem

[testdat-prob.xml.gz](https://github.com/user-attachments/files/15887450/testdat-prob.xml.gz)

kruegerha, 2024-06-18:
reminds me on  #3960. behavior seems somehow similar...

kruegerha, 2025-01-09:
just tried again and cannot reproduce anymore. Maybe the problem disappeared, or I did something weird before.

I guess this bug can be closed

closed 2025-01-09 by kruegerha