subsurface/subsurface #4255: Cylinder pressure tags for printing are not working correctly (and some more)

Patrick-BS, 2024-06-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When using the printing Tags, the cylinder pressures are not exported, when stored via tanksensor data.

### Operating system:

Win 10, German
### Subsurface version:

6.0.5214
### Steps to reproduce:

Store a dive with tank sensor data to the logbook. The gas tank start and end pressures will be prefilled (italic letters) with the data logged via the sensor. Print it via the following html loop with tags:

```
{% for cylinder in cylinderObjects %}
<tr>
<td class="fieldTitle">
     <h1>{{ cylinder.description }}</h1>
</td>
<td style="width:75%" class="fieldData">
      <p>{{ cylinder.gasMix }} ({{ cylinder.size }} / {{ cylinder.startPressure }} ⇒ {{ cylinder.endPressure }})</p>
</td>
</tr>
{% endfor %}
```

### Current behavior:

It will show 0bar -> 0bar in the preview and on the printed page

### Expected behavior:

It should show the same values as in the tank list.

### Additional information:

When using the "dive.gas" tag, the pressure data is exported, but for technical dives i need to handle that separately. See Screenshot.
The "dive.salinity", "dive.waterType", "dive.weights" Tags are not working too....

[image]
[image]

### Mentions:

Patrick-BS, 2024-11-04:
Not relevant? xD

mikeller, 2024-11-04:
> Not relevant? xD

More like 'none of the usual contributors in the project felt an irresistible urge yet to work on a fix for this'.

Feel free to develop a fix and open a pull request for it.  😉