subsurface/subsurface #4256: Aqualung i100 Dataimport

nsflatspin, 2024-06-20:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Dataimport of Aqualung i100 shows wrong Data, as I never dives 73 m deep ;-)
Failure notice appaers: unlikely dive gas data from libdivecomputer: o2 = 2.200 he = 0.000
Date ist wrong (eg. 24.07.2025) / depth is wrong / Watertemperature ist wrong / ....

### Operating system:

### Subsurface version:

Subsurface Version 6.0.5214-CICD-release for Windows
Downloaded from https://subsurface-divelog.org/current-release/?lang=de_DE

### Steps to reproduce:

1) install Subsurface
2) Import Data from Divecomputer
3) select "Aqualung / i100 / COM3

### Current behavior:

All data semm to be wrong imported

### Expected behavior:

Import data correct

### Additional information:

### Mentions:

[Data.zip](https://github.com/user-attachments/files/15917159/Data.zip)

jefdriesen, 2024-07-17:
Can you download with both the "Save libdivecomputer dumpfile" and "Save libdivecomputer log file" checkboxes enabled and attach those files to the bug report?

jefdriesen, 2024-07-18:
It looks like you have a new variant of the i100 that requires some changes to the code. I'll take care of that. Are you able to download your dives into the official Aqualung application and then export the data into some human readable format (csv, xml, txt, ...)? Have something to compare against would make my work a bit easier.

nsflatspin, 2024-07-19:
Hi Jeff, thanks for taking care of this issue! I have only diverlog lite. It's the free version, which have no option to export data. But I can give you some examples  of dives in detail. Here are Excel-files of dive 5 and dive 6. This because the first 4 dive were manually added and not importet from the i100. Please tell me, if this helps. If you need more, I'm happy to do it.
[DiveDetailsi100_Dive5.xlsx](http
[DiveDetailsi100_Dive6.xlsx](https://github.com/user-attachments/files/16309300/DiveDetailsi100_Dive6.xlsx)
s://github.com/user-attachments/files/16309299/DiveDetailsi100_Dive5.xlsx)

jefdriesen, 2024-07-22:
Can you carefully check the attached file for errors? Rename the .txt extension to .ssrf and open the file in subsurface.
[i100.txt](https://github.com/user-attachments/files/16334641/i100.txt)

nsflatspin, 2024-07-28:
Hi Jef, sorry for late response. Was on holiday for some days. :-)) I checked the imported data from your file. The only difference I noticed, is in the duration of the dive. The dive on 2023-11-03 lasted 49 min, but with your file it lasted 51 min. The dive on 2023-11-04 lasted 59 min, but with your file it lasted 1 hour. All dives have a little difference. I send you a cvs-file from this 2 dives. Not all data, but the duration and depth, I can eyport. Everything ales seems to be good. Thank you very much for you refforts!
[Dive_2023_11_03.csv](https://github.com/user-attachments/files/16404144/Dive_2023_11_03.csv)
[Dive_2023_11_04.csv](https://github.com/user-attachments/files/16404145/Dive_2023_11_04.csv)

jefdriesen, 2024-07-31:
Very small differences in the dive time are indeed possible. Looks good, so I pushed the fix. You'll have to wait until it gets integrated into subsurface.

nsflatspin, 2024-08-01:
Thank you very much again! Will wait until new version and than use it, I think! Great peace of software!

jefdriesen, 2024-08-02:
You can try this build: https://github.com/jefdriesen/subsurface/actions/runs/10158382908

nsflatspin, 2024-08-02:
Installed your build and everything seems to be ok. 👍 Thank you very much!

mikeller, 2026-10-17:
Hi @nsflatspin,

Thanks for all the back-and-forth testing on this, it made it possible to track down. As you confirmed above after testing jefdriesen's CI build: "Installed your build and everything seems to be ok." That fix has long since been part of standard releases.

Closing this as fixed and confirmed working. Please reopen if you still see wrong depth/date/temperature importing from your i100 on a current release.

closed 2026-10-17 by mikeller