subsurface/subsurface #4258: Connection with Sunnto D5 from bluetooth

milhaushk1981, 2024-06-24:
Hi, I have a problem connecting to SubSurface on my iPhone 12 and Sunnto D5. When I try to connect, I get the error Import error: Input/output error. and I have no device in the Connection field, but I am connected to sunnto via bluetooth. I can see that in the phone settings. Does anyone have experience?

### Operating system:
IoS 17.5.1 - iPhone 12

### Subsurface version:
6.0.5209

MarkusAndrea, 2024-09-20:
Same problem. It works with Subsurface on Mac via cable.

mikeller, 2026-10-17:
Hi @milhaushk1981,

Thanks for reporting the Bluetooth connection issue with your Suunto D5. To dig into this we'd need a log capture of the failed connection attempt, enabling verbose logging in Subsurface's preferences before connecting and sharing the output would help us see exactly where it's failing.

If we don't get a log within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.