subsurface/subsurface #4261: Not able to delete useless

OlivierV78, 2024-07-01:
### Describe the issue:

- [ ] Bug
- [ X] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Not possible to delete useless gas lines.

### Operating system:

Kubuntu 20.04, French

### Subsurface version:

6.0.5214

yes

Officiel

### Current behavior:

Not possible to delete some gas lines (see screenshot)

### Expected behavior:

Delete every gas lines if necessary

### Additional information:

### Mentions:

[image]

[image]

atdotde, 2024-07-02:
Subsurface prevents you from deleting cylinders that it thinks you are breathing from during the dive. I assume this is a CCR dive. But I cannot see from the screen shots which gases are used during which part of the dive (I guess Subsurface thinks you rebreather uses the third gas as diluent and of course the O2 as well). That's why it offers your to delete the first gas.

mikeller, 2024-07-02:
@OlivierV78: Subsurface normally prevents gas mixes that are reported as used during the dive from being deleted, as these are needed in order to calculate the tissue loading and compression obligation.
What is odd is that in your case the log seems to be showing a dive where only a single gas (presumably a diluent in a rebreather) was used, yet two gases are marked as non-deletable.
And even more oddly one of these gases is the 100% oxygen feed gas for the rebreather - which is normally not tracked at all by the dive computer.
What is the configuration of your dive computer, is it standalone, or connected to the CCR? If so, analogue or via DiveCAN? And what model CCR is it?
Are you able to enable the 'Use' and 'Sensor' columns in the gas list by right clicking on one of the column headings and selecting these, and then sharing another screenshot?
And would you be willing to share the log file for one of the dives in question?

OlivierV78, 2024-07-02:
@mikeller My rebreather is a Revo with a Petrel 3 in divecan and a perdix backup with 2 transmitters (1 on the diluent and the other on the oxygen). I import the dives from the 2 computers. Sometimes the shearwater transmitters (old non swift models) are not picked up by the 2 computers but only by 1. This is why I only want to keep the one that has been read in this case. I have to specify the gas manually because subsurface defaults to my bail-out gas (33%) whereas the probes are on oxygen and diluent.

Attached is the requested capture and the log file of this dive.

In attachment a new import from today with **no manual change**

[image]

[logfile.txt](https://github.com/user-attachments/files/16071561/logfile.txt)

[image]

[logfile2.txt](https://github.com/user-attachments/files/16071606/logfile2.txt)

mikeller, 2024-07-03:
Ah, this makes sense then. There are two shortcomings in how Subsurface handles gases at the moment:
- when tank pressure sensors are used, Subsurface does a 1 : 1 mapping of the list of gasmixes and the list of tank sensors reported by the dive computer. This is not necessarily correct, as the dive computer treats gas mixes and tank sensors as completely independent things;
- when more than one dive computers are used, then Subsurface just concatenates the two gas lists, and does not try to merge identical gases from both computers.
Unfortunately both of these apply to your case, and there is no easy way to fix this in the application, as adding a way to define and maintain a custom mapping between gas mixes and tank sensors will require a UI for this, and also need a change in the formats that dive logs are stored in.

OlivierV78, 2024-07-03:
@mikeller Thank you for the explanations. So I will continue to correct by hand

tdhammer, 2024-08-21:
@mikeller -

RE:
> when more than one dive computers are used, then Subsurface just concatenates the two gas lists, and does not try to merge identical gases from both computers.
This seems to be true also when combining a dive plan with the actual dive uploaded from the computer, correct?

Since I do not have an air-integrated computer, would I be better off not adding a cylinder to the dive created by the upload and then after merging modify the cylinder with the actual start & end pressures (replacing what the planning filled in)?

mikeller, 2026-10-17:
Hi @OlivierV78,

Thanks for the detailed description of your Revo/Petrel 3/Perdix setup. This traces back to the same gasmix-to-tank-sensor mapping gap as GH#4291 (compounded here because your probes are on diluent/oxygen while Subsurface defaults to interpreting readings against the bailout gas), plus a second limitation: when a dive imports from more than one dive computer, Subsurface concatenates the two gas lists rather than merging duplicates, so gases from a secondary/redundant DC source can't be cleanly removed. Fixing this needs a UI for a user-defined gas/sensor mapping plus a format change, not something we can do as a quick fix.

Closing this citing those two compounding architectural gaps, with manually editing the gas list/cylinder data after import as the current workaround. Please reopen if you hit a case that workaround doesn't resolve.

closed 2026-10-17 by mikeller