subsurface/subsurface #4264: Import from ProPlus X has faulty data despite log being correct

knottpak, 2024-07-09:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I had 2 dives on my Oceanic ProPlus X; first showed correct depth/time (9ft/10min) but second shows 165' depth. The log on the computer shows the correct information. 

### Operating system:

Windows 11 computer with ProPlus X and downloading dives by BT on iPhone. 
### Subsurface version:

iPhone subsurface version 6.0.5209/windows 6.0.5144
### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

[knott error log 2x dive.zip](https://github.com/user-attachments/files/16135719/knott.error.log.2x.dive.zip)

jefdriesen, 2024-07-17:
Can you download with both the "Save libdivecomputer dumpfile" and  "Save libdivecomputer log file" checkboxes enabled and attach those files to the bug report?

knottpak, 2024-07-18:
Unfortunately, I haven't been able to get subsurface to be able to download the proplus X data on my virtual windows 11 machine (the only windows machine I have access to ATM. I usually import the dives on my phone, then they are saved to cloud, where I edit them. That's been my workaround. But I did find that when I downloaded the dives with diver log lite (official oceanic software) the dive shows a max depth of 100.6M there too (incorrect), so it appears the data is coming either from the computer or the download. The max depth on the log on the ProPlus X shows correct (9 ft).

knottpak, 2024-07-18:
I will be able to download with the requested files when I am back to my normal windows machine in a few weeks.

mikeller, 2026-10-17:
Hi @knottpak,

Thanks for reporting this. To look into why the imported data doesn't match your Pro Plus X's own log, we'd need two things: which specific fields look wrong (depth, time, temperature, gas, etc.) and a sample of the raw dive file/log so we can compare it directly against what Subsurface parsed.

If we don't hear back with those details within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.