subsurface/subsurface #4275: UDDF Export: when select "all dives", exported file is empty

rafsalvioni, 2024-08-07:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When export dives to UDDF format and "all files" selecttion checked, result file is empty

### Operating system:
Windows 8

### Subsurface version:
6.0.5214

### Steps to reproduce:
In a opened logbook:
1) Go to File -> Export (Ctrl+E);
2) Select "UDDF" in export format;
3) Select "all dives" in selection box;
4) Press "OK" and choose file location

### Current behavior:
Exported file is empty

### Expected behavior:
Exported file contents should be whole logbook in UDDF format

### Additional information:

### Mentions:

rafsalvioni, 2024-08-07:
I try to analyze the code, but I didnt found the issue... I think that is in XSLT transformation

mikeller, 2024-08-10:
I had a quick check, and with my test logs the UDDF exported just fine when 'all dives' is checked.
So I suspect that some data that you have in at least one of the dives in your log is causing an error and abort during the XSLT transformation.
Did you try running Subsurface with -v -v -v command line arguments and checking if there is anything in the log hinting at where the transformation is failing?

rafsalvioni, 2024-08-10:
@mikeller Log shows a error in XSLT:

[image]

I run Subsurface 6.0.5230 on Linux

mikeller, 2024-08-10:
@rafsalvioni: Can you please test if the artifacts in #4278 fix this?

rafsalvioni, 2024-08-11:
> @rafsalvioni: Can you please test if the artifacts in #4278 fix this?

Tested. Fix OK

PR #4278 (2024-08-11): Export: Fix UDDF Export

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Add a workaround to address a syntax error in the XSLT used to export as UDDF.

### Changes made:

### Related issues:

Fixes #4275.

### Additional information:

### Release note:

### Documentation change:

<!
commit 38a0050a (2024-08-11): Export: Fix UDDF Export

Add a workaround to address a syntax error in the XSLT used to
export as UDDF.

Fixes #4275.
closed 2024-08-11 by mikeller