subsurface/subsurface #4281: Dive site flags move when resizing map vertically

pjsanfil, 2024-08-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When resizing the map portion of the main window in the vertical direction the dive flags indicating dive site location move, relative to the map. Problem occurs either shrinking or enlarging the map area.

### Operating system:

Ubuntu 20.04.6 LTS
US English

Acer Spin Laptop

### Subsurface version:

Subsurface 6.0.5214-local

Official release from repo.

### Steps to reproduce:

1. Start subsurface
2. Open cloud storage
3. Zoom in on some dive site(s)
4. Grab the horizontal bar dividing the map from the graph above it, drag the divider bar up giving the map more room.

### Current behavior:

When resizing the map vertically the dive site flag location moves relative to the map background.
When resizing the map in the horizontal direction the flags move as expected, staying in the same geographic location.

### Expected behavior:

The dive site flag should maintain it's position relative to the map. 

### Additional information:

### Mentions:

pjsanfil, 2024-08-13:
Cool thanks, I'm interested in fixing this issue myself then if that's OK.

mikeller, 2024-08-14:
> Cool thanks, I'm interested in fixing this issue myself then if that's OK.

Of course, contributions to our project are always welcome!

Sing out if you need assistance, we are happy to help.