subsurface/subsurface #4285: [macOS] version 6.0.5231 requires macOS 12, although FAQ says it is compatible with macOS 10.15+

core-code, 2024-08-14:
[image]

dirkhh, 2024-08-14:
interesting. I don't have a machine that's running an older version than macOS 12 in order to test this with.
But really - support for 10.15 (from Apple) ended more than 4 years ago, so maybe we should just acknowledge that versions older than macOS 12 can't be supported anymore (my guess is that it's something in the build chain that forces that version, but I'll need to go through all of the shared objects to figure out what exactly causes this.

mikeller, 2024-08-15:
@core-code: Where is this information shown?
@dirkhh: Not sure if this is worth spending much time on, besides updating of the FAQ to remove mentions of OS versions that are unsupported by their manufacturer.

core-code, 2024-08-15:
> But really - support for 10.15 (from Apple) ended more than 4 years ago,

agreed. this bug report is not about 10.15 && 11.0 not being supported anymore, it is about the documentation and reality being out of sync

> @core-code: Where is this information shown?

you don't see the info without third party software (Apparency in this case), but putting the app on any 10.15 or 11.0 Mac will show it with a "disabled" icon and when trying to launch it there will be a notice that 12.3 is required

core-code, 2024-08-15:
```% otool -l  Subsurface-6.0.5231-CICD-release_folder/Subsurface.app/Contents/MacOS/Subsurface
Subsurface-6.0.5231-CICD-release_folder/Subsurface.app/Contents/MacOS/Subsurface:
[…]
Load command 10
      cmd LC_BUILD_VERSION
  cmdsize 32
 platform 1
    minos 12.3
      sdk 12.3
   ntools 1
     tool 3
  version 820.1
[…]```

dirkhh, 2024-09-09:
I think I managed to build a DMG that should work with 10.15 (and, for extra credit, even support printing):
https://subsurface-divelog.org/downloads/Subsurface-6.0.5252-10-15.dmg
I'd love if you could test that (assuming you have access to a macOS 10.15 system).