subsurface/subsurface #4286: SEGV when Synchronising with the Cloud

mikeller, 2024-08-16:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface dies with a SEGV when synchronising with the cloud:

```
Thread 1 "subsurface" received signal SIGSEGV, Segmentation fault.
0x00007ffff5456d45 in __GI___libc_free (mem=0x1400000000)
    at ./malloc/malloc.c:3375
warning: 3375	./malloc/malloc.c: No such file or directory
(gdb) bt
#0  0x00007ffff5456d45 in __GI___libc_free (mem=0x1400000000)
    at ./malloc/malloc.c:3375
#1  0x00005555558e9f1b in std::__new_allocator<event>::destroy<event>
    (__p=<optimized out>, this=0x555558c73d18)
    at /usr/include/c++/13/bits/new_allocator.h:196
#2  std::allocator_traits<std::allocator<event> >::destroy<event>
    (__p=<optimized out>, __a=...)
    at /usr/include/c++/13/bits/alloc_traits.h:558
#3  std::vector<event, std::allocator<event> >::_M_erase (__position=Python Exception <class 'gdb.error'>: value has been optimized out

   , this=0x555558c73d18)
    at /usr/include/c++/13/bits/vector.tcc:187
#4  std::vector<event, std::allocator<event> >::erase (__position=Python Exception <class 'gdb.error'>: value has been optimized out

   , this=0x555558c73d18)
    at /usr/include/c++/13/bits/stl_vector.h:1535
#5  fixup_dc_events (dc=...) at /home/mikeller/git/subsurface/core/dive.cpp:801
#6  fixup_dive_dc (dc=..., dive=<optimized out>)
    at /home/mikeller/git/subsurface/core/dive.cpp:1106
#7  dive::fixup_no_cylinder (this=this@entry=0x555558c73ac0)
    at /home/mikeller/git/subsurface/core/dive.cpp:1129
#8  0x00005555558f47c8 in dive_table::fixup_dive
    (this=0x555555ce0ac0 <divelog>, dive=...)
    at /home/mikeller/git/subsurface/core/divelist.cpp:36
#9  0x00005555558f5e38 in dive_table::record_dive
--Type <RET> for more, q to quit, c to continue without paging--c
    (this=0x555555ce0ac0 <divelog>, d=std::unique_ptr<dive> = {...})
    at /usr/include/c++/13/bits/unique_ptr.h:199
#10 0x000055555593bbef in finish_active_dive (state=state@entry=0x7fffffffd1c0)
    at /usr/include/c++/13/bits/unique_ptr.h:197
#11 0x00005555559428da in dive_directory
    (state=0x7fffffffd1c0, timeoff=<optimized out>, name=<optimized out>, entry=0x555558c738d0, root=0x555557acbce0 "2013/10/")
    at /home/mikeller/git/subsurface/core/load-git.cpp:1514
#12 walk_tree_directory
    (state=0x7fffffffd1c0, entry=0x555558c738d0, root=0x555557acbce0 "2013/10/") at /home/mikeller/git/subsurface/core/load-git.cpp:1616
#13 walk_tree_cb
    (root=0x555557acbce0 "2013/10/", entry=0x555558c738d0, payload=0x7fffffffd1c0) at /home/mikeller/git/subsurface/core/load-git.cpp:1818
#14 0x00007ffff7f55571 in ??? () at /lib/x86_64-linux-gnu/libgit2.so.1.7
#15 0x00007ffff7f5569a in ??? () at /lib/x86_64-linux-gnu/libgit2.so.1.7
#16 0x00007ffff7f5569a in ??? () at /lib/x86_64-linux-gnu/libgit2.so.1.7
#17 0x00007ffff7f557b4 in git_tree_walk ()
    at /lib/x86_64-linux-gnu/libgit2.so.1.7
#18 0x0000555555941b68 in load_dives_from_tree
    (state=0x7fffffffd1c0, tree=<optimized out>, repo=<optimized out>)
    at /home/mikeller/git/subsurface/core/load-git.cpp:1827
#19 do_git_load
    (state=0x7fffffffd1c0, branch=<optimized out>, repo=<optimized out>)
    at /home/mikeller/git/subsurface/core/load-git.cpp:1867
#20 git_load_dives
    (info=info@entry=0x7fffffffd400, log=log@entry=0x555555ce0ac0 <divelog>)
    at /home/mikeller/git/subsurface/core/load-git.cpp:1904
#21 0x00005555559109b2 in parse_file
    (filename=0x5555568510f8 "https://ssrf-cloud-eu.subsurface-divelog.org/git/subsurface@ike.ch[subsurface@ike.ch]", log=0x555555ce0ac0 <divelog>)
    at /home/mikeller/git/subsurface/core/file.cpp:285
#22 0x00005555557bc528 in MainWindow::loadFiles
    (this=0x5555560f61f0, fileNames=<optimized out>)
    at /home/mikeller/git/subsurface/desktop-widgets/mainwindow.cpp:1318
#23 0x00005555556f848d in main (argc=<optimized out>, argv=<optimized out>)
    at /home/mikeller/git/subsurface/subsurface-desktop-main.cpp:98
```

@bstoeger: This seems to be a similar case to https://github.com/subsurface/subsurface/pull/4245#issuecomment-2290842053.

### Operating system:

Linux ubuntu 24.04.

### Subsurface version:

Build from a905a781784e46972ba7df0f3d06e34391802d68

### Steps to reproduce:

1. Start Subsurface
2. Synchronise with cloud

### Current behavior:

Subsurface dies

### Expected behavior:

Cloud synchronisation completes

### Additional information:

### Mentions:

bstoeger, 2024-08-17:
I can't reproduce this. sorry. This seems to trigger when deleting "redundant" events. Can you give me a small reproducing example?

Or could you run it under valgrind, to see if there is some memory corruption?

When adding new events, I get these kinds of messages:
```
INFO: non-monotonous dive stamps 2529 2440
INFO: non-monotonous dive stamps 2440 2322
INFO: non-monotonous dive stamps 2322 2240
INFO: non-monotonous dive stamps 2240 1201
```
which shows that there is a bug in the event-storage. But nothing that leads to a crash for me.

bstoeger, 2024-08-17:
> @bstoeger: This seems to be a similar case to https://github.com/subsurface/subsurface/pull/4245#issuecomment-2290842053.

Seems unrelated to me.

mikeller, 2024-08-17:
@bstoeger: This is what valgrind is giving me:

```
==40837== Invalid read of size 8
==40837==    at 0x4C3E54: _M_data (basic_string.h:223)
==40837==    by 0x4C3E54: _M_is_local (basic_string.h:264)
==40837==    by 0x4C3E54: _M_dispose (basic_string.h:282)
==40837==    by 0x4C3E54: ~basic_string (basic_string.h:804)
==40837==    by 0x4C3E54: event::~event() (event.cpp:50)
==40837==    by 0x4A284A: destroy<event> (new_allocator.h:198)
==40837==    by 0x4A284A: destroy<event> (alloc_traits.h:558)
==40837==    by 0x4A284A: _M_erase (vector.tcc:187)
==40837==    by 0x4A284A: erase (stl_vector.h:1535)
==40837==    by 0x4A284A: fixup_dc_events (dive.cpp:767)
==40837==    by 0x4A284A: fixup_dive_dc (dive.cpp:1072)
==40837==    by 0x4A284A: dive::fixup_no_cylinder() (dive.cpp:1095)
==40837==    by 0x4AF047: dive_table::fixup_dive(dive&) const (divelist.cpp:36)
==40837==    by 0x4B06B7: dive_table::record_dive(std::unique_ptr<dive, std::default_delete<dive> >) (divelist.cpp:30)
==40837==    by 0x4F79AE: finish_active_dive(git_parser_state*) (load-git.cpp:1385)
==40837==    by 0x4FEB09: dive_directory (load-git.cpp:1514)
==40837==    by 0x4FEB09: walk_tree_directory (load-git.cpp:1616)
==40837==    by 0x4FEB09: walk_tree_cb(char const*, git_tree_entry const*, void*) (load-git.cpp:1818)
==40837==    by 0x4966570: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x4966699: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x4966699: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x49667B3: git_tree_walk (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x4FDD97: load_dives_from_tree (load-git.cpp:1827)
==40837==    by 0x4FDD97: do_git_load (load-git.cpp:1867)
==40837==    by 0x4FDD97: git_load_dives(git_info*, divelog*) (load-git.cpp:1904)
==40837==    by 0x4CC501: parse_file(char const*, divelog*) (file.cpp:285)
==40837==  Address 0x12b0e150 is 32 bytes before a block of size 512 in arena "client"
==40837== 
==40837== Invalid read of size 8
==40837==    at 0x4C3E61: _M_destroy (basic_string.h:289)
==40837==    by 0x4C3E61: _M_dispose (basic_string.h:283)
==40837==    by 0x4C3E61: ~basic_string (basic_string.h:804)
==40837==    by 0x4C3E61: event::~event() (event.cpp:50)
==40837==    by 0x4A284A: destroy<event> (new_allocator.h:198)
==40837==    by 0x4A284A: destroy<event> (alloc_traits.h:558)
==40837==    by 0x4A284A: _M_erase (vector.tcc:187)
==40837==    by 0x4A284A: erase (stl_vector.h:1535)
==40837==    by 0x4A284A: fixup_dc_events (dive.cpp:767)
==40837==    by 0x4A284A: fixup_dive_dc (dive.cpp:1072)
==40837==    by 0x4A284A: dive::fixup_no_cylinder() (dive.cpp:1095)
==40837==    by 0x4AF047: dive_table::fixup_dive(dive&) const (divelist.cpp:36)
==40837==    by 0x4B06B7: dive_table::record_dive(std::unique_ptr<dive, std::default_delete<dive> >) (divelist.cpp:30)
==40837==    by 0x4F79AE: finish_active_dive(git_parser_state*) (load-git.cpp:1385)
==40837==    by 0x4FEB09: dive_directory (load-git.cpp:1514)
==40837==    by 0x4FEB09: walk_tree_directory (load-git.cpp:1616)
==40837==    by 0x4FEB09: walk_tree_cb(char const*, git_tree_entry const*, void*) (load-git.cpp:1818)
==40837==    by 0x4966570: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x4966699: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x4966699: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x49667B3: git_tree_walk (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==40837==    by 0x4FDD97: load_dives_from_tree (load-git.cpp:1827)
==40837==    by 0x4FDD97: do_git_load (load-git.cpp:1867)
==40837==    by 0x4FDD97: git_load_dives(git_info*, divelog*) (load-git.cpp:1904)
==40837==    by 0x4CC501: parse_file(char const*, divelog*) (file.cpp:285)
==40837==  Address 0x12b0e160 is 16 bytes before a block of size 512 alloc'd
==40837==    at 0x4846FA3: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==40837==    by 0x4A83D6: allocate (new_allocator.h:151)
==40837==    by 0x4A83D6: allocate (alloc_traits.h:482)
==40837==    by 0x4A83D6: _M_allocate (stl_vector.h:381)
==40837==    by 0x4A83D6: _M_allocate (stl_vector.h:378)
==40837==    by 0x4A83D6: void std::vector<event, std::allocator<event> >::_M_realloc_insert<event const&>(__gnu_cxx::__normal_iterator<event*, std::vector<event, std::allocator<event> > >, event const&) (vector.tcc:459)
==40837==    by 0x4AACDA: insert (vector.tcc:170)
==40837==    by 0x4AACDA: add_event_to_dc(divecomputer*, event) (divecomputer.cpp:302)
==40837==    by 0x4AADC9: add_event(divecomputer*, unsigned int, int, int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (divecomputer.cpp:309)
==40837==    by 0x4FADF4: parse_dc_event(char*, git_parser_state*) (load-git.cpp:853)
==40837==    by 0x4FD4BF: parse_one_line(char const*, unsigned int, void (*)(char*, git_parser_state*), git_parser_state*) [clone .constprop.0] (load-git.cpp:1348)
[... 192 lines cut ...]
==40837==  main thread stack using the --main-stacksize= flag.
==40837==  The main thread stack size used in this run was 8388608.
==40837== 
==40837== HEAP SUMMARY:
==40837==     in use at exit: 62,817,353 bytes in 207,382 blocks
==40837==   total heap usage: 1,083,917 allocs, 876,536 frees, 284,722,674 bytes allocated
==40837== 
==40837== LEAK SUMMARY:
==40837==    definitely lost: 15,104 bytes in 47 blocks
[... cut ...]
==40837== For lists of detected and suppressed errors, rerun with: -s
==40837== ERROR SUMMARY: 503 errors from 58 contexts (suppressed: 0 from 0)
[1]    40837 segmentation fault (core dumped)  valgrind ../install-root/bin/subsurface -v -v -v
```

At the moment the only reproduction that I have is in my personal dive log. I'll try to bisect this at some point to get to a smaller sample for a reproduction (assuming the problem is not linked to the structure of the git repository itself and can be reproduced with a .ssrf file).

bstoeger, 2024-08-17:
Strange. It doesn't show the origin of the data corruption.

Can you try #4289 and perhaps this patch to look for anomalies:
```
diff --git a/core/dive.cpp b/core/dive.cpp
index 81330d1fa..2dd9538b5 100644
--- a/core/dive.cpp
+++ b/core/dive.cpp
@@ -752,19 +752,27 @@ static void fixup_dc_events(struct divecomputer &dc)
 {
 	std::vector<int> to_delete;
 
+	fprintf(stderr, "fixup_dc_events(): %lu events\n", dc.events.size());
 	for (auto [idx, event]: enumerated_range(dc.events)) {
 		if (!is_potentially_redundant(event))
 			continue;
 		for (int idx2 = idx - 1; idx2 > 0; --idx2) {
 			const auto &prev = dc.events[idx2];
-			if (prev.name == event.name && prev.flags == event.flags &&
-			    event.time.seconds - prev.time.seconds < 61)
+			if (event.time.seconds - prev.time.seconds > 60)
+				break;
+			if (prev.name == event.name && prev.flags == event.flags) {
+				fprintf(stderr, "fixup_dc_events(): mark %d for deletion\n", idx2);
 				to_delete.push_back(idx);
+			}
 		}
 	}
 	// Delete from back to not invalidate indexes
-	for (auto it = to_delete.rbegin(); it != to_delete.rend(); ++it)
+	for (auto it = to_delete.rbegin(); it != to_delete.rend(); ++it) {
+		fprintf(stderr, "fixup_dc_events(): delete %d\n", *it);
 		dc.events.erase(dc.events.begin() + *it);
+	}
+
+	fprintf(stderr, "fixup_dc_events(): %lu events left\n", dc.events.size());
 }
 
 static int interpolate_depth(struct divecomputer &dc, int idx, int lastdepth, int lasttime, int now)
```

mikeller, 2024-08-17:
@bstoeger: #4289  doesn't fix it, but this should hopefully give you more information:

```
fixup_dc_events(): 7 events
fixup_dc_events(): mark 1 for deletion
fixup_dc_events(): mark 2 for deletion
fixup_dc_events(): mark 1 for deletion
fixup_dc_events(): mark 3 for deletion
fixup_dc_events(): mark 2 for deletion
fixup_dc_events(): mark 1 for deletion
fixup_dc_events(): mark 4 for deletion
fixup_dc_events(): mark 3 for deletion
fixup_dc_events(): mark 2 for deletion
fixup_dc_events(): delete 5
fixup_dc_events(): delete 5
fixup_dc_events(): delete 5
fixup_dc_events(): delete 4
fixup_dc_events(): delete 4
fixup_dc_events(): delete 4
fixup_dc_events(): delete 3
fixup_dc_events(): delete 3
==70854== Invalid read of size 8
==70854==    at 0x4C3ED4: _M_data (basic_string.h:223)
==70854==    by 0x4C3ED4: _M_is_local (basic_string.h:264)
==70854==    by 0x4C3ED4: _M_dispose (basic_string.h:282)
==70854==    by 0x4C3ED4: ~basic_string (basic_string.h:804)
==70854==    by 0x4C3ED4: event::~event() (event.cpp:50)
==70854==    by 0x4A28AF: destroy<event> (new_allocator.h:198)
==70854==    by 0x4A28AF: destroy<event> (alloc_traits.h:558)
==70854==    by 0x4A28AF: _M_erase (vector.tcc:187)
==70854==    by 0x4A28AF: erase (stl_vector.h:1535)
==70854==    by 0x4A28AF: fixup_dc_events (dive.cpp:772)
==70854==    by 0x4A28AF: fixup_dive_dc (dive.cpp:1080)
==70854==    by 0x4A28AF: dive::fixup_no_cylinder() (dive.cpp:1103)
==70854==    by 0x4AF0C7: dive_table::fixup_dive(dive&) const (divelist.cpp:36)
==70854==    by 0x4B0737: dive_table::record_dive(std::unique_ptr<dive, std::default_delete<dive> >) (divelist.cpp:30)
==70854==    by 0x4F7A2E: finish_active_dive(git_parser_state*) (load-git.cpp:1385)
==70854==    by 0x4FEB89: dive_directory (load-git.cpp:1514)
==70854==    by 0x4FEB89: walk_tree_directory (load-git.cpp:1616)
==70854==    by 0x4FEB89: walk_tree_cb(char const*, git_tree_entry const*, void*) (load-git.cpp:1818)
==70854==    by 0x4966570: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==70854==    by 0x4966699: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==70854==    by 0x4966699: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==70854==    by 0x49667B3: git_tree_walk (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==70854==    by 0x4FDE17: load_dives_from_tree (load-git.cpp:1827)
==70854==    by 0x4FDE17: do_git_load (load-git.cpp:1867)
==70854==    by 0x4FDE17: git_load_dives(git_info*, divelog*) (load-git.cpp:1904)
==70854==    by 0x4CC581: parse_file(char const*, divelog*) (file.cpp:285)
==70854==  Address 0x112c6330 is 32 bytes before a block of size 512 in arena "client"
==70854== 
==70854== Invalid read of size 8
==70854==    at 0x4C3EE1: _M_destroy (basic_string.h:289)
==70854==    by 0x4C3EE1: _M_dispose (basic_string.h:283)
==70854==    by 0x4C3EE1: ~basic_string (basic_string.h:804)
==70854==    by 0x4C3EE1: event::~event() (event.cpp:50)
==70854==    by 0x4A28AF: destroy<event> (new_allocator.h:198)
==70854==    by 0x4A28AF: destroy<event> (alloc_traits.h:558)
==70854==    by 0x4A28AF: _M_erase (vector.tcc:187)
==70854==    by 0x4A28AF: erase (stl_vector.h:1535)
==70854==    by 0x4A28AF: fixup_dc_events (dive.cpp:772)
==70854==    by 0x4A28AF: fixup_dive_dc (dive.cpp:1080)
==70854==    by 0x4A28AF: dive::fixup_no_cylinder() (dive.cpp:1103)
==70854==    by 0x4AF0C7: dive_table::fixup_dive(dive&) const (divelist.cpp:36)
==70854==    by 0x4B0737: dive_table::record_dive(std::unique_ptr<dive, std::default_delete<dive> >) (divelist.cpp:30)
==70854==    by 0x4F7A2E: finish_active_dive(git_parser_state*) (load-git.cpp:1385)
==70854==    by 0x4FEB89: dive_directory (load-git.cpp:1514)
==70854==    by 0x4FEB89: walk_tree_directory (load-git.cpp:1616)
==70854==    by 0x4FEB89: walk_tree_cb(char const*, git_tree_entry const*, void*) (load-git.cpp:1818)
==70854==    by 0x4966570: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
==70854==    by 0x4966699: ??? (in /usr/lib/x86_64-linux-gnu/libgit2.so.1.7.2)
[... 151 lines cut ...]
==70854==  If you believe this happened as a result of a stack
==70854==  overflow in your program's main thread (unlikely but
==70854==  possible), you can try to increase the size of the
==70854==  main thread stack using the --main-stacksize= flag.
==70854==  The main thread stack size used in this run was 8388608.
==70854== 
==70854== HEAP SUMMARY:
==70854==     in use at exit: 62,750,651 bytes in 206,939 blocks
==70854==   total heap usage: 1,080,512 allocs, 873,575 frees, 290,474,158 bytes allocated
==70854== 
==70854== LEAK SUMMARY:
==70854==    definitely lost: 6,856 bytes in 46 blocks
==70854==    indirectly lost: 18,679 bytes in 104 blocks
==70854==      possibly lost: 2,471,613 bytes in 1,945 blocks
==70854==    still reachable: 60,158,567 bytes in 203,981 blocks
==70854==                       of which reachable via heuristic:
==70854==                         newarray           : 130,776 bytes in 79 blocks
==70854==                         multipleinheritance: 44,536 bytes in 49 blocks
==70854==         suppressed: 0 bytes in 0 blocks
==70854== Rerun with --leak-check=full to see details of leaked memory
==70854== 
==70854== Use --track-origins=yes to see where uninitialised values come from
==70854== For lists of detected and suppressed errors, rerun with: -s
==70854== ERROR SUMMARY: 2640 errors from 54 contexts (suppressed: 0 from 0)
```

PR #4289 (2024-08-19): Fix sorting of events

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [x] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fixes sorting of events. The wrong order messed up the profile. And perhaps was also responsible for other weirdness.

### Changes made:

1) Fix less than operator for events.
2) Trivial code hygiene improvement.

Fixes #4286.
closed 2024-08-19 by bstoeger