subsurface/subsurface #4287: Latest Dives stored in cloud do not show up on phone.

ShadowWizard11, 2024-08-16:
### Describe the issue:

- [ ] Bug
- x Change request
- x New feature request
- x Discussion request

### Issue long description:

Dives synced to cloud show up on multiple computers, but not in the subsurface app on android.

### Operating system:

Windows 10 computer at home.  Windows 10 computer at work, Android 12 phone

### Subsurface version:

Official and latest (As of this date) android 3.4.7
Offical and latest (Downloaded today) on Windows 10 (6.0.5231)
### Steps to reproduce:

Add dives from dive computer (From dive 466-477.  Computer shows dives from 1-477) (Was only able to do this once, as I have not done dives since)
Upload dives to cloud
Download dives on phone (failed, no error, but only have 1-465)
Download dives on computer at work (proves dives are stored in cloud)

### Current behavior:

See "Steps to reproduce"
### Expected behavior:

Dives show up everywhere the same.
### Additional information:

### Mentions:

mikeller, 2024-08-17:
@ShadowWizard11: Have you tried logging into your Subsurface cloud account on https://cloud.subsurface-divelog.org/ and checking if your latest dives are showing up there?

ShadowWizard11, 2024-08-17:
I have not, but if it happens again I will check that first.  Since I posted (I just checked now) its working.
Without knowing the inner working of Subsurface at all and working only from my personal IT experience if I had to make a guess I would say the cloud servers for the program on the computer, and either Android, or mobile as a whole are different and there was a sync delay.

jb2cool, 2025-02-07:
@ShadowWizard11 can i ask you to consider if this issue can be closed. To my understanding this is all working for you currently

ShadowWizard1, 2025-02-07:
So yes, dirkhh confirmed by suspicion, it was likely just some lag that happens from time to time.  So yes jb2cool, it looks as though the only issue was me not being patient enough.  You can close this issue.

closed 2025-02-07 by mikeller