subsurface/subsurface #4288: Calculated green ceiling disappears when gas elements are changed

OlivierV78, 2024-08-17:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

On import the green calculated ceiling appears but as soon as I modify gaz information this ceiling disappears

### Operating system:

Android14 with One UI 6.1 - French
Kubuntu 22.04 - French

Galaxy S21 with Android
Computer with Kubuntu  

Yes

### Subsurface version:

6.0.5230 on Android
6.0.5214 on Kubuntu

Yes

Official release

### Steps to reproduce:

1) Download a dive. Green calculated ceiling appears (screenshot 1)
2) Change some settings as gaz (screenshot 2) 
3) Green calculated ceiling disappears (screenshot 3)

### Current behavior:

### Expected behavior:

### Additional information:

The file of the dive is in attachment dive_ssrf.txt

### Mentions:

[image]

[image]

[image]

[dive_ssrf.txt](https://github.com/user-attachments/files/16644845/dive_ssrf.txt)

mikeller, 2024-08-17:
This is expected behaviour - if you are telling Subsurface that you have done this dive breating pure oxygen (which I hope for your safety's sake you haven't done) then Subsurface will tell you (correctly) that there was no deco ceiling for your dive.
Reading between the lines, this was a CCR dive with air diluent and EAN33 as bailout, and you were using tank sensors on your O2 and diluent?
Unfortunately at the moment there is no way to change the (1 : 1) mapping between gasmixes and tank sensor information that is assumed in Subsurface - and forcing Subsurface to use a gasmix different from what you actually used certainly isn't a good idea to get this done.
What you could do is swap the order of registration of your tank sensors (or swap the sensors on the regulators) - in this case at least the gasmix and pressure readings for the diluent will be combined correctly.
Since the dive computer isn't reporting any gasmix information for O2 on CCR dives, there is no workaround to get an entry for O2 in the list of gasmixes that could be matched with the tank sensor information.

OlivierV78, 2024-08-17:
@mikeller Thank you

I thought it was a bug because sometimes the green background appears (see new screenshot).

>  Reading between the lines, this was a CCR dive with air diluent and EAN33 as bailout, and you were using tank sensors on your O2 and diluent? 

Yes

>  What you could do is swap the order of registration of your tank sensors (or swap the sensors on the regulators) - in this case at least the gasmix and pressure readings for the diluent will be combined correctly. 

How can I do that ?
I cannot change the sensors on the regulators as one is oxy clean and one not.

[image]

mikeller, 2024-08-17:
> How can I do that ? I cannot change the sensors on the regulators as one is oxy clean and one not.

Unregister both tank sensors from the dive computer, then register them in the opposit order of how they were registered previously.

OlivierV78, 2024-08-18:
OK I will have a look.
Perhaps the order of import (from both Petrel and Perdix) also has an impact too

OlivierV78, 2024-08-22:
Hi, 

I tried to change the order but it does nothing better
The importation order doesn"t change anything too.

mikeller, 2024-08-23:
I don't quite get it why your dive computers are reporting that you are breathing a 100% O2 mix on these dives - either they are badly misconfigured, or you are doing something really dangerous.

OlivierV78, 2024-08-23:
No I changed the mix myself after import because the sensor is on my 02 CCR tank and other sensor on my air diluent tank.

If I don't change and stay with 21 % then the green ceiling appears ok, but 21 % is not what was in my tank...
I can do this change back (see capture) and then the green ceiling appears again.

Finally I don't understand why subsurface use this 21 % information (which is wrong as it is 100% O2 tank) to calculate the green ceiling because in CCR I never breathe 21 %...
It should only need the diluent information (air or trimix) and pp02 setting and not need information about the 02 tank.

[image]

OlivierV78, 2024-08-23:
But on this today dive, the 4 informations tanks were imported and the green ceiling was ok even if I changed the tanks information.

[image]

mikeller, 2024-08-24:
> Finally I don't understand why subsurface use this 21 % information (which is wrong as it is 100% O2 tank) to calculate the green ceiling because in CCR I never breathe 21 %...

Please read [this](https://github.com/subsurface/subsurface/issues/4288#issuecomment-2294852643) again.

OlivierV78, 2024-08-25:
OK perhaps something to do for the CCR in the the future for the developers...

Thank you very much for the work !

closed 2024-08-25 by OlivierV78