subsurface/subsurface #4324: Won't load software

SteveR369, 2024-09-14:
I have installed the software on a windows 11 laptop agreeing to the windows prompts. It was working fine a month ago but now will not load.
I have tried uninstalling and reinstalling, using administrator rights, setting exemption in windows defender but regardless of what I do the software will not load. All my dive history seems to have disappeared as well.
All and any help appreciated as this was by far my preferred dive log software and having to start everything again with other software would be a real pain.
Many thanks
Steve

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2024-09-14:
See https://groups.google.com/g/subsurface-divelog/c/ZPoRMO3OI-U for some hints.

Also, please refrain from cross-posting the same issue in multiple places - this just creates extra work for the people in the project.

mikeller, 2026-10-17:
Hi @SteveR369,

This matches a known class of "worked before, won't launch now" Windows issue that already has a troubleshooting thread with relevant steps: https://groups.google.com/g/subsurface-divelog/c/ZPoRMO3OI-U. Worth noting: your dive data lives separately from the application itself, so a launch failure is very unlikely to have deleted your dive history, once launching is sorted, your dives should still be there (or recoverable from cloud storage if you use it).

Closing this and pointing at that existing thread rather than treating it as a new defect. Please reopen with specifics (error messages, Windows Event Viewer entries) if working through that thread doesn't get you launching again.

closed 2026-10-17 by mikeller