subsurface/subsurface #4329: Import of FIT files into Subsurface app on IPhone or IPad

MarkusAndrea, 2024-09-20:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X ] New feature request
- [ ] Discussion request

### Issue long description:

I‘ve two issues:
1) I can‘t Import Suunto D5 via Bluetooth. It works only via cable on my Mac.
2) Please built an option into the iPad and iPhone version to import FIT files, as it is possible in the Mac version.

### Operating system:
 iOS 
 IPad, IPhone

### Subsurface version: 6.0.5231

 latest on IPad and IPhone
 yes
 official 

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2024-09-21:
I don't think there is anything in particular on the Subsurface side that would make it difficult to import .FIT files on a mobile device (@dirkh, @torvalds please correct me here if I am wrong). But from memory iOS makes it quite difficult to deal with files on the device, unless the app handling the files is officially registered with Apple as the 'owner' of the file type.
How are you generating or moving the .FIT files onto your iOS device?

mikeller, 2026-10-17:
Hi @MarkusAndrea,

Thanks for reporting this FIT-file import issue on iOS. Could you confirm the exact steps you're using to get the FIT file into Subsurface on your iPhone/iPad, specifically how you're accessing/selecting the file (Files app, AirDrop, share sheet, etc.)? iOS's sandboxed file access can behave differently depending on the method, and we want to make sure we're looking at the same scenario you're hitting.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.