subsurface/subsurface #4347: Seac Apnea Hr

minomit, 2024-10-09:
I would like to know if the seac apnea hr is supported, since I used subsurface with the cressi drake (it worked perfectly), I just bought the seac but I can't download the logbooks. How can I use it? If it is not supported, can I make my seac available for testing, or do you recommend that I send it back and replace it with a supported one? thanks

jefdriesen, 2024-10-29:
Can you try to download your dive computer with logging enabled? Select the Seac Screen or Action as the dive computer, and enable both the "Save libdivecomputer logfile" and "Save libdivecomputer dumpfile" checkboxes. When finished attach both files to the bugreport.

jefdriesen, 2025-08-11:
How many dives are there on your dive computer?

Can you download your dives with the Seac Sync application, and then share the log file (main.log) and database file (divesDB.db)? The location of those files can be found in the settings of the Seac Sync application.