subsurface/subsurface #4348: Meta: Please tag commit of latest weekly release

mguentner, 2024-10-09:
It would be super helpful to have the current weekly release tagged here in `subsurface/subsurface` to create automated package update pipelines for distributions.

### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description

I am the maintainer of the `subsurface` package in [nixpkgs](https://search.nixos.org/packages?channel=unstable&show=subsurface&from=0&size=50&sort=relevance&type=packages&query=subsurface). There are [multiple](https://github.com/Mic92/nix-update) [tools](https://nix-community.github.io/nixpkgs-update/), even [bots](https://nix-community.github.io/nixpkgs-update/r-ryantm/#r-ryantm) available that automatically fetch the latest tag of a repository and update the attributes of the derivation accordingly.

Currently there is no semantic link from what is published on the website to the `nightly-builds` release and finally to the actual commit.  A lot of scripting is necessary to get to the commit that was used to build the nightly release that was selected to be published on the website as the "weekly release". The release body in `nightly-builds` only points to the merged pull request GitHub page, not the actual commit that was merged.

### Desired Outcome:

Publish the weekly release here on `subsurface/subsurface`, currently this would mean having the following tag:

**6.0.5231** -> a905a781784e46972ba7df0f3d06e34391802d68

A prefix (no suffix!) would also be possible, suggestion: **weekly-6.0.5231**
This would allow to have automation in place to get the latest weekly Subsurface "weekly" release into nixpkgs asap.

### Operating system:

NixOS, MacOS, Linux in general (with Nix)

mikeller, 2024-10-09:
Hi @mguentner.

I think we are talking about the same thing here, but this seems to be doing what you are asking for:

```
➜  subsurface git:(master) ./scripts/get-changeset-id.sh 5231
38a0050ac33566dfd34bf94cf1d7ac66034e4118
```

But for some reason this is different from the changeset that you think this should be pointing to - not sure why?

> 6.0.5231 -> https://github.com/subsurface/subsurface/commit/a905a781784e46972ba7df0f3d06e34391802d68

dirkhh, 2024-10-09:
The current release is the latest commit in the `current` branch

mguentner, 2024-10-10:
@mikeller: My path to get to the commit was

* https://subsurface-divelog.org/current-release/?lang=en (6.0.5231)
* https://github.com/subsurface/nightly-builds/releases/tag/v6.0.5231-CICD-release
* Follow the mentioned PR https://github.com/subsurface/subsurface/pull/4198
* End up there: https://github.com/subsurface/subsurface/commit/a905a781784e46972ba7df0f3d06e34391802d68

@dirkhh Thanks for pointing out the `current` branch, I was not aware of it! Running `scripts/get-version.sh` there gives `6.0.5231-local`. :+1:

mikeller, 2024-10-10:
@mguentner: Makes sense. But this is only the last commit on the pull request - due to the rebasing that happens on merge there may be changesets that came after it in `master` at the point that the pull request was merged, and so `38a0050ac33566dfd34bf94cf1d7ac66034e4118` is the changeset that the 6.0.5231 release was built from.
`./scripts/get-changeset-id.sh` will give you the changeset for any given release.

mguentner, 2024-10-10:
Thanks for the input. I was able to create an `updateScript` that uses the `current` branch and determines + updates the version automatically. More in [this PR over at NixOS/nixpkgs](https://github.com/NixOS/nixpkgs/pull/343701).

I suggest to describe the relationship between the release available on the website and the `current` branch in the `README.md` or `INSTALL.md`.

closed 2024-10-10 by mguentner