subsurface/subsurface #4357: Dive site management - country field

kluuzna, 2024-10-31:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Entry of "Country" field in "Dive site management" is not retained/stored

### Operating system:

Windows 11 Pro OS Build: 22631.4317

Lenovo laptop

### Subsurface version:

6.0.5281

Yes on 6.0.5276 

Official releases

### Steps to reproduce:

1) Click on "globe" - Edit dive site
2) Enter country name in "Country" filed
3) Click on "Done"
4) Returned to dive list, but country not shown

### Current behavior:

Entered country name is not retained/stored

### Expected behavior:

Country name to be stored as was previously the case

### Additional information:

### Mentions:

mikeller, 2024-11-02:
@kluuzna: Can you please test if this is fixed for you in https://github.com/subsurface/subsurface/suites/30407436781/artifacts/2136048667?

kluuzna, 2024-11-02:
Many thanks Mike, fix worked for me!

closed 2024-11-02 by mikeller

PT1975, 2024-12-06:
Using v5276 on Windows 11, issue is readily repeatable using the steps to reproduce in the original bug report. This is when editing or creating a new dive site. Sometimes, it works as intended but I'm unable to determine the difference (tried changing other parameters such as coordinates without success).

PR #4359 (2024-11-02): Desktop: Fix Saving of the Divesite Country.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix a bug in the edit divesite country command preventing the country of the divesite from ever being saved.

### Changes made:

### Related issues:

Fixes #4357.

### Additional information:

### Release note:

### Documentation change:
commit 0e7d3a64 (2024-11-02): Desktop: Fix Saving of the Divesite Country.

Fix a bug in the edit divesite country command preventing the
country of the divesite from ever being saved.

Fixes #4357.