subsurface/subsurface #4358: Mares Puck 4

PT1975, 2024-11-01:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Mares Puck 4 will not connect to subsurface via BLE on either PC or IOS (v6.0.5276). Puck was set to BLE mode and BT enabled on both devices. Failed to detect the device. PUCK is detected on BLE via the Mares/SSI app (IOS) and successfully downloads dive data.

### Operating system:

PC: Windows 11 23H2 22631.4391
IOS: 18.1

PC: Samsung Galaxy Book 2 
IPhone: Apple Iphone 12

Yes (As per description)

### Subsurface version:

v6.0.5276 Official Release, dowloaded from website via windows installer. 

Not tried on other subsurface versions, but I believe support for PUCK4 is a recent addition.  

### Steps to reproduce:

PC:
1. Power on dive computer, select BLE option from menu as per operating manual. Confirm BLE and MAC address is shown on display
2. Confirm bluetooth enabled and will detect other BLE devices. 
3. Start subsurface, verify correct version number displayed in about box. 
4. open cloud storage
5. select import from dive computer
6. select vendor = Mares; Dive Computer = Puck 4
7. check the "choose Bluetooth download mode" option
8. Remote bluetooth device selection dialog box is displayed
9. click scan. --> scan completes successfully message comes up but no devices listed. 
10. trying other options like turn on/off bluetooth; foce BLE mode; force classic mode has no effect on outcome.
11. Selecting show all bluetooth devices shows another BLE device with different MAC address to the DC. 

IOS: 
1. Power on dive computer, select BLE option from menu as per operating manual. Confirm BLE and MAC address is shown on display
2. Confirm bluetooth enabled on IOS device and that subsurface has Bluetooth permsisions.  Check iphone connects to other BT devices. 
3. Start subsurface, verify correct version number displayed in about box. 
4. select dive management/download from DC
5. select vendor name = Mares; dive computer = Puck 4
6. Connection field is empty/unelectable
7. Select RESCAN - Download button and quit button enabled. connection field is still empty/unelectable
8. Select Download --> Displays connecting to BLE device --> states info: Import Error: Input/Output Error.
9. Only functioning option now is to select quit. 

### Current behavior:

Unable to find the dive computer on IOS or Windows 
### Expected behavior:

Expect to see the DC and download logs. 

### Additional information:

### Mentions:

jefdriesen, 2024-11-13:
On Windows you first need to pair the bluetooth dive computer using the Windows control panel. That's because the bluetooth api used by subsurface can only discover paired devices.

PT1975, 2024-11-13:
> On Windows you first need to pair the bluetooth dive computer using the Windows control panel. That's because the bluetooth api used by subsurface can only discover paired devices.

@jefdriesen - thanks for your response. 

The Puck 4 uses BLE so I assume is needing to use "Bonding" rather than traditional BT "pairing". I can't get it to "pair" at all using the standard Windows or IOS Bluetooth discovery screens - but as I've said, it works on iPhone for the Mares App as BLE normally does. 

Note that I've also tried the "Force LE mode" option which suggests it should use BLE bonding, but I've had no success either.

lpokeefe, 2024-11-16:
I also have the same problem as the submitter, with the exact same steps and results as documented above.
I am also iOS 18.1 on iPhone 13  Subsurface version 6.0.5276

I was able to import data into subsurface on the Mac version 6.0.5276-CICD-release

Would greatly appreciate a fix for this!

lpokeefe, 2024-11-19:
Update from me here... I was looking at the subsurface-mobile manual and noticed the Bluetooth toggle in the advanced settings menu.  I turned it on, and lo and behold, I was able to connect to the dive computer.  I know it connected, as it returned a message of "No new dives downloaded from dive computer" rather than the previously seen error messages.  (I downloaded all the dives previously via bluetooth to my Mac). Here is a screenshot showing the puck listed in the connection field.

I hope this helps.  At the very least this gives me a workaround to connect the computer to my iPhone.
[image]

jefdriesen, 2024-11-25:
That workaround is needed for all models where the bluetooth detecting code in subsurface hasn't been updated yet to recognize the new model.

PT1975, 2024-12-01:
@lpokeefe - Thanks for the post. I thought i had tried this permutation of settings too without success but I  can confirm this workaround is working for me on IOS. I've tried similar with PC to no avail. 

Thanks to you all for fast replies.