subsurface/subsurface #4364: Divesoft Freedom+ log support (header "DiVE")

jakubhecko, 2024-11-07:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

Add support for Divesoft logs (*.DLF) with new header format starting with signature "DiVE".
This is a log header format only, the rest of the dive profile data format is identical.
Main differences from a v1 header "DivE":

-  Length of the header - 64B instead of 32B
- Time is handled as a separate UTC time and a timezone
- Added average depth, surface interval and device S/N
- Some values are expanded to achieve higher max value and/or better precision.

This log header format is already supported (more than a year) as a part of libdivecomputer, unfortunately Subsurface uses a custom parser for "Import / Import log files".

Some users tried already adjusting the dive log signature from "DiVE" to "DivE", with some success, unfortunately the extra 32B of header makes for erroneous data shown in the dive profile.
https://groups.google.com/g/subsurface-divelog/c/FWh8gwfE444

### Operating system:

Not relevant, this is a parser request.

### Subsurface version:

(CICD) Release: 6.0.5285

Official release

### Steps to reproduce:

Open new logbook, try importing a log file "Import / Import log files" from any Freedom+.

### Current behavior:

The dive is not imported.

### Expected behavior:

Dive being imported and properly parsed.

### Additional information:

See the log file and "DiVE" header format attached [00000031_DiVE.zip](https://github.com/user-attachments/files/17657890/00000031_DiVE.zip)

### Mentions:

@jefdriesen @jefdriesen

glance-, 2024-11-07:
The subsurface custom parser of the Divesoft files is way older than the support in libdivecomputer, and was a reverse engineered parser, and not based on any information from Divesoft.

It has outlived its usefulness and probably the best way would be to throw it away and replace it with the libdivecomputer parser.

mikeller, 2024-11-07:
> It has outlived its usefulness and probably the best way would be to throw it away and replace it with the libdivecomputer parser.

Agreed on this - and it should probably be relatively straightforward to add support for the file system transport to this dive computer model in libdivecomputer.

glance-, 2024-11-08:
Shure, file system transport would be nice for "downloading" , but I think the shorter path to success would be to replace the content of parse_dlf_buffer() with a call to libdc_buffer_parser, and just replace the content parser with the one from libdivecomputer, just like it's done for ostctools_import(), datatrak_import() and smartrak_import().

mikeller, 2024-11-08:
> Shure, file system transport would be nice for "downloading" , but I think the shorter path to success would be to replace the content of parse_dlf_buffer() with a call to libdc_buffer_parser, and [...]

Makes sense - I wasn't aware that we are already using this pattern.
In this case we should probably switch the .FIT file import to use 'Import log file' as well - now that more manufactuers than just Garmin are offering the download of logs as .FIT, users are struggling to find this.

mikeller, 2024-11-13:
@jakubhecko: #4367 should fix this - can you please test with one of the artifacts in the pull request?

jakubhecko, 2024-11-13:
@mikeller I have tested the windows build 6.0.5286-patch.1 with the test logs and all valid logs are passing, thanks a lot !
Feel free contacting me, if anything other comes up in the future around Divesoft devices.

mikeller, 2024-11-19:
Fixed by #4367.

closed 2024-11-19 by mikeller